pageauc/pi-timolo #3: Shows an error in terminal

josehb, 2015-01-18:
Hi Claude... First of all, I want to thank you for development this program, it is helping me a lot. I was working with this program but suddenly it stopped working. I wish that you could help me with this error, i don't know what happens.
[image]

pageauc, 2015-01-18:
Hi
This looks like a firmware issue.
Make sure you have the latest pi - timolo on github version 2.23
Do the firmware update per readme file.
That should fix the problem.

Let me know if this resolves your problem.
Claude

josehb, 2015-01-18:
I just made the whole installation process again, but I get the same error. Even when I try to display the camera pi in python shell I get another error when I import the library "picamera". Do you know if there is any way to fix this?
[image]
[image]

pageauc, 2015-01-18:
pi-timolo assumes a working pi camera module.  It is possible a library
module install failed and did not
complete due to communications or some other problem.  Here are basic steps
to trouble shoot
http://www.raspberrypi.org/learning/python-picamera-setup

Stop before the step CAMERA PROGRAMMING: CAPTURE WHEN ACTIVATED

If the above does not work can you do the following for me.

What output do you get when you manually install python-picamera library.
It should say you already have the latest version.
if there is a problem with python-picamera install then it may be due to a
failed install.  sometimes this can happen.
Anyway let me know and we can walk through additional steps to fix an
incomplete install.

sudo apt-get install python-picamera

did you do a firmware update?

sudo rpi-update

Is there enough room on the SD card

df -h

Does the camera work with raspistill and can you view the image?

raspistill -o test.jpg

It would appear the problem is with picamera module seeing a hardware or
incomplete software problem.

pageauc, 2015-01-18:
Read you post again and saw that the camera was working OK with pi-timolo
and failed later on.  This means the original install was successful but it
is a good idea to do the trouble shooting steps.  Might be a hardware
problem (camera, SD card etc).  If so then some of the troubleshooting
steps will point you in the right direction.  Also try a hard boot of the
RPI.  sudo halt then unplug and re plug USB power connector.

josehb, 2015-01-18:
I did the steps until "CAMERA PROGRAMMING: CAPTURE WHEN ACTIVATED", but when i put "raspistill -o image.jpg" on terminal it shows me: "mmal: Received unexpected camera control callback event, 0x4f525245" and stops here. I have been looking for a solution for this error, but this looks like a camera module problem, or even that the ribbon which connects the raspberry with the module has been damaged. Tomorrow, i will borrow another camera module and try to make it run.
I thank you for the time that you take to help me. i will write you to tell you if it solved.

pageauc, 2015-01-18:
If you have not already tried it, try re seating the camera ribbon cable
into the RPI as well as at the camera end.  There are youtube videos and
instructions on how to do this. I have been pretty rough sometimes with my
camera modules and found they are very durable.  The ribbon cable
connections would be the first thing I would try.

josehb, 2015-01-18:
I have been plugging and unplugging the ribbon and the usb connector, too. But nothing changes. When i put raspistill in terminal window, it shows me the same error and the led in camera module turns on. In raspberry forums, some people says that they bought another camera because that problems always occurs and there is no solution yet. Do you know how i can test if the module still works?

[image]
[image]

pageauc, 2015-01-18:
Looks like a 3rd party camera module. Try contacting the vendor if you can.
They may have diagnostic tool or warranty policy

josehb, 2015-01-18:
Is there any way to make the pi-timolo code run with a USB web cam?

pageauc, 2015-01-18:
I did have code based on opencv but not to the level of pi - timolo. The
picamera python module I believe only supports the pi camera not usb. The
pi camera module is only $25 USD at Newark element14

watou, 2015-01-18:
I have had serious performance and reliability issues with USB webcams on the Raspberry Pi, including the more modern USB standardised interface (as well as with the old, vendor-specific drivers).  I think you would find it to be a problem.  The quality and throughput of the pi camera is vastly superior to USB.  There is USB interface code for DSLR cameras that work on the RPi, which I've used in a commercial photo booth, but they are also very slow in comparison.

closed 2015-05-22 by pageauc