pageauc/pi-timolo #4: Error when trying to run/setup grive

andrewmetersky, 2015-02-09:
I am receiving this error in terminal:

./grive: error while loading shared libraries: libboost_filesystem.so.1.49.0: cannot open shared object file: No such file or directory

I think it must be some type of linking issue.  Any idea?

pageauc, 2015-02-09:
Try installing this library.  Somehow it got missed but not sure how.  Let
me know if this solves your problem.

sudo apt-get install libboost1.49-dev

andrewmetersky, 2015-02-10:
I'm still getting the same error.  Is there a specific place i need to install this?

pageauc, 2015-02-10:
Libraries are installed per the installation apt-get scripts.  You may have
a corrupt system or an corrupt installation.

Do you get any errors during the setup.sh script?.
Are the grive, .py and .sh scripts executable?
Are you able to run sudo ./grive -a

If you still have problems you may want to try another RPI boot SD card
with the latest raspbian distro and repeat the installation per the
pi-timolo installation instructions and setup.sh script.

andrewmetersky, 2015-02-10:
I will try this! Thanks for your time.

andrewmetersky, 2015-02-10:
So, I get not errors from the .sh or .py scripts.  I am looking in /usr/bin/ where all the lib files are and I do not see libboost_filesystem.so.1.49.0, but when i use sudo apt-get install libboost1.49-dev the output is:

Reading package lists... Done
Building dependency tree
Reading state information... Done
libboost1.49-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

pageauc, 2015-05-22:
Until grive is fixed this item will remain closed.

closed 2015-05-22 by pageauc