pageauc/pi-timolo #5: Combine motion and timelapse

krist10an, 2015-04-02:
Thanks for the excellent program, worked right out of the box for me :+1: 

Would it be possible to combine motion detection and timelapse in order to create a timelapse movie that updates more often when there is motion? You can see what I had in mind here:
https://youtu.be/gyeTsl8ghcQ?t=3m

I like the effect, allthou the street in the video is a bit too busy to see the effect very clearly :-)

pageauc, 2015-04-02:
> Thanks for the excellent program, worked right out of the box for me [image:

Krist.

Looks like a great idea.  Pi Timolo does have video when motion is detected
but I like the idea of increasing the rate once motion is detected.  We
will be going back to Canada shortly so I might not get chance to do this
right away but definitely will implement and will let you know. Should just
need an additional function and boolean toggle.   I suspect there will need
to be three (3) additional settings. One to turn the fast motion capture
setting on and off.  Second for fast motion time between images and third
would be the duration after motion detected before going back to normal
motion detect mode.

Is this what you were thinking?.
Claude ..

krist10an, 2015-04-02:
Hi Claude,

Thanks, sounds like a nice way of doing it.

Come to think of it, it might be possible to get this effect by just set the output directory for the motion detection and timelapse in the same output folder and using the same name prefix? In that case you get extra images every time there is motion and when generating a movie it will look like it slows down when motion is present. In that case I guess the numbering might not work directly, but date/timestamps should.

pageauc, 2015-04-30:
Hi Krist

I have fixed some bugs and added the fast time lapse feature.  Was
wondering if you wanted to test pi-timolo version 2.3 before it is posted
to github.  I have attached the two new files.

I have fixed motion video, text colour display bugs and added new quick
timelapse after motion detected option. I have attached a new version 2.3
of pi-timolo.py and a new version of config.py since there are new
variables and some cleanup.  I have done testing and everything seems to
work OK but would appreciate if you give it a try.  If you turn off write
text to images then the quick time lapse will work faster.  I was thinking
of only writing text to first image but still thinking about this.  I have
not updated github and will wait till you have had a chance to test for me.

When you copy the attached files to the RPI you should run dos2unix command
for the new pi-timolo.p file.  if you do not have dos2unix on the rpi it
can be added with the following command.  This will avoid any init.d
problems when running headless as a daemon

*sudo apt-get install dos2unix*
*cd ~*
*cd pi-timolo*
*dos2unix pi-timolo.py*
*chmod +x pi-timolo.py*

to test from putty ssh or desktop terminal execute the following

*cd ~*
*cd pi-timolo*
*sudo ./pi-timolo.py*

Note.  The old config.py and templates files will not work due to new and
deleted variables.

Thanks Claude ...

-- 
See my YouTube Channel at http://www.youtube.com/user/pageaucp

# User Configuration variable settings for pitimolo
# Purpose - Motion Detection Security Cam
# Created - 13-Dec-2014
# Done by - Claude Pageau
 
configTitle = "pitimolo default configuration motion and timelapse 1080p images"
configName = "pitimolo-default-config"

# These settings should both be False if this script is run as a background /etc/init.d daemon
verbose = True             # Sends detailed logging info to console. set to False if running script as daeman
logDataToFile = False      # logs diagnostic data to a disk file for review  default=False
# print a test image
imageTestPrint = False     # default=False Set to True to print one image and exit (useful for aligning camera)

# Image Settings
imageNamePrefix = 'cam1-'  # Prefix for all image file names. Eg front-
imageWidth = 1920          # Full Size Image Width in px  default=1980
imageHeight = 1080         # Full Size Image Height in px default=1080
imageVFlip = False         # Flip image Vertically    default=False
imageHFlip = False         # Flip image Horizontally  default=False
imagePreview = False       # Preview image on connected RPI Monitor default=False
noNightShots = False       # Don't Take images at Night default=False
noDayShots = False         # Don't Take images during day time default=False  

# Low Light Night Settings
nightMaxShut = 5.8         # default=5 sec Highest cam shut exposure time. IMPORTANT 6 sec works sometimes but occasionally locks RPI and HARD reboot required to clear
nightMinShut = .001        # default=.002 sec Lowest camera shut exposure time for transition from day to night (or visa versa)
nightMaxISO = 800          # default=800  Max cam ISO night setting
nightMinISO = 100          # lowest ISO camera setting for transition from day to night (or visa versa)  
nightSleepSec = 10         # default=10 Sec - Time period to allow camera to calculate low light AWB   
[... 656 lines cut ...]
                                takeDayImage(filename)
                            else:
                                takeNightImage(filename)
                        motionNumCount = postImageProcessing(motionNumOn, motionNumStart, motionNumMax, motionNumCount, motionNumRecycle, motionNumPath, filename, daymode)
                    if motionFound:
                        # =========================================================================
                        # Put your user code in userMotionCodeHere() function at top of this script
                        # =========================================================================                    
                        userMotionCodeHere()
                        dotCount = showDots(motionMaxDots)           
                else:
                    dotCount = showDots(dotCount)  # show progress dots when no motion found
        data1 = data2
    return
    
#-----------------------------------------------------------------------------------------------    
if __name__ == '__main__':
    try:
        Main()
    finally:
        print("")
        print("+++++++++++++++++++++++++++++++++++")
        print("%s - Exiting Program" % progName)
        print("+++++++++++++++++++++++++++++++++++")
        print("")

krist10an, 2015-05-07:
Hi Claude,

Thanks! We have a new member of the family joining very-soon-now(tm), so I cannot promise a speedy response, but I'll try it when I have some time.

krist10an, 2015-05-19:
Hi Claude,

Sorry for taking so long to test this. We just had a baby, so he got most my attention lately :baby:  I tried to download the attachments you made, but all line breaks are removed which makes it a bit difficult to test Python code :smile: . Would it be an option for you to push the changes to a branch so I can test that?

pageauc, 2015-05-19:
> my attention lately [image: :baby:] I tried to download the attachments
> test Python code [image: :smile:] . Would it be an option for you to push

Kristen

Files would be in Unix Format.

Source and instructions are on Github  at
https://github.com/pageauc/pi-timolo.

Here are brief instructions FYI.  Please note you will need the new
config.py due to new variables.  Also grive is no longer working since
google changed the docs api.

Quick Setup
===========
SSH (putty) or Desktop Terminal login to RPI and perform the following

cd ~
mkdir pi-timolo (or a folder name of your choice)
cd ./pi-timolo
wget https://raw.github.com/pageauc/pi-timolo/master/pi-timolo.tar
tar -pxvf pi-timolo.tar
# Install dependancies and required software
sudo ./setup-timolo.sh
# Initialize pi-timolo.py files, motion and test motion.
python ./pi-timolo.py
# Verify motion then ctrl-c to exit pi-timolo.py
# Edit config.py file using nano editor to change any desired settings
per comments.
# ctrl-x y to Save
nano config.py
# test edit changes.
sudo ./pi-timolo.py

Regards

Claude ...

jeffisfast, 2015-05-19:
Thanks Claude for this.  

I'm wondering if there is a way for me to get both a JPG image when motion is detected and an H264 video?  From playing with the settings it seems I can get one or the other, but not both.  I was hoping to get a single JPG when motion is detected, and then a video for whatever duration I specify in the settings.  

Jeff

pageauc, 2015-05-19:
I think you can do this by adding some of the takeVideo(filename) function
logic to the userMotionCodeHere() function.  If you have the correct
configuration settings for just taking a single image when motion is
detected, then the user MotionCodeHere() function will trigger and take a
video for specified motionVideoTimer time period. Make sure the
motionVideoOn is set to False in the config.py.  This should generate a JPG
and a video after motion is detected.  Give it a try.

Sample code to add to userMotionCodeHere() function

        filename = getVideoName(motionPath, imagePrefix, motionNumOn,
motionNumCount)
        with picamera.PiCamera() as camera:
            camera.resolution = (imageWidth, imageHeight)
            camera.start_recording(filename)
            camera.wait_recording(motionVideoTimer)
            camera.stop_recording()

jeffisfast, 2015-05-20:
Makes sense.  Will give it a try.  Thanks again for everything.

jeffisfast, 2015-05-20:
Claude,

Thanks again.  Got it working, but had to manually build the filename since getVideoName breaks if MotionVideoOn is False. 

Separate question now - seems that sometimes my camera thinks it should be in "night" mode when in fact it should be in "day" mode.  Could you offer any suggestions for how to make it more reluctant to switching to night?

Thanks again!

pageauc, 2015-05-20:
Change the twilightThreshold setting.  This will adjust when the camera
tries to switch from night to day.

If the ambient day lighting is lower then reduce from default 90 to a lower
setting.  You might have to experiment or check detailed logs at dusk or
dawn.  I have a camera under a canopy that does not get lighting directly
from the sky.  I set it to 50 and this seems to help although sometimes if
it is a darker day it tries to switch to night mode once in a while.
Similar issues exist if camera is indoors with lower ambient lighting.

You can also increase the wait time in the takeDayImage() function to maybe
1 or 2 seconds since this will allow the camera more time to establish a
good white balance. This will also slow the camera down if you are trying
to catch motion as quickly as possible.  If there is good lighting I have
found you can get away with .1 second delay.  It usually takes a little
trial and error.  My default settings are for a camera that is out in the
open with sky above.

I am working on automating this setting in the checkIfDay() function but it
is a little tricky since you are looking for extreme pixel averages near 0
or 255 usually near dusk or dawn. This works but I am trying to not slow
the camera performance and that is the tricky part.

Hope this helps
Let me know how you make out

Regards
Claude ...

krist10an, 2015-05-21:
Claude,

I was not able to download the files, so I had to copy&paste them from the source here which was causing me issues since newlines were stripped for some reason. Anyway, I got that sorted out and tested the code now (finally).

Quick motion detection works and it generates several motion detection images upon detection! 6 images were captured in my first test, with showDateOnImage disabled it captured 9 images in my second test. Seems to work great, is there anything special you'd like me to test?

How do you suggest to use both the motion and timelapse images together in a movie? Motion images are stored in motion/ and timelapse in timelapse/, while the makemovie.py only uses images from timelapse/ as far as I can see.

pageauc, 2015-05-21:
I would recommend you change settings so both timelapse and motion go into
into the same subfolder. Also change settings both have the same prefix for
the filename settings so all files are named the same except the date/time
sequence
This will capture a regular sequence of events on a steady frequency and
them capture quick timelapse when motion is found.  To do this I would
recommend setting both time lapse and motion with number sequence turned
off.  This will put the file naming in date/time order. . That way you can
sort them by filename when you make a movie.

This setup will work for shorter sequences that you are interested in that
have a longer leadin but need more detail when motion is detected.  An
example would be animals that take a while to show up (lead in timelapse)
and capture some activity then leave after a while (quick timelapse on
motion detect).  If you have an external hard drive you can connect to the
pi then that would allow faster saving of more images and less chance of
running out of space on the SD card.

If you want to speed things up then reduce image dimensions to maybe eg
720p or similar instead of 1080p.

Let me know how you make out.

Claude ...

krist10an, 2015-05-22:
Hi Claude,

Thanks for describing how to do it for me :-)
I tested the combined motion and timelapse by outputing them to the same folder with date filenames, and it seems to work great! Unfortunately there wasn't really anything interesting out the window I pointed the camera so I don't really see the effect in the generated video. I'll have to put it in another window and try again!

Thanks again for the fix, much appreciated :+1:

pageauc, 2015-05-22:
Kristen

Time lapse is deceptive sometimes.  We humans live at a different faster
time frame that most of nature.  You might want to think of watching grass
or plants grow and catching birds with quick motion once in a while.  You
will need to set an appropriate time interval.  I usually calculate how
long the video should be after editing.  At 30 FPS for a 3 minute video
that would be approx 11k images.  Then calculate the time period you want
to cover.  To see grass grow take 3 or 4 days @ 12 hours per day would be
approx 2700 images per day or about every 3-4 minutes timelapse interval.
You should then factor in any fast motion sequences.and this depends on the
quick timelapse duration you set (lets say 10 seconds at fastest rate
possible or 0).  Make sure the camera does not pick up too much random
motion of vegetation or other movement.  You can set the motion threshold
higher so a pixel would need to change more before being counted.  default
is 10 so raise to maybe 40 or 50 and do a little test.  I like to keep
sensitivity lower if possible depending on how close to the camera and how
large the moving objects are.  Usually 150 or 200 is a good start.  There
is some trial and error but it is nice when it is tuned and running good
since you can just let it run unattended.  I was pleasantly surprised at my
winter time lapse project.

It is here if you have not already seen it.  https://youtu.be/pFkt0ZOrPzg
There is some wonderful time lapse in the documentary Antarctica a year on
the ice http://www.imdb.com/title/tt2361700/?ref_=fn_al_tt_1
Let me know what you discover.
Good Luck Claude ...

pageauc, 2015-05-22:
I believe this enhancement is implemented and tested.  I had fun with this and hope it is useful for others

closed 2015-05-22 by pageauc

pageauc, 2015-06-12:
fyi I have released version 2.6 of pi-timolo at
https://github.com/pageauc/pi-timolo

*Version 2.6 release notes*

   - fixed bug that caused a hang when video mode was True
   - Added a new quick time lapse after motion detected feature.
   - cleaned up console display
   - misc bug fixes
   - Added gdrive binary for easy syncing of files with internet based
   google drive seehttps://github.com/odeke-em/drive/releases amd binary
   and https://github.com/odeke-em/drive for additional implementation
   details

see Readme.md for details.

Please note, I am also using another very good web based interface that is
mentioned in my Readme.md file extract below.  I have this setup with a 3
second buffer prior to motion.  Let me know if you have any questions about
my setup.  I still like pi-timolo for headless security camera operation.

If you are looking for a good web based interactive RPI camera interface I
would highly recommend this application. It is easy to install and works
well. But I still prefer pi-timolo for remote headless camera situations.

http://elinux.org/RPi-Cam-Web-Interface

This interface can also be setup to use gdrive syncing. Contact me if you
need details.