pageauc/pi-timolo #8: In makemovie.py the print statements are missing paranthesis

bradleycundari, 2015-07-20:
I noticed today when I went to check over the makemovie.py script that all of the print statements are missing parenthesis which results in a syntax error. Is there a reason behind this that I'm missing?

pageauc, 2015-07-20:
The original code was written for python 2.7.I updated pi-timolo.py to be
python 3 compliant but obviously missed makemovie.py.  I have updated the
pi-timolo repo files and tar file.  If you want to do a quick fix change to
the pi-timolo folder on the RPI and execute

2to3 -w makemovie.py

This will convert the file to be python3 compatible.  then delete the bak
file.

You can also download an updated tar file or the updated makemovie.py from
the github repo

Let me know if you have problems

Regards Claude ....

pageauc, 2015-07-20:
FYI

I have recently posted a python opencv motion tracking demo here
https://github.com/pageauc/motion-track
here is the YouTube demo video https://youtu.be/09JS7twPBsQ

Regards
Claude ...

bradleycundari, 2015-07-20:
I had a feeling it might be because it was for 2.7. Slowly trying to immerse myself to help in projects, very new to GitHub. This is a very well written software package. Thanks Claude!

pageauc, 2015-09-03:
New update to 2.8 released.  update gdrive and changed mencoder to avconv in makemovie scipts

closed 2015-09-03 by pageauc