pageauc/pi-timolo #9: Raspbian jessie : no package for mencoder

jschaeff, 2015-09-01:
Hi,

since mencoder has been removed from debian jessie (hence raspbian), the setup.py complains.
How could we avoid the problem ?

pageauc, 2015-09-01:
mencoder may come back but it looks like it has not been maintained for a while and it is not compatible with jessie but may get resolved later.  I will take a look at maybe mplayer2 or mpv per 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=732159
Might take a while.  I also have an updated binary for gdrive so might bundle these together and update pi-timolo. 

Thanks for letting me know about this problem.  I am sure there are others in the same situation.

closed 2015-09-01 by pageauc

jschaeff, 2015-09-01:
maybe ffmpeg would do the trick for building the timelapse video.

Like : 

`ffmpeg -framerate 10 -i "tl_%04d.jpg" -c:v libx264 -r 10 timelapse.mp4`

pageauc, 2015-09-01:
See this link about ffmpeg
http://superuser.com/questions/286675/how-to-install-ffmpeg-on-debian

It would appear ffmpeg will also be removed from jessie.  Might try the
sudo apt-get install libav-tools
and see if avconv is an alternative that works
Claude...

pageauc, 2015-09-01:
Here is doc and examples for avconv

https://libav.org/avconv.html#Video-and-Audio-file-format-conversion

For creating a video from many images:

avconv -f image2 -i foo-%03d.jpeg -r 12 -s WxH foo.avi

I will give this a try and see if it is a good solution.that will be
supported

Claude ..

pageauc, 2015-09-02:
Johnathan

I have attached a test bash script that will use avconv to create a movie.
Can be avi or mp4.
The script creates a temporary working folder and creates numbers soft
links pointing to the source image folder.  This could be motion. timelapse
or something else as desired.

It then runs avconv against this temporary folder and creates a dailymovie
file that is named with date and time to be unique.

When you copy the script to the raspberry pi  into the pi-timolo folder or
whatever you named it then make it executable

chmod +x *.sh

I use filezilla with fstp protocol to move files to/from my RPI from my
windows computers.  It works quite well

Let me know how you make out.

if you do not have avconv install then use the command below

sudo apt-get install libav-tools

Let me know how you make out with your testing.

pageauc, 2015-09-02:
I have changed to working folder logic to skip the most recent file and
sort the folder by time.  This is similar logic to original.  I have
also increased the working folder numbering to 05d instead of 04d.  This
will allow for folders with more than ten thousand images although
not sure how long this would take.  This makedailymovie.sh script is mostly
for the motion folder since that would normally be limited
to a circular number sequence of 500 or so images.  I have change the
source folder in the script to motion instead of timelapse.

Anyway give it a try and let me know if there are any bugs or problems.

Thanks Claude ....

jschaeff, 2015-09-02:
Hi,

your script seems cool, but I could'nt find where to get it from. Did I miss something ? Can you send me the downloaded URL ?

pageauc, 2015-09-02:
If you downloaded it from the attachment then it should be on your local
drive somewhere.  Check for the download folder location in your email
program.

I am working on a github update for pi-timolo  It will be ready before the
end of the day.  Will let you know when if is available.
Claude ...

jschaeff, 2015-09-03:
Hi Claude,

sorry, but I don't find any attachement on github or with the mail. Maybe
Github doesn't allow attachement to mails.
Anyway, here is the command I used to generate the movie. My project will
export the photos on another host for the rendering of the timelapse.

 avconv -y -r 20 -i tl-rade-%4d.jpg -r 20 -vcodec libx264 -q:v 3  -vf
crop=1920:1080,scale=iw:ih tl2

Regards,

Jonathan

pageauc, 2015-09-03:
The new pi-timolo version is on github and the makemovie.sh and
makedailymovie.sh have been updated to avconv
See Readme.md for details https://github.com/pageauc/pi-timolo

Let me know what you think.  Claude ....