pageauc/pi-timolo #10: Takes 1 motion image then stops

303CoreyM, 2015-12-16:
Hi Claude - thank you for all your hard work on this project, it looks great, but I have run into an issue and am not sure what I am doing wrong, hopefully you can point me in the right direction. I am using a model B with a clean install of Raspbian Jessie Lite 2015-11-21. I have updated/upgraded and rpi-updates as well as several reboots. I am using the official Raspberry Pi camera and raspistill works as well as the test image from pi-timolo. I have it configured for motion detection, but when I start it either manually or from init.d it detects motion and takes the first picture then stops. When I check the status it says "failed" and the red led on the camera has gone out. I have tried turning on logging, but it just shows that it took the first picture. It is a 16GB SD class 10 card with 97% free space.

Thanks in advance for any suggestions you might be able to offer.

Corey

pageauc, 2015-12-17:
Here is a suggestion made to another user.  Try reinstalling fresh from
github and set verbose mode to True in the config.py file.  Send me the
output of the top section of the log file if things still don't work.  I
have not tested using Jessie but would not expect any problem although it
is possible.
https://www.raspberrypi.org/forums/viewtopic.php?p=863646#p863646

303CoreyM, 2015-12-17:
Hi Claude -

That sounds like a great suggestion, I should be able to give it a try 
about this time tomorrow and I'll let you know. If it does come down to 
an older version of of Raspbian I can do that too.

Thanks

Corey

303CoreyM, 2015-12-18:
I have renamed the old directory, downloaded a fresh copy, reinstalled it, set logging to verbose mode, rebooted and sadly I am still getting the same results every time I run pi-timolo. What follows is the screen capture, the last couple of lines are the pi-timolo.log contents. Hopefully this isn't too much to post, if it is please delete it and I will come up with a different way of sending the information. Thanks again for the help.

pi@raspberrypi:~/pi-timolo $ ./pi-timolo.py
------------------------------ Loading Python Libraries --------------------------------------

Note: To Send Full Output to File Use command -   python -u ./pi-timolo.py | tee -a log.txt
      Set logDataToFile=True to Send checkIfDay Data to File pi-timolo.py.log

pi-timolo.py ver 2.8
-------------------------------------- Settings ----------------------------------------------
Config File .. Title=pitimolo default configuration motion and timelapse 1080p images
               config-template filename=pitimolo-default-config
Image Info ... Size=1920x1080   Prefix=cam1-   VFlip=False   HFlip=False   Preview=False
    Low Light. twilightThreshold=90  nightMaxShut=5.800 sec  nightMaxISO=800   nightSleepSec=10 sec
    No Shots . noNightShots=False   noDayShots=False
    Img Text . On=True  Bottom=True (False=Top)  WhiteText=True (False=Black)  showTextWhiteNight=True
Motion ....... On=True  Prefix=mo-  threshold=10(How Much)  sensitivity=200(How Many)  forceTimer=60 min(If No Motion)
               motionPath=/home/pi/pi-timolo/motion
    Num Seq .. motionNumOn=True  current=1000   numStart=1000   numMax=500   numRecycle=True
               motionNumPath=/home/pi/pi-timolo/mo-pi-timolo.dat 
    Quick TL . motionQuickTLOn=True   motionQuickTLTimer=10 sec  motionQuickTLInterval=0 sec (0=fastest)
    Video .... motionVideoOn=False  Motion Video Disabled
Time Lapse ... On=False  Prefix=tl-   Timer=300 sec   timeLapseExit=0 sec (0=Continuous)
               timelapsePath=/home/pi/pi-timolo/timelapse
    Num Seq .. On=True  current=non   numStart=1000   numMax=1000   numRecycle=True
               numPath=/home/pi/pi-timolo/tl-pi-timolo.dat
gdrive Sync .. On=True  Path=/home/pi/pi-timolo/pi-timolo.sync  Note: syncs for motion images only.
Logging ...... verbose=True (Details to Console)    logDataToFile=True  logfile=/home/pi/pi-timolo/pi-timolo.log
------------------------------------ Log Activity --------------------------------------------
20151217_20:25:03 Main - Entering Motion Detect - Time Lapse Loop  Please Wait ... 

20151217_20:25:03 ..
20151217_20:26:48 checkForMotion - Found Motion - threshold=10 sensitivity=200 Exceeded ... 

20151217_20:26:50   getNightCamSettings - dayPixAve=15 ratio=0.833 ISO=666 shut=4833333 4.833 sec 
20151217_20:27:38   takeNightImage - Size=1920x1080 dayPixAve=15 ISO=666 shut=4.833 sec /home/pi/pi-timolo/motion/mo-cam1-1000.jpg 

+++++++++++++++++++++++++++++++++++
pi-timolo.py - Exiting Program
+++++++++++++++++++++++++++++++++++

Traceback (most recent call last):
  File "./pi-timolo.py", line 676, in <module>
    Main()
  File "./pi-timolo.py", line 658, in Main
    motionNumCount = postImageProcessing(motionNumOn, motionNumStart, motionNumMax, motionNumCount, motionNumRecycle, motionNumPath, filename, daymode)
  File "./pi-timolo.py", line 278, in postImageProcessing
    writeTextToImage(filename, imageText, daymode)
  File "./pi-timolo.py", line 248, in writeTextToImage
    font = ImageFont.truetype(font_path, 24, encoding='unic')
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 240, in truetype
    return FreeTypeFont(font, size, index, encoding)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 137, in __init__
    self.font = core.getfont(font, size, index, encoding)
IOError: cannot open resource
pi@raspberrypi:~/pi-timolo $ 

20151217_20:19:25 takeNightImage Size=1920x1080 dayPixAve=1 ISO=791 shut=5.736 sec /home/pi/pi-timolo/motion/mo-cam1-1000.jpg
20151217_20:27:38 takeNightImage Size=1920x1080 dayPixAve=15 ISO=666 shut=4.833 sec /home/pi/pi-timolo/motion/mo-cam1-1000.jpg

pageauc, 2015-12-18:
First try running command as sudo
Eg
sudo ./pi-timolo.py
In case it might be a permission problem

If above fails then it Looks like it might be a problem with the python
image library that is used to write text to the images. Probably due to
Jessie. Try setting image text on to False in the config.py so text is not
written to the images, then retry.  In the mean time I will install Jessie
and do some testing to check out problem.  Jessie is relatively new and has
had a few issues. I did some initial work with pi-timolo using Jessie for
some mencoder makemovie issues and had to rewrite for avconv but have not
tested recently. Anyway give this a try and let me know.
Claude
Let me know how you do.

303CoreyM, 2015-12-18:
Claude -

I tried running it as sudo, but that didn't get it to work. I set image text to False and that got it working perfectly!

Thank you!

I think for now I will grab Wheezy and get it working that way.

Thanks again.

Corey

P.S. Is there a page I haven't found yet for recommended settings? I would like to set this up as remote game cam and won't be able to get back to it for several months. I am thinking maybe some cron jobs for file management and maybe an occasional reboot to insure it continues to run ok.

pageauc, 2015-12-18:
Looks like image library or font might be at issue. Will investigate
Thanks
Default settings are for normal daylight so twighlight to night happens at
twilight setting of 90 but if day mode is with lower light level (not
bright. Outside light) then try a twilight setting of maybe 35 or so until
camera transitions to night. If cam is inside just tun night off

pageauc, 2015-12-20:
Cory

I have setup a fresh copy of latest Raspbian Jessie Full version (Not Lite)
and did a normal raspi-config and did not do a sudo apt-get update or
upgrade

Did a fresh install of pi-timolo per these instructions that are on the
github repo

cd ~
mkdir pi-timolo
cd ./pi-timolo
wget https://raw.github.com/pageauc/pi-timolo/master/pi-timolo.tar
tar -pxvf pi-timolo.tar
./setup-timolo.sh
./pi-timolo.py

exited out using ctrl-c
I was in my computer room and therefore had to change a few settings

edited the config.py
nano config.py
and changed the following variables
twilightThreshold = 35
motionQuickTLOn = False
sensitivity = 500

These changes are not absolutely necessary but did this for the test
twilightTreshhold has to be set lower for inside lighting.
motionQuickTLOn turns off taking a series of images continuously for 10
seconds when motion is found.  Turning this off will take only one motion
image.
sensitivity is increased to more pixels.  If you want to trigger on few
pixels then set this lower.

I ran this and had NO problems at all.  I would recommend you retry the
setup.  I have not tried with Jessie Lite but will do that when I get a
chance.  Was having issues setting up wifi adapter using Lite since there
is not GUI desktop and was having problems doing a manual setup for some
reason.

Anyway this worked great on RPI B2 but should also work with other systems
You might try a fresh format and install of Jessie on the problem SD card.

Let me know if you resolve your problem using these instructions

Claude ....

303CoreyM, 2015-12-22:
Claude -

Sorry for the delay. I was able to try a fresh download of Jessie Lite (don't need or want all the extra stuff) and Pi-timolo. It ended up doing the same exact thing for me. It looks like like Jessie-Lite may be missing some of the required files associated with date/time stamp text. This isn't a requirement for my project, although it would be nice.I am going to move forward with this current setup.

Regarding wifi setup on Jessie Lite, all I do to set it up was to add the ssid & password to the wpa_supplicant.conf file. I found that info here: https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

Thanks again for providing a great tool and helping me out.

Corey

pageauc, 2015-12-22:
Corey

I will do some testing with Jessie Lite and pi-timolo.  Yes I believe Lite
probably is missing some libraries that are not installed but required for
python-image.  I will see if I can find a work around.  May just need to
add some extra apt-get commands to setup.  This is may be a similar problem
like when mencoder code was replaced by avconv.

Will let you know how I do
Have a happy holiday season.  Claude
https://youtu.be/nAqMf9I55iI

closed 2016-04-07 by pageauc

klec00, 2016-04-17:
On Jessie Lite, just run 
`sudo apt-get install fonts-freefont-ttf`
to install the missing fonts.

pageauc, 2016-04-17:
Thanks

I have added this to the speed-track docs but will also add this to the
pi-timolo setup script.

Claude ...