pageauc/pi-timolo #11: Running really slow and does not really work. 6 minutes 2 pictures in motion dir and nothing in timelapse , is this expected?

sgon00, 2016-01-16:
Hi, I never use pi-timolo before, thus I don't know what the expected behavior is. I am running the latest fresh pi-timolo build in raspberry pi 2. It's running really slow. The cpu is not busy at all. I have no idea what pi-timolo was doing. 
It took ***1*** minute to see "Entering Motion Detect", another ***3*** minutes to create `./motion/mo-cam1-1001.jpg` and another ***2*** minutes to create `./motion/mo-cam1-1002.jpg`. 
There is **nothing** in the directory `timelapse`. 
I killed it after ***6*** minutes running. I have no idea how this will work. I keep moving my body in the front of the camera during these 6 minutes. 
I did test built-in commands `raspistill -o test01.jpg -w 1920 -h 1080` and `raspivid -t 20000 -o test01.h264 -w 1920 -h 1080`. these commands are working fine. 

PS: I ran this program once before. thus there is no `mo-cam1-1000.jpg` log in this time.

I read the issue **"Takes 1 motion image then stops"** comments, so I did modify `config.py` accordingly, but no luck. 
Please check my complete `config.py` file at http://pastebin.ubuntu.com/14523004/
Please check the following outputs: 

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

 $ date
Sat 16 Jan 22:33:48 CST 2016
pi@pi2:~/pi-timolo $ ./pi-timolo.py
------------------------------ Loading Python Libraries --------------------------------------

Note: To Send Full Output to File Use command -   python -u ./pi-timolo.py | tee -a log.txt
      Set logDataToFile=True to Send checkIfDay Data to File pi-timolo.py.log

pi-timolo.py ver 2.8
-------------------------------------- Settings ----------------------------------------------
Config File .. Title=pitimolo default configuration motion and timelapse 1080p images
               config-template filename=pitimolo-default-config
Image Info ... Size=800x600   Prefix=cam1-   VFlip=False   HFlip=False   Preview=False
    Low Light. twilightThreshold=35  nightMaxShut=5.800 sec  nightMaxISO=800   nightSleepSec=10 sec
    No Shots . noNightShots=False   noDayShots=False
    No Text .. showDateOnImage=False  Text on Image Disabled
Motion ....... On=True  Prefix=mo-  threshold=10(How Much)  sensitivity=500(How Many)  forceTimer=60 min(If No Motion)
               motionPath=/home/pi/pi-timolo/motion
    Num Seq .. motionNumOn=True  current=1001   numStart=1000   numMax=500   numRecycle=True
               motionNumPath=/home/pi/pi-timolo/mo-pi-timolo.dat
    Quick TL . motionQuickTLOn=False  Quick Time Lapse Disabled
    Video .... motionVideoOn=False  Motion Video Disabled
Time Lapse ... On=True  Prefix=tl-   Timer=300 sec   timeLapseExit=0 sec (0=Continuous)
               timelapsePath=/home/pi/pi-timolo/timelapse
    Num Seq .. On=True  current=1000   numStart=1000   numMax=1000   numRecycle=True
               numPath=/home/pi/pi-timolo/tl-pi-timolo.dat
gdrive Sync .. On=True  Path=/home/pi/pi-timolo/pi-timolo.sync  Note: syncs for motion images only.
Logging ...... verbose=True (Details to Console)    logDataToFile=True  logfile=/home/pi/pi-timolo/pi-timolo.log
------------------------------------ Log Activity --------------------------------------------
20160116_22:34:48 Main - Entering Motion Detect - Time Lapse Loop  Please Wait ...

20160116_22:34:48 ..
20160116_22:36:33 checkForMotion - Found Motion - threshold=10 sensitivity=500 Exceeded ...

20160116_22:36:35   getNightCamSettings - dayPixAve=19 ratio=0.457 ISO=365 shut=2651428 2.651 sec
20160116_22:37:18   takeNightImage - Size=800x600 dayPixAve=19 ISO=365 shut=2.651 sec /home/pi/pi-timolo/motion/mo-cam1-1001.jpg
20160116_22:37:18   postImageProcessing - Next Counter=1002 /home/pi/pi-timolo/mo-pi-timolo.dat

20160116_22:37:18 ..
20160116_22:39:03 checkForMotion - Found Motion - threshold=10 sensitivity=500 Exceeded ...

20160116_22:39:05   getNightCamSettings - dayPixAve=20 ratio=0.429 ISO=342 shut=2485714 2.486 sec
20160116_22:39:48   takeNightImage - Size=800x600 dayPixAve=20 ISO=342 shut=2.486 sec /home/pi/pi-timolo/motion/mo-cam1-1002.jpg
20160116_22:39:48   postImageProcessing - Next Counter=1003 /home/pi/pi-timolo/mo-pi-timolo.dat

<I killed the program here>
```

Thank you very much.

pageauc, 2016-01-16:
The camera is still using night settings  with long exposure.  The camera does not have enough light for taking a day light shot.  Try putting camera in an area with better light.  The PixAve is 19 or 20 so it is below the twilightThreshold of 35 therefore it is using a twilight setting to take the image using a long exposure.  This is the correct behavior.   Try pointing the camera outside during daylight conditions and test motion.   Motion during night shots will be very slow due to long exposure.
Hope this helps.
Claude ...

sgon00, 2016-01-17:
Dear Claude, thanks a lot for the quick reply. Day or night settings look too complex to me. I did this test with enough light (from my point of view) at night. How can I make it work at both day and night 24 hours with quick response? How can I make it work just like `"raspivid"` command? `"raspivid"` can record videos with 1920x1080 in the same condition (night with light) without any problems. The results of `"raspivid"` is OK to me. Thanks a lot.

sgon00, 2016-01-17:
PS: when there is no visible light, I will also use Infrared light. Thus I want pi-timolo work 24 hours with quick response. Is that possible? thanks a lot.

sgon00, 2016-01-17:
PS2:
I did another test with `twilightThreshold=0`. The pictures created in `./motion` directory a bit faster than before. It can create pictures btw 10s, 18s and 4s randomly. BUT 10 seconds per image, or 18 seconds per image, or 4 seconds per image are all not acceptable by me. I want a at least 25fps video. (maybe even less 15fps?). I don't know how `raspivid` works. It creates a very smooth video on 1920x1080 resolution. My current `config.py` is using 800x600 only. Btw, I am not a native English speaker. I am not sure what `timelapse` directory does. There is no images in this directory. 

If pi-timolo will not be able to create at least 15fps or even 25fps video, I think this tool is useless. I never see a CCTV only creates a picture within 10 seconds. How will that work for monitoring and surveillance?  

Maybe I will try the package `motion` and use `on_event_start raspivid ....` in the config file `/etc/motion/motion.conf` just like what other people did in this post: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=45574

Package `motion` will just be a workaround if pi-timolo doesn't. I really want pi-timolo works. 
Thank you very much for your time and hard work. I am looking forward to your solution.

pageauc, 2016-01-17:
pi-timolo does have a video mode and quick timelapse mode as well for
motion detection.  You will still have issues with low light conditions and
detecting motion at night using the camera.  In dark conditions a sensor
works best.
Two packages that I have also used are
http://elinux.org/RPi-Cam-Web-Interface and motionPie that is run from an
SD image wiki https://github.com/ccrisan/motionPie/wiki
https://github.com/ccrisan/motionpie
I have used both of these but both will still have low light issues.
Claude ...

sgon00, 2016-01-17:
I haven't tried those two packages yet. 
motionpie looks like an image. I don't want to use other people's image for now. 
Why doesn't the command "raspivid" have low light issue? I don't need too bright pictures/videos. All I want is just the same video result as the outputs of "raspivid". 
Currently, pi-timolo gives too bright images which I don't need. 
I will try to use the package "motion" and its "on_event_start <self script>" config, hopefully, that will give me better result. Thanks a lot.

sroller, 2016-01-17:
You can also try [motion](http://www.lavrsen.dk/foswiki/bin/view/Motion/WebHome), but it needs video4linux and a special driver for the PI camera board. I found it flaky and it doesn't work with high resolution on my PI2. The maximum I could get to operate was 800x600.
But it offers those features you're looking for. Perhaps you have more luck than I did.

sgon00, 2016-01-18:
Dear Steffen, I did many tests before. I used `motion` before trying `pi-timolo`. I tried two motion builds. one is motion-mmal build and another one is the repo's one by loading "modprobe bcm2835-v4l2". Both works with low resolution and low fps (4fps) as you mentioned. That's the reason why I tried pi-timolo after trying `motion`. I thought `pi-timolo` was using official python module, it should be as fast as `raspivid` command. But so far, the result is bad. 
My current workaround as I mentioned earlier, it's using two cameras. One pi camera and one usb camera. The usb camera uses `motion` to detect motion and once the motion is detected, it calls `raspivid` command to output 1920x1080 videos (30fps or 15fps). The reason why I have to use two cameras is because two apps can not access the pi camera at the same time. If I use pi camera in `motion` for the detection, the command `raspivid` will fail because `motion` is occupying the pi camera resource already. This workaround is a bit ugly but works. I can enjoy 30fps/15fps 1920x1080 video outputs with this workaround. So far, I haven't found a way to use one pi camera to achieve the same result.

sroller, 2016-01-18:
Hi,

That sounds very sophisticated. I had a similar idea but I didn't realized
it: Why not using two PIs, one for the motion trigger and the second one
for the actual recording. One would have to send a signal from one system
to the other to start or to stop recording. I had in mind to use Ruby's DRB
distributed objects for it. But one one could even setup a web server and
just call a certain URL with some JSON payload to do this job.
As I said, that was only the idea and I don't have the time to spend on
this type of project. In the end Real Life is more important :-).

sgon00, 2016-01-18:
Hi Steffen, yeah, we can do that with two PIs. It should work. I had that idea before. but since one PI with two cameras work so far. why do we use two PIs? ^_^ yeah, real life is more important. :)

closed 2016-04-07 by pageauc

kn00tcn, 2016-04-17:
i am seeing the same issue at night, the thing is i'm moving a white max brightness phone to trigger motion

i also thought i told the config to save timelapse shots as fast as possible with max 0.25sec shutter for 5 secs, yet it only saved a single image?

pageauc, 2016-04-17:
Running really slow makes me think you are indoors and it is taking night
shots that have long exposure times. As a result motion detection with
night shots can be very slow.  If that is the case then I suggest you lower
the twilightThreshold variable in the config.py file.  The default is about
50 so lower the setting until camera goes into daymode (monitor the verbose
messages and images are not too dark.  You can improve the awb for daymode
issues by increasing the time.sleep(.5) at about line 346 in the def
takeDayImage(filename):
function.  This is hard coded so try increasing value to 1 or 2 seconds.
This will give the camera a chance to calculate a better white balance and
increase brightness of inside images.  Be warned that as this value is
increased, it will slow the camera response for day images.(takes longer).
I have it set for .5 but this also causes images to be sometimes be a
little brighter or darker event with the same lighting conditions.  The
default setting for raspistill is five seconds.

Give that a try and let me know how you make out
Claude ...

kn00tcn, 2016-04-17:
certainly working much faster in the day, but it detects its own exposure changes as motion, guess there's a lot of tweaking to do

pageauc, 2016-04-17:
You can dampen exposure variations by increasing the time.sleep(.5) that is
hard coded in the takeDayImage function.  The only down side is that if
object is moving quickly there is a chance it will be out of frame before
image is taken.  That is why I left it as a short duration even at the
expense of some false positives due to exposure variations.  See below for
other options.

You can increase the sensitivity and threshold variables in the config.py.
Try increasing one at a time.

The sensitivity controls how much a pixel has to change before it is
considered changed. default is 10 which is quite sensitive.  If you
increase sensitivity then it will ignore smaller green shade changes and
exposure variations.

Threshold is the number of changed pixels for motion detection.  You can
increase this somewhat but as the number increases objects farther away
will be ignored.  This is a good way to trigger on moving objects that are
closer.
Your right there are some things to tune but I have tried to make most
variables.

My default settings are for pointing outside full light.

Claude ...

kn00tcn, 2016-04-17:
are all time units in seconds? not every commented line specifies & other tools like raspistill use miliseconds

maybe i'll eventually make a pull request for some documentation improvements & potentially code improvements (such as a few suggested preset values for specific lighting conditions, maybe with example images)

pageauc, 2016-04-17:
I used to provide custom examples of config.py files in a subfolder for
various situations but stopped due to additional maintenance.  You can
still keep copies of config.py files for various situations.  If you look
at the code I convert camera ms times to seconds.  The config.py should
indicate if value is in seconds.