pageauc/pi-timolo #12: pageauc/pi-timolo

dazbaker, 2016-03-08:
hi can some one help me about pageauc/pi-timolo

pageauc, 2016-03-08:
Most of what you need will be in the Readme.md here https://github.com/pageauc/pi-timolo/blob/master/Readme.md
also in the Raspberry Pi forum posts here https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=94467&p=735113#p735113

Let me know if you have a specific question or issue.  The Raspberry Pi is a learning tool as much as it is a hobby or for use in project solution.

Also note there are other more gui solutions out there.  See the Readme.md for links to those as well.

dazbaker, 2016-03-08:
well what i want to do is to run pi-timolo timelapse ervery 10 sec and ive done that but i have a webseit and i want take the lates photo from tim lapse and up load it to my website but i cant fined a way to input the comand

pageauc, 2016-03-08:
It depends on what the web site will allow and the security.  SFTP would be one way to upload if the web site allows it.  Here is a link to a RPI forum post discussing this solution.
https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=112444

The other way would be to simply use the existing GDrive program to push files to a google drive folder with appropriate security (public or private) and then put a URL in the web page to point the the files on the google drive.  That way you don't actually have to host the files on your web site.  BTW.  Uploading a file every 10 seconds can be problematic if the files are large since it may take longer than 10 seconds to transmit the file depending on the band width.  In this case I would suggest making a copy of the current file with a specific file name so the URL can be a fixed name.  This makes it easier than having to deal with a variable name.  This can be done by modifying the pi-timolo code a little.
You would not need pi-timolo to do all this.  A Bash script using raspistill would do the job of taking a single image every 10 seconds with a set name and then uploading.  I wrote one for sending short video files to a google drive using a web cam per link below.  It can be modified to use the RPI camera module or if you are just using images then modify a copy of the code in the post
https://www.raspberrypi.org/forums/viewtopic.php?p=841762#p841762

dazbaker, 2016-03-08:
so ant there a comand that can ony upload the lates photo been taken and
dump on my webserver

pageauc, 2016-03-08:
Please read the information and links provided.  This should be enough for you to develop a bash script or modify the existing code to suit you purposes depending on what security and rights you have on the web server you are trying to upload to.

closed 2016-03-08 by pageauc

dazbaker, 2016-03-08:
ok just one more thing can just copy the lates  photo out of the folder
 and put it in a nother folder

sroller, 2016-03-08:
Dude, this is not a "learn to program" forum. Try this http://www.tldp.org/LDP/Bash-Beginners-Guide/html/

pageauc, 2016-03-08:
Thanks Steffen.  I agree.  That is why I closed the case.

Regards Claude ...