pageauc/pi-timolo #13: Image Rotation

dcrand2012, 2016-03-30:
Hi Claude- This is what I have been looking for however the case and camera mounting I have requires the image to be rotated 270 degrees is there a way to do this?

dcrand2012, 2016-03-30:
Nevermind when I made imageRotation Setting I spelled Rotation wrong Senior moments get you evry time. Thanks for a great application.

pageauc, 2016-03-30:
I have commented out the camera rotation in the function takeDayImage(filename):
You can un comment the  camera.rotation line the comment out the vflip and hflip lines so the rotation is not overridden.  You will need to add a variable to config.py eg
cameraRotate = 270

   with picamera.PiCamera() as camera:
        camera.resolution = (imageWidth, imageHeight) 
        # camera.rotation = cameraRotate #Note use imageVFlip and imageHFlip variables
        time.sleep(0.5)   # sleep for a little while so camera can get adjustments
        if imagePreview:
            camera.start_preview()
        camera.vflip = imageVFlip
        camera.hflip = imageHFlip

Let me know if you have any issues.

pageauc, 2016-03-30:
Also please note you will need to add similar logic to takeNightImage
Just add the line 
camera.rotation = cameraRotate

dcrand2012, 2016-03-30:
Thanks for the fast reply It's working as I wanted.

closed 2016-03-30 by dcrand2012