pageauc/pi-timolo #14: Settings in the config.py

hulkhaugen, 2016-04-06:
This is less of an issue than a clarification. I noticed several values in the config.py were different than their "default" value according to the comments. Like resolution is set to 1280x720, but according to the comments, default is 1920x1080. There was also the shuttertime(?) that should be no more than 5 according to the comments, but was set to 5.8 by default.

But my main question would be about the timer for the timelapse setting, which is by default set to 5*60, which i assume means 5 x 60 sec = 5 minutes. What if i wanted it at 30 sec? Would that be 0.5*60 or 1*30? Maybe both would work? I dunno, but the comments could very well give an example.

If i may be very picky, the installation guide could use a little tidy-up, especially the 'Setup init.d to run pi-timolo on boot' part was confusing to me. I should probably add that English is my 2nd language, and Linux is pretty new to me. I don't wan't to sound like a brat, so i hope you read this as constructive criticism and not whining. And thanks for these great scripts, i'm about to start a 24hr timelapse of my view :)

pageauc, 2016-04-07:
Hi

No offence taken.  I am not a developer, just a retired guy who does
projects for himself and shares.  The settings as they say in the Pirates
of the Caribbean is just a guide line.  I used my Remote Texas camera
config in the last update.  It is set for 720p since images (max 1000) are
sync'd with my google drive every 15 minutes and I wanted to reduce the
transmission traffic.  On the other hand I have my other Ontario time lapse
setup  that records the winter while I am away See
https://youtu.be/pFkt0ZOrPzg.  I have it set to 1080p since images are
being saved to a hard disk that are later made into a video so I want a
better resolution.

I agree there may be some room for comment improvement to make things more
consistent.  I made the variables a calculation since it was easier to let
the program do the math.  You can certainly make the timer a straight
integer in seconds.  The camera actually uses milliseconds but I wanted to
use more human time values in seconds. Just set the value to 30 if you want
30 seconds or 300 for 5 minutes etc  As far as the shuttertime this is also
a variable that can be customized.  It represents the maximum exposure time
for night images.  During twilight this value varies depending on the
variable outside lighting conditions. The twilight threshold determines at
what light level the twilight ramp up or down starts There is also and ISO
setting.  The maximum depends on the night lighting conditions.  If you
have bright street lights or inside lights then the value can be lower so
images are less likely to be over exposed.  Please note there is also a
default 10 seconds wait for the camera to calculate the white balance after
which the camera takes the long exposure of the specified time period up to
the max.  If it is normally darker (inside or outside) then the max needs
to be higher so the image is brighter

If I get some time I will make the changes.  I have a remote camera in
Texas that I can view motion detection images uploaded to my google drive.
Before I left I implemented a way to remotely change the camera settings by
putting a config.py file on my google drive.  A crontab bash script then
checks for a new file and then updates the pi-timolo config.py and reboots
the camera.  I may implement a feature where the program will check for an
updated config.py and reload the values without having to restart the
program.

The init.d documentation is pretty basic because there are various other
better explanations on the web.  Just google if you need more detail.  It
is always difficult determining what skill level the user is at.  I figure
if the information is already out there then basic instructions should be
adequate.
Also things change  Raspbian now runs systemd under Jessie and some things
are compatible but others are not.  I had to change the makemovie to
accommodate removal of support for mencoder and a change to avconv.  Just
an example of things that work one day and don't work the next.  Also I had
to change the program for syning with google drive since the google changed
their API and the original program I used was no longer being supported by
the original developer.  I actually prefer the new gdrive interface and I
have passed on suggestions to the developer of that code.

Anyway thanks for your email.  It always amazes me how something you do as
a personal project for your own needs becomes useful to others.
It is also interesting to find out how other users use the program and
modify it for other purposes.  By the way in my Readme.md I do mention
other GUI motion/timelapse programs that I use and why I still prefer
pi-timolo in a lot of cases.  Like Eben I think the RPI is a good
educational/development tool and if you just compile all code or create a
custom image then it may be easier for a casual user to use but there may
not be much of a learning experience.

Regards
Claude ...

hulkhaugen, 2016-04-07:
Thank you so much for your answers! My Pi is currently taking 60s timelapse photos, but i had trouble with my WiFi dongle and there's no monitor connected. Looking forward to checking out the content of the SD card this evening.

pageauc, 2016-04-07:
If you post the results on Youtube or someplace else public then send me
the link.  I had a person in New Zealand update a public dropbox share with
daily timelapse progress of his house construction.  I thought it was a
pretty good way to keep tabs on things and record for future reference.

sroller, 2016-04-07:
Claude,

did your RPi ran through the entire Winter without reboot? I'm having
problems with mine (Pi classic and Pi2) when on Wifi. Every 3 days or so I
loose the network and have to pull the plug.

hulkhaugen, 2016-04-07:
So here it is, my first attempt. Pretty happy with it, but I will continue tweaking it :) https://www.youtube.com/watch?v=8UbHJahR5-g

pageauc, 2016-04-07:
I also posted the comment below on your video. What fps did you encode the
video at.
Wow. This is gorgeous. You should think about adding some music. You can do
this from within YouTube video manager. Edit the video and select the Audio
icon at the top (should be third one) You can then select a google audio
track that you like for the video. Claude ...

hulkhaugen, 2016-04-07:
I put all my relevant settings in the comments. I took a photo each 60s, and used 10 fps. Next time, i want to use 30 sec between photos and maybe 30 fps. I want it to be a bit faster and smoother. I want a day closer to summer when it's never pitch black, more green colors, many boats on the sea and hopefully get a full moon rising. I wish the lens had a slight wider angle.

I thought about music, but it's hard whith all them copyrighters.

pageauc, 2016-04-07:
I added a comment to the video with a step by step to add google royalty
free music duplicated below. The free music does not put adverts on the
video.  You can do this all within youtube so no need to re upload.  google
does everything on their side.  Give it a try and let me know how you make
out.

Log into youtube
1 - Go to YouTube Creator Studio by clicking on your icon in the right
corner. 2 - Select VIDEO MANAGER from left side menu 3 - Select Edit for
the video 4 - Select the Audio icon at the top (should be third one from
left) 5 - Select a google audio track that you like for the video. There
are different genre's available from pulldown menu . 6 - Save selection and
YouTube will then add the selected audio track to the video.

  Claude ...

hulkhaugen, 2016-04-28:
Hi! So i've been playing some more, and there is one issue i've got. I'm currently running a headless and no internet connected setup. When done, i just bring the keyboard and type sudo shutdown now. It seems that a pretty large portion of the latest photos doesn't get saved. I'm waiting for pitch black outside, but the latest photos are from the afternoon. Are they only stored in the memory and not the SD card?

Also, couldn't help myself so i ordered the Pi3 and the new 8MP camera yesterday. Do you think that will work straight away, just applying the new ISO values and resolution? I'm also wondering if the 1920x10890 res is cutting anything of the photo, the width and/or the height?

pageauc, 2016-04-28:
Greetings

Sun is just rising here in Canada.

[image: Inline image 3]

Not sure if picamera module needs to be updated for the new 8mp camera.
There is an issue about picamera V2 on github here
https://github.com/waveform80/picamera/issues/278 I expect there will be a
picamera module update if needed.  In the mean time the existing code
should work except for possibly some new features.

If you want to safely shutdown the pi when it is headless then I put jumper
wires on gpio pins 5 and 6 with or without momentary switch and use a
shutdown script attached.  Here is an article with additional detail.
https://www.element14.com/community/docs/DOC-78055/l/adding-a-shutdown-button-to-the-raspberry-pi-b
This will safely shutdown the pi and avoid any corruption of files.  This
works great for me and I just use jumper wires and short them out when
needed.  Just make sure there are male and female ends to avoid accidental
shorting of the wires.  I also have some with a small momentary switch but
the plain jumper wires works OK.  You can also reboot the pi after it is
halted if you use pins 5 and 6 (Does not work on other pins for reboot but
can shutdown OK using other gpio pins)

[image: Inline image 1][image: Inline image 2]

Images are taken in memory by the camera module and saved to the SD card.
I have several cameras with hard drives attached.  One is setup in a
plastic case with the adapters for the PI and hard drive inside the case.
I use the camera hdmi extension for this one.
https://www.tindie.com/products/freto/pi-camera-hdmi-cable-extension/
This works good and the camera can be up to 15 ft away if you want to mount
the camera head away from the rpi.  You just need to make a protective case
for the picamera end of the cable to protect the camera from the elements
if it is outside.

Also I saw that you added music to your first YouTube time lapse video.
Nice ...

Claude ...

-- 
See my YouTube Channel at http://www.youtube.com/user/pageaucp

# This entry should be added to the sudo crontab -e.
# This will launch the shutdown script.
# when a momentary switch is between GPIO pins 5 and 6
# After initial power boot press switch to safe auto shutdown
# After a few seconds press switch again to boot.
# repeat as desired.
# make sure shutdown.py is executable
# chmod +x shutdown.py

# add line below to sudo crontab -e

@reboot /home/pi/shutdown/shutdown.py

#!/usr/bin/env python

# Safely shutdown Raspberry pi using a momentary switch
# Connect momentary switch to pin 5 and 6 (default)
# or set gpio_pin variable to desired pin below
#
# Set button_hold variable to number of seconds to
# hold down momentary switch before action taken
# default is 2 seconds.  0=No delay
#
# Add line below to sudo crontab -e (without #)
# @reboot /home/pi/shutdown/shutdown.py
#
# make sure shutdown.py is executable
# chmod +x shutdown.py
#
# After initial power boot press switch for specfied time
# to Initiate safe shutown (halt)
# After shutdown wait 5 seconds then press switch again
# for specified time to intiate a startup
# Wait a few seconds between between operations
# 

import RPi.GPIO as GPIO
import time
import subprocess

button_hold = 2       # number of seconds to hold button before action taken
gpio_pin = 5   # Set GPIO pin you want to use.  default is 5 and ground is 6

GPIO.setmode(GPIO.BOARD)
# set GPIO pin to input, and enable the internal pull-up resistor
GPIO.setup(gpio_pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)

start_time  = time.time()
shutdown = False
while not shutdown:
    if GPIO.input(gpio_pin):
        start_time = time.time()
    else:
        hold_time = time.time() - start_time
        if hold_time > button_hold:
            shutdown = True
    time.sleep(.1)     
subprocess.call("/sbin/shutdown -h now", shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)

hulkhaugen, 2016-04-28:
Thanks for your feedback. However, it seems that the python script only assigns the same command to a GPIO input as i type manually. I can't see how that should be different:
`os.system("sudo shutdown -h now")`
I guess the error could be related to the memory card. 8GB doesn't last long with a photo each 20 sec, and I guess some of that storage is reserved to the OS and it's various functions. I've got a 32GB coming in the mail :)

But what i'm more interested in, is the resolution. Lets use the 5MP camera as an example. It has a resolution of 2592x1944, so if i take a 1920x1080 picture, does it just pick that from the center of the sensor and leave out the rest, or does it use the full width of the sensor? In other words, does it take a 2592x1458 photo and resize it to 1920x1080, or does it leave out the remaining 612x864? I want to use full width of the camera sensor.

I suspect that it will only take from the center. If that's the case, lets say i change the resolution of the stills to 2592x1458 for maximum resolution and FOV 16:9 photo. Then i use the video converter to convert it to 1920x1080 videofile. Would that be optimal for the maximum FOV and FullHD video file? Would the converter resize or crop the images?

pageauc, 2016-04-28:
Here is a good explanation of the 5mp camera native resolutions
http://picamera.readthedocs.io/en/release-1.10/fov.html .  If you want to
check the remaining space on the SD card use

DF -h

This will show how much space is free and used.

I used to take full resolution images 2592x1458 and convert them to 1080p.
It will still crop, stretch or resize according to what you specify.
Normally it will crop whatever will not fit in the resized image so you may
loose something.

sroller, 2016-04-28:
If you change the resolution when you take pictures then you'll lose around the outer edges. To prevent that take the full picture and use the ImageMagick[http://www.imagemagick.org] library tool `convert` to resize the picture to your liking.

pageauc, 2016-07-22:
Resolved issues

closed 2016-07-22 by pageauc