pageauc/pi-timolo #15: adding pi-timolo.sh to init.d causes RPI to restart every few minutes

byramsewell, 2016-04-07:
I've been trying to get pi-timolo to autostart at boot so I've been following the instructions. When RPI first boots I can run ps -ef | grep pi-timolo and I see the python routine running but it dies after a few minutes and does not capture any motion. I completely re-imaged my SD card and started from scratch. All was well until I went through the instruction again to add pi-timolo.sh to /etc/init.d. Once I do this then my RPI is stuck in a reboot itself mode. I have attempted to remove the entry form init.d by running 
sudo update-rc.d -f pi-timolo.sh remove

This apparently stops the pi-timolo.py service from running but it does not stop my RPI from restarting every few minutes. 
The only way I know to fix the problem is to re-image the SD card, completely start from scratch and AVOID attempting to autostart pi-timolo.py

Thanks

pageauc, 2016-04-07:
If you look at sync.sh it has a routine that monitors if pi-timolo is
running and if it is not found then it will reboot the pi to try to
reload.  I guess this is a bug/feature that I put in since I have a remote
camera that I wanted to auto boot if pi-timolo was not running for some
reason.  You can comment out the reboot code in sync.sh to stop this
behaviour.  I am assuming you have sync.sh running in a sudo crontab -e.

It would appear pi-timolo is failing for some reason.  I would suggest you
run it manually from an ssh session and monitor to see what is causing the
problem.  python will give error tracking info.  I have run several of my
setups for years without issue but you may have a specific issue that is
causing python to have a run time error.  I have not built very much error
trapping code into the program.

Run the program manually to see if you can trap the error and send me the
python error messages.

Claude ...

pageauc, 2016-04-07:
Also you might try deleting the .dat files if you are not running the
lastest 2.91 version.  I ran across the case where a dat file was corrupt
and could not be converted to an integer.  The new pi-timolo code has error
trapping code to recover from this error.  This can sometimes happen if the
pi is rebooted just during the dat file update and before closing the dat
file.
Claude ..

byramsewell, 2016-04-07:
I tried running update-rc.d -f pi-timolo.py several times and it appears it finally removed it from init.d and now the pi is no longer rebooting.

pageauc, 2016-04-07:
The init.d script should be run using the pi-timolo.sh bash script.  This
shell script is in the init.d format.  Do Not run init.d against
pi-timolo.py since it is not compatible with the init.d format.  Please
read the instructions again.

You may need to edit the shell script variables if installation location is
other than pi-timolo folder.  Note the init.d shell script pi-timolo.sh
will run the pi-timolo.py script as sudo user.  Make sure the pi-timolo.py
is executable as well as the pi-timolo.sh script.  If you untarred the
github file per instructions then the permissions should be executable.

Claude ...

byramsewell, 2016-04-07:
Yes I am running sync.sh but I'm also running the latest version. I commented out the reboot lines.
it ran fine manually as user pi all night but seemed to hang this morning around 8:13. It was still running just no longer doing anything. I killed it, remove mo-pi-timolo.dat and all the images. It would not get past the first log activity line, no log file was produced. I restarted the RPI and again it works fine manually. The problem is that when I run it as root, I get a permission denied error. That perhaps in conjunction with the reboot code in sync.sh is my problem.
[Error.txt](https://github.com/pageauc/pi-timolo/files/208517/Error.txt)

byramsewell, 2016-04-07:
I have an nfs mount to a NAS device. the motion folder inside ./pi-timolo is a symbolic link to my nfs share. this setup works fine running pi-timolo as user pi but for some reason does not work as root. I will investigate further.

pageauc, 2016-04-07:
Are you running timelapse, motion or both?  It sounds like just motion from
your description but not sure.  The problem may be with the share across
the network if there is an interruption when files are being written.
Normally I would send the files to the local SD card and run the
mvleavelast.sh script (You would need to edit variables)  This may avoid a
problem with network minor network interuptions.  You would need to put the
script in a crontab to schedule the transfers.  This would move the files
not copy them.

Also it sounds like the the camera is hanging since it does not appear that
python is crashing with an error.  The fact that you have to reboot to get
the camera to work is a good indication the the camera is hanging for some
reason.  Double check connections.  Also You might try changing the
nightMaxShut
variable in the config.py to maybe 5 or 5.5 The exposure time can be a
problem if it goes over 6 seconds.
I have experienced this problem myself. It might be crashing after coming
out of twilight mode and changing to daymode.
Also try changing the twilightThreshold setting if you see overexposed or
underexposed images (very black or very white).
I have one of my camera's pointing outside east into the sunrise and have
it set to 80 but if you are in a lower light situation the
you can set it to 50 or so. You can experiment with this setting to see if
twilight transitions are free of very dark or very light images

Hope this helps
Claude ..

byramsewell, 2016-04-07:
Thank you Claude, This is very helpful. My problem was indeed NFS mount permissions on my Synology NAS. I have resolved that and will go back and try the autostart at boot feature.

byramsewell, 2016-04-07:
Autostart at boot is working now.
However all my images are very dark. I have tried to adjust the twilightThreshold setting. I have set it at 90, 50, 30, and 5. All of these settings seem to be about the same except for 90 which was very dark. The pi-timolo.log file shows the same entries for each shot:
20160407_15:40:50 takeDayImage Size=1280x720 exp=auto awb=auto /home/pi/pi-timolo/motion/mo-cam1-1054.jpg I am attaching both images from the logs I mentioned.

again except for the one shot I took with it at 90. That shows the following log entry:
20160407_15:28:00 takeNightImage Size=1280x720 dayPixAve=92 ISO=100 shut=0.001 sec /home/pi/pi-timolo/motion/mo-cam1-1037.jpg
[image]
[image]

byramsewell, 2016-04-07:
The same lighting using a simple raspistill command gives a normal photo.
[image]

pageauc, 2016-04-07:
When the camera is set for day the awb is auto and exp is auto as per the
log entry.  However to speed up the camera I set the sleep delay to 0.5
sec.  The normal raspistill waits approx 5 seconds so it has a better
chance of calculating the AWB and Exp settings.  You can change this by
editing the pi-timolo.py file for the takeDayImage() function near line 342
and edit approx line 346  time.sleep(0.5)  try a longer time period
starting at maybe 2 seconds eg time.sleep(2) to see if the awb improves.
You can then increase or lower the sleep period to optimize.  My setup is
for mostly outdoors so I keep the sleep to 0.5 seconds since AWB is pretty
fast in bright light but in lower light the AWB takes more time.  That
should improve things.  I should make the sleep a variable.

Let me know how you make out
Claude ....

closed 2016-07-22 by pageauc