pageauc/pi-timolo #19: Stream overflow trying to get higher resolution images

hendrixjl, 2016-05-25:
First: Great software!
I am trying to get higher resolutions than what are the actual defaults config.py (1024 x 768). I'm using that new raspberry pi camera (V2.1).
I'm getting the following errors:

> ++++++++++++++++++++++++++++++++++
> pi-timolo.py - Exiting Program
> +++++++++++++++++++++++++++++++++++
> 
> Traceback (most recent call last):
>   File "pi-timolo.py", line 689, in <module>
>     Main()
>   File "pi-timolo.py", line 603, in Main
>     data1 = getStreamImage(False)
>   File "pi-timolo.py", line 458, in getStreamImage
>     camera.capture(stream, format='rgb')
>   File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1577, in capture
>     encoder.close()
>   File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 893, in close
>     self.stop()
>   File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 881, in stop
>     self._close_output()
>   File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 795, in _close_output
>     output.flush()
>   File "/usr/lib/python2.7/dist-packages/picamera/array.py", line 285, in flush
>     self.array = bytes_to_rgb(self.getvalue(), self.size or self.camera.resolution)
>   File "/usr/lib/python2.7/dist-packages/picamera/array.py", line 174, in bytes_to_rgb
>     'Incorrect buffer length for resolution %dx%d' % (width, height))
> picamera.exc.PiCameraValueError: Incorrect buffer length for resolution 100x75
> 
Are there other parameters I should change to make this work?
Thanks!

pageauc, 2016-05-25:
The pi-camera python library internally compares the camera output buffer
with the specified image size to see if they match.  If the two do not
match then the above error is raised.  In my experience the problem is
usually due to a camera cable or Raspbian firmware issue.  I have not yet
tested things with the new 8mb camera module although I have received one.
It is possible there is an issue between the camera hardware and the
pi-camera library.

You can try reseating the cable and doing a raspbian firmware update that
is documented in the github instructions.

hendrixjl, 2016-05-26:
I did a firmware update and that seems to have done it.
Thanks!

closed 2016-05-26 by hendrixjl