pageauc/pi-timolo #22: Has this been tested with Python 3 ?

git-user44, 2016-10-24:
Struggling to get this code working. I'm starting to wonder if it has been tested with Python 3. I'm trying with Python 3.5.2 and getting all sorts of errors. This is the latest.
  
  Main()
  File "./pi-timolo.py", line 712, in Main
    takeNightImage(filename)
  File "./pi-timolo.py", line 410, in takeNightImage
    logToFile(dataToLog)
  File "./pi-timolo.py", line 131, in logToFile
    f.write(filecontents)
TypeError: a bytes-like object is required, not 'str'

pageauc, 2016-10-27:
Should be a not w+.  Code changed but cannot test at this time.  Let me know if this works
Thanks Claude ...

pageauc, 2016-10-27:
I have edited line 130 so it does not try to write in  binary, just a.  Download a new copy of pi-timolo.py from github 

    wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

or just redo the install script again.  I edited the code directly on github since I am in the middle of a move.   Also I cannot test at this time so would appreciate if you can test and let me know status

Thanks Claude ...

pageauc, 2016-10-27:
> Taking a look at the code You should be able to change line 130 to just
>
> f = open(logFilePath, 'w+')
>
> I am in the middle of moving so cannot test this so If you can do that let
> me know if this resolves issue
>
> Claude ...

I have change code line to

f = open(logFilePath, 'a')

Can you please test this change by downloading latest pi-timolo.py off
github.
Thanks

pageauc, 2016-10-27:
Taking a look at the code You should be able to change line 130 to just

f = open(logFilePath, 'w+')

I am in the middle of moving so cannot test this so If you can do that let
me know if this resolves issue

Claude ...

pageauc, 2016-10-27:
pi-timolo code has has never been tested by myself under python3.

The bytes type was introduced in Python 3 and that is why the code works in
Python 2. In Python 2 there was no data type for bytes: See link below for
more details

http://stackoverflow.com/questions/33054527/python-3-5-typeerror-a-bytes-like-object-is-required-not-str

git-user44, 2016-10-27:
The latest code runs, but I think there are other problems. I'm not sure that pyexiv2 exists anymore, or if it does, that it is compatible with Python3.
Thanks for your help. This is the first Python code I have ever looked at and and thanks for confirming that it's not been tested with Python3. I thought I must be doing something stupid, or my environment was wrong.

pageauc, 2016-10-28:
I ran across the pyexiv2 issue with python3.  There is supposed to be a
solution here http://www.py3exiv2.tuxfamily.org/downloads but can't seem to
get it to work.  pi-timolo works good with python2 but will see if I can
resolve the dependencies with python3 so it will at least work.  I will see
if one version can be used for both python2 and 3.  In the mean time I
suggest just use python2 version.

Claude ..

closed 2016-12-16 by pageauc