pageauc/pi-timolo #23: Pi NoIR camera

cartercherry, 2016-10-29:
Despite extensive fussing over the sensitivity, threshold, and motionAverage settings of config.py of pi-timolo, I experience many false positive motion events with the NoIR pi camera.  The regular pi camera works well.

  Have you or others found the pi NoIR camera to work acceptably with pi-timolo?  If so I will continue to try to tune the variables in config.py 

Great application!  
Thanks,  
Carter, a loyal follower of your project

pageauc, 2016-10-29:
I have not tested with noir although I bought one a while back.  Just have not had time to try it out.  The motion algorithm for pi-timolo is very basic and can give false readings of true motion since it just looks at changed pixels using a pixel level and number of changed pixels counts.  
OpenCV can do a much better job by using a tracking method (see my other motion projects) so small objects or stationary object motions can be excluded.  Better filtering of data would be possible to zero in on only specific types of objects by contour sq px area size range, ratio of contour Length and Width, direction of motion, specific location on image that motion takes place, Etc,  This would allow specifying objects within a size range, L/W ratio (Eg shapes like people, vehicle, animal rectangular shapes and sizes and avoid things like trees or bushes that basically move back and forth in one spot.
If you look at some of the other projects I have done using motion tracking you and data filtering you can see what I mean.
I have been thinking of making a new updated version of pi-timolo that would add a feature to use OpenCV and motion tracking.  Not sure how well it would do for regular camera night conditions and other things, but I believe it would allow much better control for filtering data to trigger on what user wants to see.

Let me know what you think.
Claude ...

cartercherry, 2016-10-29:
Claude,

I am very interested in experimenting with OpenCV.  I will delve into your
other github projects.  The long exposure times at night make the security
aspect of the timolo project difficult (as you pointed out) - but great to
experiment with.  I hope there will be support for the newer  8 megapixel
pi camera in forthcoming projects.

Thank you for sharing your knowledge and time. I've learned a great deal
studying your source code.  You are very thoughtful.

Cordially,
Carter

davidhaile, 2016-11-16:
I'd also be interested in an OpenCV motion detector. Someday, someday, someday I'll also contribute. 

I have it working with RPi 3 and Cam 2. I ran it a few weeks ago, but this evening I put it out on the edge of the yard where it has a better view of the mountains. A snowstorm will arrive by morning. The camera is encased in plastic-wrap. Seems OK so far. I'll check it after it starts snowing.

My sister-in-law lives in San Jose, CA and has had numerous break-ins and last week a car stolen out of her driveway. She lives in a million-dollar neighborhood. She'll be interested in a good motion detector and timelapse camera. I may have to fly out to setup her network to support it, though. She's not technically literate.

pageauc, 2017-05-09:
I now have a noir camera.  pi-timolo has been upgraded to version 6.0 with support for sub-folders, recent images folder,  disk management and motion detect using threaded video stream.  I also have a version of pi-timolo that uses my speed camera code logic.  I have tested this and it works very well for eliminating motion false positives since you can set tracking variables.  The normal pi-timolo.py does not use opencv so I am thinking of releasing pi-timolost  PI - TImelapse,  MOtion, LOlight, Speed Tracking,

Still need to integrate the new ver 6.0 pi-timolo.py logic into pi-timolost.py.   I will also see what settings work best with the noir and it may just need a separate config.py default settings file.
Regards Claude ...

davidhaile, 2017-05-10:
I will duplicate your setup when I get back home and run a few tests.

My goal is to track trail usage for cross country ski trails that I help
manage in the winter. I've been experimenting with PIR sensors (75% false
positives, can't be tuned), ultrasonic (complete failure  - misses all soft
fuzzy targets), and IR laser (too directional - can't track coming & going
without missing part of the trail). I'll get back into CV to solve the
problem. The negative point is high power requirements compared to the
others. This must be a 100% battery-powered installation. I can visit every
two weeks but not every two days. It will shutdown and startup with the
daylight.

Thank you for your efforts.

pageauc, 2017-05-10:
Have you looked at blink  https://www.theverge.com/circuitbreaker/2016/9/29/13104268/Blink-budget-security-camera.  You would need a hub put away in the ground or somewhere safe and run off a larger battery .  Not sure how secure things would get if these things were not secured or hidden well.  I would assume they could be put in a waterproof case if need be.  
https://www.amazon.com/Blink-Home-Security-Battery-Powered-Smartphone/dp/B0172DDZ5E/ref=sr_1_4?s=electronics&ie=UTF8&qid=1494411276&sr=1-4&keywords=blink+home+security+camera+system

pageauc, 2025-02-23:
See NEW pi-timolo2 libcamera picamera2 at https://github.com/pageauc/pi-timolo2

Not tested with noir camera yet

closed 2025-02-23 by pageauc