pageauc/pi-timolo #24: Crontab example in the wiki may not be correct.

kaushik-d, 2016-12-15:
Crontab did not work for me following the example in the wiki.
After some searching around online I tried this and it worked. 

*/5 * * * * cd /home/pi/pi-timolo && /home/pi/pi-timolo/sync.sh

I suggest that we check and correct the example if necessary.

pageauc, 2016-12-15:
Thanks for finding this bug.  The problem is that sync.sh was failing when checking for  a matching folder on google drive due to the crontab sync.sh not being in the home folder. 
I had made recent changes to sync.sh and it ran from a console session but as you found it failed in a crontab.  I have added a cd $DIR so now the script runs under a crontab since the folder check succeeds.  I have updated sync.sh and  the wiki to add basic trouble shooting  by sending output to a log file.  That is how I found the issue.
You can download the updated sync.sh here

    wget -O sync.sh https://raw.github.com/pageauc/pi-timolo/master/source/sync.sh

Give it a try with the default crontab entry and let me know.  

I will review the logic of sync.sh to see if I can improve on operation

Let me know how you do
Thanks Claude ...

pageauc, 2016-12-16:
Ver 1.96 Release
I have restructured sync.sh and improved messages.  Tested this on local console and from crontab.  Should work better and display appropriate messages.

New version can be downloaded from ssh or console session per commands below

    cd ~/pi-timolo
    wget -O sync.sh https://raw.github.com/pageauc/pi-timolo/master/source/sync.sh
   chmod +x sync.sh

Please raise a new GitHub issue if required
Claude

closed 2016-12-16 by pageauc