pageauc/pi-timolo #26: webserver.py not starting?

Monongahela, 2016-12-21:
Hello Claude, I'm having an issue getting webserver.py to start and then continue running on the RPi 3. I've installed your latest release from December 14. Running ./webserver.py on the CLI reports "webserver.py is Running" and returns the PID. But netstat -tln can not find a server listening on port 8080 (but does find SSH on port 22), and pgrep can not find the PID that was returned by the webserver.py script. Attempts to reach the RPi at port 8080 from a browser on any other machine on my local network are met with the message "Connection Refused".

Any ideas on the cause, or recommendations to fix this?

Thank you,
Mono

pageauc, 2016-12-21:
You need to check that ./webserver.py runs on it's own without failing
otherwise webserver.sh will not show any errors if webserver.py fails.  I
suggest you run

./webserver.py

From you description it looks like you have done this.

See if there are any error messages webserver.py console messages.  Note
you cannot ctrl-c or end the terminal session when this is running since it
will stop the webserver port from opening.  The webserver should be
accessible on a web browser via the ip and specified port and web activity
will be shown in the console or terminal session that is running
pi-timolo.py.  It appears that nothing else is using port 8080.  You could
always try another port eg 8081, 8090 or similar

Open a second terminal session and type

pgrep -f webserver,py

to check if the webserver.py started OK.  A pid number will be returned if
it is running OK. pgrep -f parameter looks at the complete command since
the primary command is python and webserver.py is a parameter.  You can
always do a

ps -ef

to see all processes and look for python webserver.py

If webserver.py fails then review the error message, correct the problem
and try again.  Usually it would be a misconfigured variable setting in
webserver.py.  Use nano to edit webserver.py file variables if that is the
problem.

Make sure you are not running under python3.  I get a ImportError: No
module named 'SocketServer' error if I try to run python3 webserver.py

If ./webserver.py appears to run OK then the problem may be with the
network, router of a windows firewall setting, Etc.  It may be possible
that another machine has the same IP address as the RPI.  This can
sometimes happen if you configure a fixed IP address in the same range as
the router DHCP scope. I have actually had this problem on a network.  I
usually run fring on my ipod to check ip and hosts on my network since I
have some devices with fixed IP's and also the router will sometimes change
a DHCP ip address for one of my RPI's that I am trying to SSH, filezilla,
mobaxterm or VNC to.

Other than a port being blocked (by RPI, Browser, OS firewall, or used by
another process, misconfigured webserver.py variable, permissions or
network issue that is the only things I can think of.  I have run this on
quite a few RPI's with different Raspbian versions and hardware vintages
without problem.

Let me know how you make out.  I am thinking something either the client
OS, or browser is blocking a port.  I would be interested and would
document issue for others.

FYI I run chrome browser on Windows 7 so not sure about other combinations.

Thanks
Claude ...

Monongahela, 2016-12-21:
Thanks for the response Claude. I discovered the problem, and it was a newb mistake. I clobbered the webserver.py file when I installed it. Downloaded a fresh install and now it starts fine. But now I'm seeing a new issue. When I try to connect to the webserver at port 8080 from my client, I get a long list of error messages apparently thrown from python. Here's a snippet of what I'm seeing:

----------------------------------------
Exception happened during processing of request from ('10.0.0.10', 35946)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 655, in __init__
    self.handle()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "/usr/lib/python2.7/SimpleHTTPServer.py", line 44, in do_GET
    f = self.send_head()
  File "/usr/lib/python2.7/SimpleHTTPServer.py", line 83, in send_head
    return self.list_directory(path)
  File "./webserver.py", line 93, in list_directory
    % (list[1], web_page_title))
IndexError: list index out of range
----------------------------------------

Could this be an issue with the version of python I'm running?

Thank you,
Mono

pageauc, 2016-12-22:
Looks like you have No jpg files in the web server root folder or the pi
user cannot access them.  I did a quick test on an empty folder and got the
same error message.  Try putting more than two jpg images in the folder.
Edit the variable web_server_root to point to the folder containing the
images.  Otherwise it should work OK.

See my wiki here https://github.com/pageauc/pi-timolo/wiki/Utilities

Let me know how you make out.  You can have folders in folders and the
webserver will navigate them but create a tab for each new folder.  Just
close the tabs as required since you cannot navigate backwards in the tabs.

Let me know how you make out

Monongahela, 2016-12-22:
Yes, that was it. web_server_root was pointing to "motion" and I was capturing everything in the timelapse folder. I stopped the webserver with webserver.sh stop, edited webserver.py to point to the correct root folder, then started the webserver with webserver.sh start. Then pointed my browser to the ipaddress:8080 and it came right up.

Thanks for your patience.
Mono

pageauc, 2016-12-22:
Please note there is a setting to limit the max number of files to show in
the right pane listing.  This feature works better if sorted by date
descending (default) or by filename if named by number not date.  That way
you can browse the last 100 .. 200 or so images Up to 500 works OK as well,
 Note with a long list you may want to turn off refresh or set to a longer
time period.  Default is 120 seconds.   Also you can setup a second web
server by copying the existing webserver.py and sh to another name eg
webserver-mo.py and sh.  Change the web_server_root and port number
different from the first one.  You can then have two web servers pointing
at two locations if you wish.

Claude ...

Monongahela, 2016-12-22:
Great suggestions, thanks. I'll give them a go.

Mono

pageauc, 2016-12-22:
Just did some testing with other files besides jpg.  If I create a folder
and put txt, html or other similar files in the folder, then you can create
a sub web server.  When you click on the folder the list of files is
displayed. html files only seem to work locally, no access to internet. If
you have an index.html then no right pane is displayed and the index.html
takes up the whole page.  You could use this for looking at txt, html or
other types of files. besides jpg.  I tried putting a Readme.md file in and
it did not recognize it but did pull up download window.  I have not tried
javascript yet.

See wiki here
https://github.com/pageauc/pi-timolo/wiki/Access-images-via-webserver

Claude ...

Monongahela, 2016-12-22:
Great, thank you Claude. Keep up the good work. It will be interesting to see where you can go with this.

Mono

closed 2017-01-04 by pageauc