pageauc/pi-timolo #28: Add sync to DropBox feature

MartinStolle, 2016-12-21:


PhyNerd, 2017-04-05:
This might be a starting point: 
https://github.com/andreafabrizi/Dropbox-Uploader

It allows for recursive directory upload. 
But there sadly is no check for uploading changed files only.
This however could be solved by writing uploaded files with their last change date to a txt file and crosschecking before uploading.
Or alternatively a List could be created where files get added, after being created by timolo, that will then be wiped on the next sync.

pageauc, 2017-04-05:
Jan

I have done quite a bit of work with dropbox uploading using the bash
script and dropbox and other python api/sdk libraries
https://www.dropbox.com/developers/documentation/python .  The big problem
is that you have to setup a dropbox project and this is not intuitive, so
requires detailed instructions.  I also ran across the problem of syncing
changed files and yes you would need to keep track of snapshots of changes
to sync or do like google drive and do a comparison of files to determine
changes prior to sync.  I have not seen too much interest in a dropbox
feature so I have not done pursued this feature.  Wiki has a message about
dropbox
https://github.com/pageauc/pi-timolo/wiki/sync.sh---Automate-gdrive-Uploads-and-More

Let me know your feelings on this.

Claude ...

PhyNerd, 2017-04-05:
I must have missed that passage in the wiki.
Since it seems to be quite a bit of work and gdrive sync works nicely I also didn't look into it much. 
Just wanted to leave the link here so someone that has time and interest has a starting point.
Maybe after tackling the threading of the camera module(which comes with it's own set of little problems) I'll come back to dropbox.

pageauc, 2017-04-05:
Dropbox section is at the bottom of the wiki link on the previous
discussion here  https://github.com/pageauc/pi-timolo/wiki/sync.
sh---Automate-gdrive-Uploads-and-More

marciano1, 2017-11-15:
What about using rsync/scp ? I've seen something called rclone

pageauc, 2017-11-15:
I have done some work with dropbox  using https://github.com/andreafabrizi/Dropbox-Uploader/
It works but integrating it into pi-timolo seemed a bit more work than I wanted to do for a feature I would not use.  I prefer google drive.
If someone wants to develop this feature and give me a github push I would be willing to test and document.  see Alternative DropBox solution https://github.com/pageauc/pi-timolo/wiki/sync.sh---Automate-gdrive-Uploads-and-More#alternative-dropbox-solution

Claude ...

marciano1, 2017-11-15:
I also don't use Dropbox nor GDrive in these cases, I rent a dedicated server for other purposes where I easily can upload and other options rsync and scp have.

closed 2017-11-15 by pageauc