pageauc/pi-timolo #29: Add easier/automated way to mount and redirect files to an external HD or USB dongle

MartinStolle, 2016-12-21:


pageauc, 2017-06-27:
I have updated the wiki to include usbmount.  I plan on adding some of this to the install script to make it easier for novice users but in the mean time the wiki should give enough information.  Let me know if this resolves this issue and I will close.  Claude ...

See usbmount section at top of wiki page
https://github.com/pageauc/pi-timolo/wiki/How-to-mount-external-storage-drive

pageauc, 2017-12-14:
I have improved the wiki and latest version of pi-timolo now includes rclone as default.  Rclone allows syncing files with a large number of remote storage services including google drive.  See pi-timolo wiki for more details https://github.com/pageauc/pi-timolo/wiki

closed 2017-12-14 by pageauc