pageauc/pi-timolo #33: hook for video files missing ... can i convert h264 to webm

ozett, 2017-01-04:
hi,
i found your code very useful for my case, 
but i am viewing my files in directory view of my local apache on the rasperry with this:
https://larsjung.de/h5ai/

but there is no web-support for h264 files, i would like to convert them after recording to mp4/webm with ffmpeg. 
do you see a way to configure/or trigger a script for converstion after recording motionvideo. 

this way i could view the video in browser, like in motioneye..

thanks for help.
tozett

pageauc, 2017-01-04:
I have written a little bash script called convid.sh.  This utility will process any h264 files in the specified folder and convert them to mp4 and either delete or rename the h264 per variable del_h264 per comment in convid.sh script.  I have also added image flip variables to pi-timolo.py video mode function (bug).  You can add the convid.sh script to a crontab per script comments.

To install the new version execute the command below in a logged in SSH or terminal session on the RPI.  This will update the files in pi-timolo.py and not touch your data.  Note you will need to copy the config.py.prev to config.py per below to save your previous settings.  Also the gpac will be installed so MP4Box will be available. 

    curl -L https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo-install.sh | bash
   cd ~/pi-timolo
   cp config.py.prev config.py

I would recommend you keep video's short eg 10-20 sec

Also I have used my webserver.sh script and the mp4 video's play OK in chrome.  You do not need to run Apache server.  See my wiki for more details

https://github.com/pageauc/pi-timolo/wiki

Thanks for your request.  I wrote this for myself and don't use video mode so it is nice to fix things for users who actually use other features.   I have not documented convid.sh but you can read comments in the script.  The variables should be obvious.

IMPORTANT.  The latest (most recent) video will not be processed since it could be in the process of being written to.  There are ways around this so do some testing and let me know if this is an issue.  Normally when motion is detected there will be more than one video generated and the latest may not be that significant.

Your feedback is appreciated.

Regards Claude ....

pageauc, 2017-01-04:
See wiki convid.sh section for more details per link below

https://github.com/pageauc/pi-timolo/wiki/Utilities

closed 2017-01-04 by pageauc

ozett, 2017-01-05:
hi, thanks for reply.. 😄 

at the moment i am updating my install and afterwards see how it goes with convid.sh.
thanks a lot.

i like to have some kind of surveillance-cam, so even the last videos could be interesting.
may i have to consider the finishing of writing a video to disk, before a convertion can start.

may this is an option: inotify-deamon...
https://www.cyberciti.biz/faq/linux-inotify-examples-to-replicate-directories/

another idea was, to have a thumbnail from the video. 
as i read on your side that you tested with motion-detection, i could be another feature to record motion-movies and store a thumbnail image from the most moving sequence ot the movie.
what do you think?

i come back with some exprience of mp4-conversion.

👍

ozett, 2017-01-05:
i started convid.sh and looke with my directory-enabled webserverscript (h5ai) into the nfs-mount.
now i get thumbs for all the mp4 inside the folder "magically".
see how it looks:

[image]

i will test your webserver.py and give this also a try.

ozett, 2017-01-05:
i forget to click on a mp4-file....
and i did this. wow. now the video plays in the browser windows without any other component.
looks promising....

[image]