pageauc/pi-timolo #34: motion-video duration independent of motion?

ozett, 2017-01-05:
hi,
i read the wiki about convid.sh and i think i misunderstood the conception of recording motion from motion-detection.

as far as i understand the wiki, a "motion" triggers only a videorecording for a fix period of time. 
after that the process starts.

it is _not_ that the video is recorded _as long_ as there is somethin moving in the camera/scene.

am i right? is this by default/architekture of python-video. 
or is this the same with the linux "motion" binary? 

or could it be coded (opencv?) that the recording is going on from first trigger to the end of a moving?

thanks,
tozett

pageauc, 2017-01-05:
The design of pi-timolo is that it can record video, images or do motion detection but not at the same time.  I do a lot of opencv work for motion tracking etc but this application is designed to be simple so it works on older RPI hardware.  Therefore it does not use OpenCV to analyse video stream for motion.  Currently when motion is detected the camera is switched to record video for a predetermined time period (eg 10 seconds)  After the recording stops the camera goes back to motion detection and if there is motion again then another video is recorded.  In most cases this works well enough and the gap between recordings would be short.  The main thing is that this works reliably as you have found out with other RPI security apps.  I found that some of these other projects can sometimes crash.  This can be a problem for a remote unattended camera.
At this point in time there is no intention of integrating opencv analysis of video for motion into pi-timolo.py

See some of my other github opencv projects like speed camera, motion tracking, camera tracking, Face Tracking, Hot Spot Game Etc.  See these projects for code that does real time analysis of camera video for motion detection, tracking, face detection Etc.
My Github home page is here https://github.com/pageauc

Regards
Claude ...

closed 2017-01-05 by pageauc

ozett, 2017-01-05:
thanks for this longer explanation.  👍 
helps understanding the softwaredesign a lot.

pageauc, 2017-01-05:
Thanks for you interest in my project.  I think that as you use pi-timolo and some of it's features you will find that it is good enough and stable.  I have had remote camera's run for several years unattended without serious problems and uploading images to my google drive.  Google drive works well for browsing uploaded files and can seamlessly sync to a local PC directory for further processing.  I quite often process images into a video on my PC.  Since band width can be an issue on a remote camera, I try to keep things to basics.  gdrive sync does this quite well.

Regards
Claude ....