pageauc/pi-timolo #35: makemovie.sh  --avconf destination path

ozett, 2017-01-05:
hi,
after modification of the destination path, the movie was still created in the pi-timolo homedir.
may there should be destination-path-variables in the avconv line?

[image]

❔

pageauc, 2017-01-05:
The script has a variable
    folder_destination=$DIR
$DIR is the default folder that the script is launched from but can be changed to any valid path you wish
Eg.
    folder_destination="/home/pi/video/cam1"
path can be full or relative
Eg 
    folder_destination="$DIR/video/cam1"
    folder_destination="./video/cam1"

Just make sure the destination folder exists since the script will not create it.
Hope this answers your question.

ozett, 2017-01-05:
yes, but look at the avconv line.

you coded there the use for the destination path the variablename of $DIR
(and i am no coder, but i think...) -- it should be $folder_destination.

avconv creates the destination in $DIR, but should create it in $folder_destination

you see my point?

pageauc, 2017-01-05:
Thanks
I have fixed code and released ver 2.9 of makemovie.sh.  You can download it per command below

    cd ~/pi-timolo 
   wget -O makemovie.sh https://raw.github.com/pageauc/pi-timolo/master/source/makemovie.sh

permissions should be OK.
Give this a try and let me know if you find anything else.  I will close this issue.
I appreciate your input.  Note folder names with blanks need to be in quotes but otherwise do not need quotes.  

Regards
Claude ...

closed 2017-01-05 by pageauc

ozett, 2017-01-05:
i found the same variable-typo also in "makedailymovie.sh".
i changed it by hand accordingly. may you update your source?

[image]

ozett, 2017-01-05:
another misunderstanding on my side.
"makedailymovie.sh" converts only images (.jpg) from the motion-folder to a video.
like the "makemovie.sh"-script does for timelapse-images.

my misunderstandig is, that i _thought first_ that the dailymovie script could create one movie-file from the motion-dir. Out of all the (daily) recorded movie-files. 

this way - i thought wrong - i could get a daily movie from all the events of the motion-movies from one day. than i could keep the daily-movie and store the single-files in an daily archive-folder. this would clean up about the mass of files..

but that was not intended. i guess...

💡 maybe an idea for development ❔