pageauc/pi-timolo #36: feature-request: timelapse-movie, archive images

ozett, 2017-01-05:
hi,
i made a movie from the timelapse-images. fantastic. 🛩 
as i thought of now let the rasperry make a daily movie from timelaps, i could run makemovie.sh from cron.

did you consider to delete timelapse-images after finishing the movie? 
like in the convis.sh script the deletion of the original .h264 after convertion to mp4?
or move all the daily timelapse-images after making the movie to an daily/montly archive-folder?

for me this would be a nice feature. so i could let cron make a daily timelapse movie and the script renames the folder to an daily archive, or moves the images there into.

what do you think ❔

pageauc, 2017-01-05:
I will look at making it an option that can be turned on or off.  I do not like to arbitrarily delete user data since it can cause problems.  I have a timelapse setup that runs all year long and saves files to an older attached ext USB 1TB Hard drive formatted ext4.  This means space is not a problem and the RPI SD card does not get overused.  The makemovie.sh script was done pretty quickly when Jessie eliminated ffmpg and replaced with avconv.  avconv needs files named consecutively and that is why I create a temporary working folder with links to the images.  This avoids the encoding stopping prematurely when an image in the source folder is deleted and stops the encoding at that point.  Basically the links create a number sequence even though the original source folder may have some files deleted.  Normally I would delete very small size files since they may be very black or very white.  This setup allows for successfully encoding everything correctly.

Hope this explains the program design

Claude Pageau

ozett, 2017-01-05:
wonderful. thanks for that. 
i wait for the option 😃 , and i appreciate your comments. 🥇 

regards,
ozett

pageauc, 2017-01-05:
release 3.0 now includes option to delete source files (see caution in code comments)

closed 2017-01-05 by pageauc

pageauc, 2017-01-07:
makevideo.sh now replaces makemovie.sh and makedailymovie.sh
For more details see issue  https://github.com/pageauc/pi-timolo/issues/39

ozett, 2017-01-11:
should'nt the deleten of the original timelapse-images also reset the image-counter?
it counts further with the last number, but the images are deleted. it could start over with fresh numbering?

or maybe consider a numbering scheme like YEARMONTHDAYTIME, ie. 201701110515, 2017-01-11-0515

pageauc, 2017-01-11:
In config.py you can set motionNumMax = 0.  This will make numbering continuous forever and not overwrite after a specific count.   You have to be careful with this option since it can fill up disk space so you would have to use an attached hard drive or move files to a network share with more disk capacity

Also you can set  motionNumOn = False and the filesnames will be displayed by date and time so you can tell by the video file name when it was recorded.  Give it a try. 

Claude ...

ozett, 2017-01-11:
helpful hint, i looked this up afterwards and will give it a try..
thanks for pointing me in the right direction ... 🏅 

[image]