pageauc/pi-timolo #38: convid.sh with delete of original files = deletes original creation date

ozett, 2017-01-06:
hi,
as i am playing with convid.sh i am observing that the file creation time on the mp4 files is the creation time of my mass-conversion date, ie. hour:minute.
here i am loosing the original recording date (hr:min) from the original h264 file.

maybe you want to consider building in the original source an option to transfer the date,time,minute to the converted file (before deletion)?

apreciate your opinion,
regards
tozett

pageauc, 2017-01-07:
Did some research and does not appear to be an easy solution.  You may want to change pi-timolo setting in config.py called motionNumOn=False.  This will create a filename with a date/time creation stamp as part of the video name.  Give this a try.

BTW I have written makevideo.sh to replace makemovie.sh and makedailymovie.sh.  Rerun the easy install curl command on the pi-timolo main page to update.  See wiki for more details.
Claude

closed 2017-01-07 by pageauc

pageauc, 2017-01-09:
Updated convid.sh to ver 0.7 that should now copy the original h264 file date to the mp4.  This should keep the files in order since convid.sh would have the mp4 date the run time of convid.sh.  This should work even if convid.sh is run days later.  The mp4 will have the same date as the h264.  This uses the touch -r command

pageauc, 2017-01-09:
Just posted updated convid.sh that process all h264 files.  It will wait for the first file in case it it still recording so you need to adjust sleep.  Also it MP4 dates are same as origina h264.  See code at link below
https://github.com/pageauc/pi-timolo/blob/master/source/convid.sh

You can upgrade or just download updated file per
wget -O convid.sh https://raw.github.com/pageauc/pi-timolo/master/source/convid.sh

ozett, 2017-02-10:
after a while i triggerd convid.sh... lets see how it performs with this amount on my nfs-share
 😀 

[image]

pageauc, 2017-02-10:
I previously added a touch -r command in convid that syncs the date/time info of the original h264 to the output MP4.  

else
    # sync mp4 file dates from corresponding h264 file 
    /bin/touch -r $fileparam $MP4filename.mp4
    if [ "$del_h264" = true ]; then
        echo "STATUS - Deleting" $fileparam rm $fileparam

I have verified this is working but not across a NAS.  You may be getting issues with copying across to NFS mount.  Make sure you have the latest version from github.  Unfortunately whenever a file is copied the original date/time can be changed and it is hard to carry the original flawlessly over the whole chain.  especially if a user does a copy somewhere.  Please check to see if this is the case...  You may have to insert additional touch -r commands

Let me know your findings ....

Also 
I am working on a new version of pi-timolo that is going to be called pi-timolos  (timelapse, motion detect, low light, speed)  The speed uses my speed camera code (motion tracking) as an option instead of motion detection.  Using speed tracking of motion will confirm an object is actually moving and not just spurious motion.  Also revamped all the variable names to make them more consistent.  This will take a while but have completed initial integration and it is working but still needs more work, since speed camera did not have all the bells and whistles of pi-timolo..

Regards
Claude ....

ozett, 2017-02-10:
Hi Claude,
thanks for always rich explanations, helps a lot.

i nfs-mounted the nas and used there the subfolder directly as storage-folder for pi-timolo.

```192.168.14.243:/data01/data01 /root/pi-timolo/source/media nfs auto,rw 0 0```

i only have .mp4 files in there, "touch" seems to work on the nas. i guess 😕 

the raspberry-one rebooted over time (maybe powersupply issues, i am in the process of checking this), but that way there grew the collection of .mp4. as i wanted them to be a daily video a least, i now triggered the concatenation over all my motion-recordings to clean up the past 😄 

i will keep up with your updates on github, thanks for sharing👍 

(for the next version, you are working on, i add one of my ideas in another post. maybe you'll find some interest  in this ...who knows...)

ozett, 2017-02-10:
as i am in the progress of testing motion-detection with the raspberry cam and your piece of software has prooven to run stable over time, i had the idea of opening our house-door after a face-comparison.

first con against this: i am no softwaredeveloper...
pro: i found apis on the internet, which seems to make this easy..

http://blog.mashape.com/list-of-10-face-detection-recognition-apis/

this is, what i thought of: 
as there are coming a lot of parcel services to our house, which i gave them RFID-keys to open the door between 08:00-and 17:00, they often dont carry the chip/key with the them. or in the car.
now your software (and i saw your opencv experiments) let me think, that i could be possible to have a cam on the door, do a face-recoginition and compare this image with a list of allowed face.images. and use this to open the door. 
i am aware of a lot of security-issues, but the door opens up only to a secure area for delivery. no problem here. and i wanted to test the process of face-detection with a cam, looking to the street-side of our house.

i see the problem, that the sofware must identify the best shot of a face, to compare it (with opencv , or an online-api) against a stored face-image while the person is in the view of the cam and approaching the door.
i could build a PIR-trigger (or a button) for shooting the door-opening image for face-recognition, but i guess it could be build smarter...

thats the rough outline of the idea, maybe it strikes you and you can consider some points while building your new version of pi-timolo? your are some kind of coder, that i  a not ... 😢 

what do you thing 👼 ?

i did some test with two apis, looke really promising to me...

ozett, 2017-02-10:
[image]

# **Even better Features: FACE COMPARISON...**

[image]

pageauc, 2017-02-10:
I have github code for magic box but have not done anything with it.
Programming is just a hobby so can't work on everything but have it in the
back of my mind.  Even mashing someone else's code can take quite a bit of
time.

Claude ...