pageauc/pi-timolo #40: numbering scheme for images/motion

ozett, 2017-01-11:
hi,
as i look up the folder with the saved motion-video and image files daily (i am trying to use pi-timolo) as some kind of surveillance-cam), the files are numbered from 1000 upwards.
at the moment i try this with directory browse (for some other reasons), and not with your webserver...

on my small mobile display i see the file-name and get no clue at wich time the video/image was shot.

do you think it could be an option to have a choice of single-counting the images/video upwards or get a counting-number in the scheme of YEARMONTHDAYTIME (201701110525). or maybe even more human readable like 2017-01-11-0525 ❔ 

[image]

what do you think? :thought_balloon:

pageauc, 2017-01-12:
Change pi-timolo config.py variable motionNumOn=False.  This will output files with a date time stamp.
that way you can see creation date of videos.  Also I have added video join to convid.sh    Type convid.sh with not parameters to see help.  I don't have the wiki updated yet.  makevideo.sh also has a minor update 
You can update these

wget -O convid.sh https://raw.github.com/pageauc/pi-timolo/master/source/convid.sh
wget -O makevideo.sh https://raw.github.com/pageauc/pi-timolo/master/source/makevideo.sh

Doing a convid.sh join    will slow a quadcore RPI but I can still run pi-timolo.py and the h264 video conversion can happen in parallel but just a little slower.

Let me know you comments
Claude ...

pageauc, 2017-01-13:
I have added start DateTime text annotation on the video recording.  Use the image config.py settings for text to turn on and control text size.  Default colors are black text on white background

Use these settings for the text annotation on the video recordings.
showDateOnImage = True     # Set to False for No display of date/time on image default= True
showTextFontSize = 18      # Size of image Font in pixel height

closed 2017-01-13 by pageauc