pageauc/pi-timolo #41: cleanup of all scripts into one: makemovie/makevideo/makedailymovie

ozett, 2017-01-14:
my usecase is a little surveillance cam on the raspberry.
and i want to keep a simple and clear archive of daily-happenings.

1) therefore i use the scripts to create a daily movie of the daily motion-recordings
2) a script for making a movie of the daily-timelapse.jpgs
3) an let the scripts clean up the recordings-folder (deletion/archive in daily folder would be nice).

now as there are three scripts for that in the source, i could get confusing.
my suggestion would be, to make one clear script with options and a single config-file.

what do you think? 😇

pageauc, 2017-01-14:
Greetings

Yes combining and reading settings from a conf file is a good idea.  This
should be pretty simple with a bash script since you can just specify a
source filename to read variables (or code) from another file.  Will see if
it will override default variable settings already in script..  I will give
this a try.  Making one large file might affect performance for the auto
conversion of h264 to mp4 since it is done as a subprocess, so keeping the
size of that might be better (but may be cached by os so may not be an
issue.

Remember I just wrote this for myself and decided to share, so a lot of
what I did was adhoc.  It is nice to have another perspective.

Will see if I can have one project conf file that will override existing
default variables.

Claude ...

pageauc, 2017-03-24:
makevideo now implemented

closed 2017-03-24 by pageauc