pageauc/pi-timolo #42: convid.sh -- join does not delete mp4

ozett, 2017-01-15:
hi,

i used convid.sh to join my daily recordings. these are in .mp4 format since i set pi-timolo in your new version to immediately convert all h264 recordings into mp4.

as i would like to join them into one file, i would also like to cleanup/delete the single-files afterwares.
but convid.sh does not have this option. 

if you are considering one-script for all movie-operations, you may want to consider deletion after 
join 😵  😁

pageauc, 2017-01-15:
Default is to delete.  Can you check permissions.  I pi-timolo is running
from /etc/rc.local and is not being run as pi user then root may own the
files and they are not getting deleted.  Entry should be

    su pi -c "/home/pi/pi-timolo/pi-timolo.sh start > /dev/null"

Check permissions.

See wiki here
https://github.com/pageauc/pi-timolo/wiki/How-to-Run-pi-timolo-on-Boot

ozett, 2017-01-16:
i meant deletion of original mp4 files, but it seems to be my fault...
the old ones are deleted, but as the process finished, there are still (again) 52 files.

at first sight this looked like a lot an i thought there had been no deletion,
but this count was the reserve or meanwhile new files ... (?)

so the only thing left of convid.sh join seems to me the resulting filename. 
evertime the same confid__.mp4
maybe better to use time-day-options... or as a name from-time_to_time...?

[image]

closed 2017-01-30 by pageauc

ozett, 2017-02-25:
just to give some feedback: 
convid was not put on cronjob by me. i forgot. than i started it after a while.
now it runs since 3 (or more) days now, on my raspi-one. and its still running  😄 

[image]