pageauc/pi-timolo #43: Gdrive sync delay and a typo in config.py file

Lama81, 2017-01-25:
When I use the sync.sh script, it stay at "resolving" for near 4-5 min before uploading. Is there anything I can do to shorten that delay? Everything else work without any flaw. 

......

little typo in config.py file. The directory to webserver script is "homw/pi/pi-timmolo/" should read home. Little correction.

Last error that I was unable to solve is probably because of my knowledge level.  I had an error will executing the webserver.py script. Error with line 155 os.chdir(web_server_root).  I put a # to disable the line and the websever was running after that.

Good work with this app. Thanks !

pageauc, 2017-01-25:
Lama

slow gdrive
If you have a lot of files then gdrive will take quite a while to review
all the files for changes before actually doing the sync.  This takes
longer the more files you have and that is why I set my motion images to
500 max and then recycle the numbering.

The latest github version corrects the config.py typo You can install with
wget but will loose come of your previous settings.  Since there has been a
major update to 4.0  you might consider a fresh install since I have a new
menubox.sh admin menuing and config.py now has the webserver.py variables.
The install will by design not overwrite existing config.py or .conf
files.  If you run the curl bash script from the github repo main page it
will refresh the executable files

curl -L https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo-install.sh
| bash

The config.py latest update will be config_new.py  You can copy this over
the existing

cp config_new.py config.py

to get the changes and edit settings to match prev file.  A copy of the
original will be config.py.prev so you can transfer changes.  The shell
script conf files will have .1 , .2 etc ending.  the highest number will be
the latest.  If you want to update these then mv the original eg

mv convid.conf convid.conf.bak
 then
cp convid.conf.1 convid.conf

You might also want to consider a fresh install since the new config is
setup to put motion, timelapse and video in a media folder.
See the wiki for more details
https://github.com/pageauc/pi-timolo/wiki

Let me know how you make out.  I would like feedback on the new version

Lama81, 2017-01-27:
Hi,

First, thank for that fast reply. I didn't expect such a fast answer. I 
juste did a fresh install. Everything run as aspect. Awesome work here.

I have manually edit the config file and look after in your menu script 
and the changes show up there too. Nice !

The gdrive sync his working too. Slow.. but it work. I have just see one 
little problem. The webserver go down often. I think that is because of 
the process of the Raspberry Pi 2 that is max out. l Not sure if it's 
the cause but usualy it's happening when the gdrive is running. Any cue 
? On my side I have made a script how set up the

On my side I will look for a ftp pull or push solution when I will have 
the time.

By your name, I suppose that you speak french too which is my first 
language. If it's the case I will email you in french it's easier for me 
than english.

Again Thanks !

Mathieu Lacharite

pageauc, 2017-01-27:
> source/pi-timolo-install.sh

Yes my heritage is french. Although my eight younger brothers and sisters
can speak french the older boys including myself went to english only
schools so we are not very fluent in the french language.  I do speak quite
a few programming languages though.

I have run the web server on a RPI model A and some RP2's as well as RPI3's
without issue.  If you have gdrive syncing,  it will web refresh will bog
down due to network response times, especially with wifi.  Not so bad with
wired connection.   I have these set to motion image (not video).  One of
my security cameras has run the webserver continuously for well over a
month without issue but I only check it once every few days.  When I access
the web page tab on my browser it takes a few seconds to come up but
otherwise is functional and responsive after that.  It is possible to hang
the webserver due to a file io problem.  I have had bad sd cards that
caused me headaches until I replaced with another fresh sd card and the
problem went away.  This sometimes can happen with Kodi and depends on the
quality of sd.  That said I have had serverl RPI's running continuously for
several years and only updated to Jessie recently.  One is an original 256k
RAM RPI that runs Pi-Hole add blocker and rarely gets shutdown or rebooted
(It is on a UPS)

Claude ...

closed 2017-01-30 by pageauc

Lama81, 2017-02-23:
Hi Claude,

Sorry to bother you, in fact told me if I have too many question lol. Is 
it possible to change the delay between each capture on timelapse 
setting in pi-timolo. What I would like to do is to raise the number of 
image took when motion is detect. Like motion detect : took 1 picture 
each 1- or 2 seconds for 10 seconds. Is it possible? The less I was able 
to get was like one picture each 5-10 secondes at his best. 
Quciktimevideo is not an option for my project but it's working great.

Thanks !

Mathieu Lacharite

pageauc, 2017-02-27:
You can try using the quick timelapse feature.  Change the config.py setting

    motionQuickTLOn = True

Then you can adjust the time period that quick timelapse feature is run for using 

    motionQuickTLTimer = 10

It is the number of seconds that the quick timelapse runs for after motion is detected.  Set this to the number of seconds.

The camera needs time to warm up and establish white balance.  During the pi-camera initialization there is a sleep to allow camera to establish white balance.  If a very short delay is specified the camera white balance will not be correct and images will not be consistent.  The default is .5 seconds on line 364 
def takeDayImage(filename):
    # Take a Day image using exp=auto and awb=auto
    with picamera.PiCamera() as camera:
        camera.resolution = (imageWidth, imageHeight) 
        time.sleep(0.5)   # sleep for a little while so camera can get adjustments

.5 seconds is about as short a warm up as reasonable and still get half decent image consistency.  You will still find variations in white balance a this low level.  BTW the default rpistill is 5 seconds.

Lama81, 2017-02-27:
Thanks for the information. I didn't know about the White balance delay. I had notice thoses variation in WB between some picture. Didn't know the cause. Now I do. I'm still exploring the features of pi-timolo and I enjoy it. I was expecting that I will be able to take burst picture like a do with my dslr camera (like 5 fps). The sleep time to adjust White balance will not make it possible. It will still do the job with less frame. 

Have a Nice day
Mathieu

pageauc, 2017-02-27:
Using raspistill you can do a timelapse sequence that only has one warm up but pi-timolo uses pi-camera python library and needs to switch between motion detection and taking images or video sequences.  This means the camera needs to switch between these modes and warm up between.  It would be possible to shell and run raspistill in a timelapse sequence

See this link for details

https://computers.tutsplus.com/tutorials/creating-time-lapse-photography-with-a-raspberry-pi--cms-20794