pageauc/pi-timolo #44: Dependency on pyexiv2

trunksio, 2017-03-05:
There seems to be a dependency on pyexiv2 which no longer has a pip install option? 

sudo pip install pyexiv2
Downloading/unpacking pyexiv2
Could not find any downloads that satisfy the requirement pyexiv2
Cleaning up...
No distributions at all found for pyexiv2

or am I missing something in the setup. 
raspberry pi zero: raspbian 8 jessie

pageauc, 2017-03-05:
Dependency is installed as part of  https://github.com/pageauc/pi-timolo/blob/master/source/pi-timolo-install.sh line 109
sudo apt-get install -yq python-picamera python-imaging dos2unix python-pyexiv2 libav-tools

Just tried
sudo apt-get install python-pyexiv2

This works so not sure where the error is coming from.  I do not do a pip install in the script

This library is just used to transfer image exif data when the image text is put on an image otherwise that meta data would be lost..  

Please provide more information on how to reproduce your problem and let me know

Claude ...

trunksio, 2017-03-05:
For some reason that library did not get installed in the initial install script. 
I have run the apt-get install and it is working fine now- thanks!

closed 2017-03-05 by trunksio