pageauc/pi-timolo #55: NameError: global name 'motionCamSleep' is not defined

ibbaman, 2017-04-15:
Hi, 

I'm new to this, but when I tried to start the application after install, it seems to fail due to a variable not being set. Did I do something wrong or is there an issue with the current version?

+++++++++++++++++++++++++++++++++++
pi-timolo.py - Exiting Program
+++++++++++++++++++++++++++++++++++

Traceback (most recent call last):
  File "./pi-timolo/pi-timolo.py", line 751, in <module>
    Main()
  File "./pi-timolo/pi-timolo.py", line 652, in Main
    displayInfo(moCnt, tlCnt)
  File "./pi-timolo/pi-timolo.py", line 182, in displayInfo
    print("               motionPath=%s  motionCamSleep=%.2f sec" % (motionPath, motionCamSleep))
NameError: global name 'motionCamSleep' is not defined

pageauc, 2017-04-15:
You probably have an older config.py
The latest version is here
https://github.com/pageauc/pi-timolo/blob/master/source/config.py

You can either do an upgrade by running the pi-timolo-install.sh again or
if you bring down a new copy of config.py and try again per commands below.
You will then need to transfer any changes.  Alternatively you can add the
two new variables to your config.py per github link above.

      cd ~/pi-timolo
      cp config.py config-bak.py
wget -O config.py
https://raw.github.com/pageauc/pi-timolo/master/source/config.py

There are now separate sleeps times for motion and timelapse to allow
camera time to stabilize and establish white balance.
motion is shorter to minimize time for motion detection.

This should resolve your problem unless I have missed something

Claude ....

ibbaman, 2017-04-15:
Seems to have resolved it indeed... Thanks for the quick support 👍

closed 2017-04-16 by pageauc