pageauc/pi-timolo #56: Released ver 4.40 of pi-timolo.py for testing of simplified logic for takeNightImage function

pageauc, 2017-04-16:
pi-timolo.py release 4.40
Since an earlier update of the picamera python library and an associated camera driver update I have been getting greenish images at night and twilight.  After testing I have simplified logic to allow camera to determine settings based on camera.framerate_range.  This seems to have resolved the igreenish image issue, although some images are slightly darker than I would like.  Please test this release and post a github issue if it works well or you discover problems.  If everything is OK, I will cleanup unused functions and optimize.  Will also add pixAve logging info somewhere since it is still used to determine day/night/twilight.  Also want to set a lower pixAve setting to avoid camera taking black images when light level is very low. 
Your comments are greatly appreciated ....
Thanks Claude .....

ozett, 2017-04-17:
👍  🥇 
(will test the next days...)

pageauc, 2017-04-17:
Thanks  Appreciate results of your testing.

Claude ....

pageauc, 2017-04-17:
Found a bug when taking motion day images.  Now uses motionCamSleep instead of timelapseCamSleep.  Now captures motion with shorter delay.  default is 0.7 sec instead of default timelapseCamSleep that is 4.0 seconds.  Found motion was out of frame due to the 4 second delay.  Things seem to work better now.  Still testing the greenish image problem for takeNightImages.  Working good so far.
Please update to this version (sorry for the bug)
Claude ...

closed 2017-04-21 by pageauc