pageauc/pi-timolo #57: Auto deleting old files

KminekMatej, 2017-04-18:
User should put config.py limit of disc size. When this size is reached, older files would gets automatically deleted to create space for the new motions.

pageauc, 2017-04-18:
Yes I have code that will do this and will take a look at adding this feature as an option.  Setting a percent of disk full before deleting files.  eg 80% full  and a value for MB to delete or recover.  This will avoid constant activation of deletion that may slow processing down.  That is the main reason I did not add feature.  Instead I use recycling to  overwrite older images.
Claude ...

pageauc, 2017-05-08:
Version 6.00 now released that includes disk space management.  Please let me know if you have any issues.

closed 2017-05-08 by pageauc