pageauc/pi-timolo #58: Release 4.50 with improved takeNightImage with better low light and should resolve most greenish images problems

pageauc, 2017-04-21:
This version must have the revised config.py file.  pi-timolo.py release 4.50 has improved night low light and should resolve most green images issues.  I have done testing with an indoors and outside facing camera.  There is still room for tuning.  There are lighting bumps between day and night transitions.  I thought I would release this version so others can work with it and let me know any issues.  It is recommended that you have the latest raspbian firmware by doing a but this update may work ok.  I have only tested on up-to-date raspbian and firmware.

    sudo rpi-update

Let me know how this update is working for you.
Thanks for your support
Claude ....

1kane, 2017-04-27:
Hay,
i am interessted in taking night-day day-night timelapse with the PI and im still looking for a working programm that can capture the transition. can you upload a example video of ur twilight captures?
or can u tell how nice/or not it is working?

pageauc, 2017-04-27:
pi-timolo release 4.5 Twilight transitions are a little bumpy due to recent
changes to I believe pi-camera firmware and/or pi-camera python library.
After an update pi-timolo started to experience  low light greenish or
badly colored night shots.  After a lot of testing and trying different
things I believe night images are better but the algorithm and settings
need to be tweaked more to avoid over dark or light images during twilight
transitions.  I am working on this when there is time.  At any rate it is
not working as well as it used to before the update but is better than it
was.  I have some YouTube videos that were taken before the problem arose.

https://youtu.be/pFkt0ZOrPzg

https://youtu.be/x06MUtLAhcU

Claude ...

pageauc, 2017-04-28:
I have updated pi-timolo.py and config.py to release 4.60.  This modifies takeNightImages for better low light logic.  There may still be a bit of a light bump in places that might need settings to be tuned.  There are now several light thresholds based on the dayPixAve.
nightTwilightThreshold  is the transition between day mode and twilight using framerate_range
nightDarkThreshold is the transition between twilight and dark.  Uses variable shutter exposure times
nightBlackThreshold is for very low light and uses maximum low light settings.

I have tested this indoors and outdoors.  Some settings may need to be tuned to reduce the lighting bump between transitions.  Settings may need to be adjusted up or down to optimize.  I have not experienced greenish images although you will get very dark or very white images if you change lighting quickly when a long exposure is in progress.  This is normal and to be expected.  The next image should be OK      
Minimal upgrade from logged in ssh or terminal session

    cd ~/pi-timolo
    cp config.py config.py.old
    wget -O config.py https://raw.github.com/pageauc/pi-timolo/master/source/config.py
    wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

Please let me know how this is working for you. good or bad
Thanks Claude ....

closed 2017-05-03 by pageauc