pageauc/pi-timolo #63: saveRecent gets stuck if it can't remove a file

dbach, 2017-06-16:
First, thanks for all your work putting this together and publishing it! I ran into the following issue, which I think is worth reporting. 

The saveRecent function catches errors when it tries to remove a file, but it does not remove the non-removable file from its list, so it tries to remove the same file over and over again. 

This issue bit me because my NAS indexed the volume where I stored my pi-timolo images and left indexing metadata directories everywhere. In my case, there was a directory (_@eaDir_) that could not be removed with os.remove and the logs filled with messages like:

```
2017-06-15 21:51:41 ERROR    saveRecent Cannot Remove /mnt/capture/recent/timelapse/@eaDir - [Errno 21] Is a directory: '/mnt/capture/recent/timelapse/@eaDir'
2017-06-15 21:51:41 ERROR    saveRecent Cannot Remove /mnt/capture/recent/timelapse/@eaDir - [Errno 21] Is a directory: '/mnt/capture/recent/timelapse/@eaDir'
2017-06-15 21:51:41 ERROR    saveRecent Cannot Remove /mnt/capture/recent/timelapse/@eaDir - [Errno 21] Is a directory: '/mnt/capture/recent/timelapse/@eaDir'
2017-06-15 21:51:41 ERROR    saveRecent Cannot Remove /mnt/capture/recent/timelapse/@eaDir - [Errno 21] Is a directory: '/mnt/capture/recent/timelapse/@eaDir'
```

pageauc, 2017-06-16:
Not sure how to resolve this issue. It appears that @eaDir is a directory and not a filename.  Not sure how the pi-timolo filename is getting substituted by a directory reference.
 Can you provide more information..  
- Type of file system on NAS   
- Pi mount command in fstab or a manual script    
- List of permissions on NAS as seen from the rpi.  eg ls -l /mnt/capture/recent/timelapse   

Does this work if indexing is turned off for the NAS recent folder.
Thanks

dbach, 2017-06-17:
The _@eadir_ directory is created by the indexing process in _every_ directory (this seems to be a bug in the Synology NAS software). The saveRecent function lists all files in the recent directory and if there are too many, tries to remove them one by one. I think the best solution is for saveRecent to only try to remove the files that match its prefix -- that will also prevent potential issues with other operations that might store sync metadata in the recent directory. 

I killed the indexer and removed the problem directories so I don't have the detailed directory list for you at this time.

pageauc, 2017-06-17:
I have implemented your solution.  This is something I never thought of but now makes sense since it is possible other directories and /or files are in the recent folder.

closed 2017-06-17 by pageauc