pageauc/pi-timolo #66: Webserver issue

nigels0, 2017-06-26:
Hi,

The webserver.py doesn't work with Safari on mac or on an iPad though works fine on firefox on Mac.

On Safari on mac and on iPad, I get:

----------------------------------------------------------------
192.168.0.2 - - [26/Jun/2017 13:03:13] "GET / HTTP/1.1" 200 -
192.168.0.2 - - [26/Jun/2017 13:03:17] "GET /20170626121316.mp4 HTTP/1.1" 200 -
----------------------------------------
Exception happened during processing of request from ('192.168.0.2', 55298)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe

pageauc, 2017-06-26:
I originally tested webserver.py with images on my ipod but did not at the time test with mp4.  video and images both work fine on my windows pc with chrome browser.  Tested images on my ipod and now does not show side menu.  I believe this might be due to updates to chrome on ios since this worked fine previously.  please confirm if images display OK on ipad with side menu visible.  Will take a look at this problem as well
Error 32 problem 
Below is a link that I believe explains the issue that is related to timeout of client before it has a chance to load the mp4 file.
https://stackoverflow.com/questions/11866792/how-to-prevent-errno-32-broken-pipe
I will take a look at problem and see if I can resolve the broken pipe issue as well as the side menu issue on IOS devices on Chrome and Safari.

Thanks  Claude ...

nigels0, 2017-06-26:
Hi,

I can confirm I'm getting the side menu visible (Images load fine but no video) on iPad with Safari and Chrome - but have the broken pipe issue.... so you must be right about the timeout.

pageauc, 2017-06-26:
OK Thanks

The iframe side menu selections are generated from A directory listing that is formatted as url links.  I will see if I can implement a SocketServer.TCPServer timeout to allow client to download mp4 file before webserver timeout..

nigels0, 2017-06-26:
Is it possible to have sub-directories on the side-menu selections, so you could have a few cameras rather than all on one flat list?

pageauc, 2017-06-26:
Yes you can have subdirectories and they open in a separate browser tab.  Not sure if you want to display a folder tree.  If you look at the webserver.py code it is just reads in a a list of the directory entries.  This list is then used to create the url links for the files and subfolders.  A new folder tab is  spawn if you click on a subfolder link.

nigels0, 2017-06-26:
Oops - my mistake! should have tried it first :)

pageauc, 2017-06-27:
Still doing some testing with IOS url links and broken pipe problem.  Android seems to work file as well as a PC web browsers.  In doing research it appears IOS video timeout/broken pipe is a recurring problem.  Some have found changing the player works.  The Apple video player seems to have some video playback issues on IOS devices.  It could be that the MP4 files encoded by RPI is not very compatible with Apple.  I will try to run some videos through handbrake and put back on the RPI.  This will confirm if issue is encoder based.  I suspect when the URL is requesting MP4 file from RPI that it closes connection when it cannot read the file format.   
I have already tried changing the webserver URL format for video player and reducing size of video, Etc with no joy.

pageauc, 2017-06-27:
Tested with 640x480 video's converted with HandBrake using web optimize, an ipod 5G Support.  Still getting broken pipe.  Not sure where the problem is.  Could be a security issue on IOS due to local web server.  Still scratching my head since same code works great on Android, windows, Raspbian (chrome) and debbian.  So the problem does not seem to be the HTML code but rather some glitch with IOS that others have experienced.  Does not seem to be an obvious solution.   You might try installing another video player on IOS device and give that a try.

nigels0, 2017-06-27:
Hi,

I think the problem is buffering in that both the iOS and Mac version of Safari may need to bring the whole file into memory before playing - but times out in its calls to the disk (I'm using goggle drive to store the video files).

closed 2017-07-06 by pageauc