pageauc/pi-timolo #69: motion unfocused

mmcalmeida, 2017-09-07:
One of my cameras running pi-timolo saves unfocused motion pictures but saves focused timelapse pictures. Is the imagewidth and imageheight different for those tho options?

pageauc, 2017-09-07:
Ok.  The default config.py motion track setting is motionTrackQuickPic=True that uses a stream image since it is much faster.  If you want better quality motion image then set motionTrackQuickPic=False  
This will stop the motion track stream and safe a pi-camera image. There will be a several second delay for the pi-camera to switch between stream mode and image mode.  This will take a much better image.

Note  You can also adjust default jpg image quality using the config.py image setting

    imageJpegQuality = 20      # default = 20  Valid jpg encoder quality values 1(high) - 40(low)

20 is the default camera quality and should be OK.

Please note I converted pi-timolo to use opencv motion tracking instead of motion detection since it greatly reduces false positives and also allows capturing a quickpic frame from the stream.

Let me know if this resolves your problem
Regards Claude ....

mmcalmeida, 2017-09-08:
I have changed motionTrackQuickPic from True to False and now the motion images are focused.
Thank you very much.
Manuel

closed 2017-10-08 by pageauc