pageauc/pi-timolo #70: Timeout after motion detect

Ebihen, 2017-10-06:
Hi Claude,

thanks for sharing your work! I'm very much enjoying your script.
I'm running it on a Pi that is installed in my temporary workshop to document the process of building a small wooden sail boat. (www.aarnoudsebihen.wordpress.com))
Would it be possible to introduce a variable in your script that maximizes the number of motion triggered pictures per time? It would be good for me if I could let my Pi run your script and have it take a picture every 10 minutes if it detects motion.

Thanks in advance for your answer! Best regards,

Aarnoud

pageauc, 2017-10-06:
Not sure if I understand your request.

When motion is detected you can take 

1 - Take a single image,  

    motionVideoOn = False

2 - Take a series of quick timelapse images for a specific time period eg 60 seconds and interval between images eg 5 seconds  see example settings below

    motionVideoOn = False
    motionQuickTLOn = True  
    motionQuickTLTimer = 60     
    motionQuickTLInterval =  5   

3 - Take a Video for a specified time period eg 10 seconds per example below

    motionVideoOn = True      
    motionVideoTimer = 10  

4 -   You can also specify what to do if there is No motion for a specified period of time eg take a picture  anyway if  there has been no motion for an hour or specified number of seconds.  zero will disable this feature.

    motionForce = 3600

Sometimes I set my camera with a motionForce = 300  so it forces an image every 5 minutes  that can act as an easy way to get regular updates but will take an image immediately if motion tracking is activated, at which time the 300 motionForce seconds wait will restart.

You can also specify if you want to automatically turn off camera settings during  daylight or night time.  The default is to turn both these off.  Setting them both to True will take no images 

    noNightShots = False     
    noDayShots = False

There are many other combinations possible.  Sometimes I will set both timelapse and motion on

These variable settings can be found in the config.py file.

If these examples do not cover your requirement please explain the situation or outcome you would like.

Thanks Claude ....

Ebihen, 2017-10-07:
Thanks for your quick response!

I’ve been tuning with the variables in config.py , but haven’t been able to get the system behaviour I had in mind.
What I would like to achieve is that I get a picture every 10 minutes of my workshop every time I do some work there, but no pictures taken if nothing moves there (which would be the case if time lapse would be turned on).
The workaround I used until now to get the system behaviour is set the variable for white balance measuring to 300 seconds, but then a picture is taken 300 seconds after motion is detected.
The variable that would be helpful for my situation would be one that times out motion detection for a period of time after motion is detected.
Please let me know if I need to be more precise in my description ;-)
Best,

Aarnoud

pageauc, 2017-10-07:
You should be able to achieve what you want by using 

Motion Quick Time Lapse Mode

When motion detected take a series of  timelapse images for a specific time period  and with a specified interval between images. See example settings below thus you will get 6 images after motion spaced 
 10 minutes apart apart.  You can use either motion or motion tracking to activate this series.

motionVideoOn = False            # Take images not video
motionQuickTLOn = True          # Turn on motion timelapse sequence mode
motionQuickTLTimer = 3600       # On motion do timelapse sequence for 60 minutes
motionQuickTLInterval =  600      # Take image every 10 minutes until timer runs out

timelapseOn = False    # Turn off normal time lapse mode so only motion mode used.

motionForce = 0    # Do not force motion image if no motion for a period of time

I think this should accomplish what you are trying to do.  When motion detect or motion tracking is triggered it will take an image every 10 minutes for 1 hour then resume monitoring for motion again.

Let me know if this is what you were looking for

Claude ...

pageauc, 2017-10-07:
Please Note I found a bug in motionForce that did not disable this feature if value set to zero.  I have fixed this in pi-timolo.py release 7.9 that can be downloaded from GitHub. or use the command below to update version

    cd ~/pi-timolo   
    wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

This should fix this bug for the configuration above
Claude ...

Ebihen, 2017-10-07:
Hi Claude,

I tried the settings you suggested with the motionQuickTLOn variable and it seems to work fine! Thanks! I set both motionQuickTLTimer and motionQuickTLInterval to the same value of 600, and I have indeed got a picture for every 10 minutes I was in my workshop.
The only thing I found was when I afterwards checked the pictures; I use the file naming option with the time stamp (motionNumOn=False) but the time stamp in the file name doesn’t always correspond to the time the file was created. The file creation times for my test pictures show a pattern that could be expected: a picture taken at a certain time and 10 minutes later another one. I’ve three sets of two pictures taken with this logic. The time stamp in the file names however show another pattern; the time stamp in the name of the first picture of a set of 2 corresponds to its file creation time, the second picture however has a creation time of 10 minutes later, but a time stamp of 5 seconds later.
Best,

Aarnoud

pageauc, 2017-10-07:
Did you download the 7.9 release of pi-timolo that fixes the motionForce=0
problem.  Now it will disable motionForce correctly when it is set to zero

Claude ...

Ebihen, 2017-10-07:
I just installed it. More testing tomorrow.
Would your bug fix have anything to do with the time stamp-issue? At the time I tested it, I had the motionForce variable not set to 0.

pageauc, 2017-10-07:
Probably.  forceMotion was designed to take an image if not motion was
detected for a specified period of time. Default is 3600 or one hour.  By
setting the value to zero is should disable this feature but the bug would
force a motion image continuously which is not what you wanted.  The new
release fixes that problem so setting the value to zero disables the force
motion option.

I would set the motionQuickTLTimer for a longer time period like  3600 or
1800 seconds.  That way when you are in your shop working it will take a 30
or 60 minute sequence with 10 minutes between images.  Once you leave, then
the time sequence will continue for a while until timer runs out and only
start again if there is another motion triggered.  Depending on the point
in the sequence you will end up with a variable number of images at the end
of the sequence where you are no longer in the shop.

Claude ...

Ebihen, 2017-10-13:
I set the motionForce variable to 0 now, and that worked fine for me.
For the time being I'm going to stick to a shorter motionQuickTLTimer period of 1190 and a motionQuickTLInterval of 600. With that I should get a picture every 10 minutes I work there, and no more pictures once I leave. I'll change both variables to the setting you recommended once I'll be spending longer periods of time in the workshop. The current building phase requires me to do 30 minute jobs and then leave for a few hours to let the epoxy dry....

pageauc, 2017-10-13:
Sounds Great.  I suspected you were working on a shop project.  It is a
good application since recording time lapse is hands free.
Thanks for your interest and a great idea that I never would have thought
of until you asked for help.

I am now thinking of adding various scenario descriptions and configuration
settings in the project GitHub Wiki

Claude ...

Ebihen, 2017-10-15:
Hi Claude,

I'm satisfied with the pictures and time lapses your script is producing, but still I observed an anomaly. The filenames the script is producing do not correspond to the file creation times my NAS records.
So, for your info, if you want to look into it, here is an example from yesterdays list with file names and creation times:

mo-Ebihencam1-20171014-161931 - 16:29
mo-Ebihencam1-20171014-162933 - 16:39
mo-Ebihencam1-20171014-164058 - 16:41
mo-Ebihencam1-20171014-164100 - 16:51
mo-Ebihencam1-20171014-165101 - 17:01
mo-Ebihencam1-20171014-170119 - 17:01
mo-Ebihencam1-20171014-170120 - 17:11
mo-Ebihencam1-20171014-171121 - 17:21
mo-Ebihencam1-20171014-172824 - 17:28
mo-Ebihencam1-20171014-172825 - 17:38
mo-Ebihencam1-20171014-173826 - 17:48

I'm running the script with the setting I mentioned above: Quicktimelapse with a timer of 1190 and an interval of 600.

pageauc, 2017-10-15:
Yes the file time will be different than the date shown on the image.  The
filename is set just before the image is taken.  The camera then needs time
to warm up and take image plus post processing time for overlaying data
onto the image file.  The only way to get around this would be to post
process the image file to change the creation date but this would slow
things down during faster image sequences.

Let me know if this answers your question..

closed 2017-10-21 by pageauc