pageauc/pi-timolo #72: Increase opencv tracking to 720p for better quickpics?

brycedjohnson, 2017-10-22:
Awesome project!  I was thinking about putting my rasp pi outside and wanted to do a daily simple timelapse then found your project and was able to get up and running in no time!  Also decided to take advantage of all the other cool features you've implemented.  So Thanks!

I noticed on your speed-camera projects (https://github.com/pageauc/speed-camera) that you are running opencv at 720x1280 instead of 240x320.  I wanted to bump that up so the quickPics came out better.  I tried `motionTrackQuickPic = False` but it seemed that it kept missing when an object would go by quickly, since it had to switch between camera modes.  I am running the newest Raspberry pi 3 so hopefully it should be able to deal with it.

I increased that here: https://github.com/pageauc/pi-timolo/blob/master/source/pi-timolo.py#L111

As well as increased `motionTrackMinArea` and `motionTrackTrigLen` since that looks to be pixel based and 720 has a bunch more pixels.

Any problems or gotchas from doing that?

pageauc, 2017-10-22:
> speed-camera) that you are running opencv at 720x1280 instead of 240x320.
> timolo/blob/master/source/pi-timolo.py#L111

Hi Bryce

pi-timolo was not meant to be a speed camera although
motionTrackQuickPic=True does take a stream image similar to speed camera
to catch fast moving objects (but does not measure speed). This can result
in a slightly blurry image and lower resolution image since as you found,
it is based on a 240x320 stream that is enlarged using config.py variable
motionTrackQPBigger=2.5.  This keeps cpu arm processor cooler on older
RPI's.  The stream size is hard coded in pi-timolo.py at approx line 111
and 112 per below.  I could have added it to config.py but decided to leave
this setting coded in pi-timolo.py.

# Video Stream Settings for motion detection using opencv motion tracking
CAMERA_WIDTH = 320 # width of video stream
CAMERA_HEIGHT = 240 # height of video stream

You can change these values to eg 720x1280 and you will save a better
resolution stream image. Note the Image might still be a bit blurry for
fast moving objects while a non stream image will be better but will miss
fast moving objects.  Also increasing stream size may also cause some
overheating of the arm processor and can give some reliability issues (lock
up) depending on ambient temperature, environmental conditions like high
air temperatures and/or a case or enclosure with poor ventilation. Also can
slow/interrupt wifi connection and give slow console response.  Just give
it a try and monitor.  You can also slow the fps stream speed down a bit as
well.  Console command below will give you temperature.

/opt/vc/bin/vcgencmd measure_temp

BTW I have just released speed camera ver 6.9
https://github.com/pageauc/speed-camera that adds an optional (default)
recent folder feature with a specified max entries so you don't have to
switch to next subfolder if viewing images via web browser with
webserver.py running on RPI. (subfolders are used as the default).  This is
similar to pi-timolo recent folder feature.

Also I have run pi-timolo with larger stream image but did occasionally get
overheating on single core cpu so I decided to stay conservative.  I plan
on running speed camera all year as well as pi-timolo which has run several
years continuously without issue.  High summer temperatures can cause
overheating as well.

I would appreciate any data and experience you have on specific type of
RPI's.  I could add this information to the wiki and maybe increase
settings a bit.

Thanks for you Question ....
Claude ...

pageauc, 2017-10-22:
> speed-camera that adds an optional (default) recent folder feature with a

Reminder remember to change  motionTrackQPBigger=1

pageauc, 2017-10-22:
I just tried speed camera at 720p stream size on a single core RPI.  Since
opencv has to do more work due to bigger numpy array size, it essentially
slows the loop speed.  Results are better with a quad core due to threading
on another core.  Did not get very good track detection with single core
RPI but lighting is not ideal right now.  cpu was 100 % and temp was 60.0 C
at room  temp.  default config is 57.3C but does better job of tracking
motion with 240x320 stream size.

Currently have my two pi-timolo machines on long term timelapse so don't
want to fiddle with them now one is single core and the other is RPI3.  The
rest of them are packed since we will soon be heading south for the winter.

Claude ...

brycedjohnson, 2017-10-22:
Hi Claude,
Thanks for the great info.  I have set `motionTrackQPBigger=1` so I don't get jumbo QP pics.

It seems to be detecting great so far with only a few false detects because of the trees moving in the background due to the wind. 

I'll keep an eye on the temp.  I'm going to be putting it outside in a hopefully waterproof box so actually a little extra heat won't be too much of an issue, at least for this fall/winter.  

Thanks again,
Bryce

pageauc, 2017-10-23:
Also I forgot to mention that speed camera crops a detection section from the full stream image and passed that smaller array to opencv for motion tracking and thus improves performance on larger stream images.  pi-timolo on the other hand uses the full image that opencv has to process so quite a much larger array to process for motion tracking.

pageauc, 2017-10-23:
If pi-timolo is getting triggered by two trees near each other you might want to change some internal arbitrary settings coded into pi-timolo.py at approx line 117

TRACK_TRIG_LEN = motionTrackTrigLen  # Length of track to trigger speed photo
This variable is set by the variable in config.py and is the pixel distance an object must move in order to record motion.  One object moving (above the motionTrackMinArea) back and forth below this triglen value will not trigger motion but will start the track and then time out.  Unfortunately there can be two trees or objects moving that can cause a false positive.  See explanation below.

TRACK_TRIG_LEN_MIN = int(motionTrackTrigLen / 4)
TRACK_TRIG_LEN_MAX = int(CAMERA_HEIGHT / 2)  # Set max over triglen allowed half cam height  

TRACK_TRIG_LEN_MIN is used to verify the object has moved by at least a quarter of the trigger length.  This starts a timeout timer just in case the object moves the stops and never achieves the full trigger length within the timeout period.  This should be OK and should catch most reasonably slow objects but you can increase the timeout in case very slow moving objects are not getting caught.

TRACK_TRIG_LEN_MAX is a length that is half the camera height (an arbitrary setting on my part)  This will filter out objects moving on two location of the screen that are greater than this value.  I may change this to be Trigger length + 1/2 or 1/3 trigger length.  All this is saying is that if the object can be greater than the trigger length to report motion cannot be more that a specific value over the trigger length  in the original case it would be half the stream height which is quit a bit.
The purpose of this filtering is to avoid two arbitrary motion like trees moving on two sides of the stream image causing a false positive because it looks like the start and end points of the track was a moving object when in fact is was two separate motions.  I did this with the 240x320 resolution and it eliminated the situation I had with two trees triggering false positives.  If I get some time I will do some testing but it is hard to simulate every situation.  That is why I think it is nice for others to see the code and make changes if they can figure out the logic and why it is there instead of a black box where no one can see what the code is doing.
So like I said sometimes you wing it with internal filtering logic.  Comments appreciated.

Tuning Suggestions
If you know the contour size you are interested in like really big things then you can just set the motionTrackMinArea to a sqpx size appropriate to the object you are interested in.  default value is pretty small motionTrackMinArea = 100.  This would eliminate any smaller sqpx contours.  So there is some control there.  

You can also shorten the track length or adjust the trig len max setting but this may not catch fast moving objects.  Those are the two best settings that can be adjusted to reduce false positives.  Otherwise the logic would have to track each event and make sure is was close enough to the previous event but within a timeout period.  I was afraid that something moving could start an event and just keep moving without reaching a trigger length and miss something else in the mean time if there was no timeout time period.

Hope this explanation helps. 

Claude ...

pageauc, 2017-10-23:
I have tried to summarize discussion above into wiki.  Let me know your comments and results in weeding out false positives.
https://github.com/pageauc/pi-timolo/wiki/Variable-Tuning#reducing-motion-tracking-false-positives

brycedjohnson, 2017-10-23:
Wow Thanks!  Today is a particularly windy day with lots of trees in the background so I should have some good testing.  The spot where I want to mount it eventually will have a lot less tree movement, but this is good to be able to tune it out.

Was there a way to overlay the movement lines or areas on the motion capture picture?  I see that when I execute the py script directly I get a lot of useful debug information such as the locations and the track length.

pageauc, 2017-10-23:
I could add option for a restricted motion area similar to speed camera.
This would be a rectangle that you can specify that determines the
detection area.  Currently the whole image is the detection area.
Claude ...

closed 2017-10-27 by pageauc