pageauc/pi-timolo #73: pi-timolo won't start on RPi Zero W with Raspbian Stretch

DrGlas, 2017-10-26:
Hi!

First of all: Quite impressive project! It looks like it would fit my needs exactly, with many tweaking options. However, pi-timolo won't start. I have installed it using 

curl -L https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo-install.sh | bash

and everything went fine.
I then tried to start it using different methods:
1. the shell script ./pi-timolo.sh start
2. the python file ./pi-timolo.py
3. via the menubox.sh

No error messages except for when I try to start via menubox: I then get a message "Failed to start pi-timolo.py. Please investigate problem"

I have made sure I use the standard config.py file.
- Running a Raspberry Pi Zero W with camera module v2.
- Raspbian Stretch

I have also tested other time-lapse SW and verified my camera module is working.

Could it be incompatible with Stretch?

Best Regards,
Nils
Sweden

pageauc, 2017-10-26:
> source/pi-timolo-install.sh | bash

I am running rpi3 with stretch with no issues. If you run from console
./pi-timolo.py and verbose=True in config.py you should get some status
and/or error  messages on console if python fails. Are you  running stretch
lite?

DrGlas, 2017-10-26:
Interesting. I have Verbose=True, but when I run ./pi-timolo.py absolutely nothing happens.
I'm running the full version of Stretch.

closed 2017-10-26 by DrGlas

pageauc, 2017-10-27:
Very strange.  You should get a pi-timolo version  message as a bare
minimum before most of the python libraries are loaded.  Can  you run other
python programs.  Can you run pi-timolo directly from python

python ./pi-timolo.py

Even if there was a problem with the camera module you will still get a
python error message.  No message indicates something is wrong.  Could be a
problem with SD card, permissions or a bad install of some kind.  I am
having trouble imagining what would cause no message to be displayed.

Let me know if you can run python interpreter or another python program?

eg
python
print(__version__)
print("hello world")
quit()

Claude ...

pageauc, 2017-10-27:
sorry print(__version__) will not work.  Was thinking of opencv version
checking after importing cv2

Just see if python is installed OK and you can execute some python
programs.  Still think it might be SD card or problematic stretch install
of some kind.