pageauc/pi-timolo #74: Only few pictures taken during timelapse

DrGlas, 2017-10-27:
Hi!

I finally got my installation to work, but during timelapse it only takes a few pictures and then stops after an error message. I have pasted the message below.

Anyone?

I have attached the config file as well...

[config.txt](https://github.com/pageauc/pi-timolo/files/1420986/config.txt)

`2017-10-27 08:24:25 INFO     timolo     Scheduled Time Lapse Image - daymode=False
2017-10-27 08:24:29 INFO     takeNightImage TwilightThresh=62/90 shutSec=range camSleepSec=30 MaxISO=800 Size=1024x768
2017-10-27 08:24:31 INFO     takeNightImage FilePath media/timelapse/tl-NisseCam-1065.jpg
2017-10-27 08:24:31 INFO     postImageProcessing Next Counter=1066 /home/pi/pi-timolo/tl-pi-timolo.dat

2017-10-27 08:25:26 INFO     timolo     Scheduled Time Lapse Image - daymode=False
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x1ec8c10 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "./pi-timolo.py", line 1313, in <module>
    timolo()
  File "./pi-timolo.py", line 1088, in timolo
    takeNightImage(filename)
  File "./pi-timolo.py", line 712, in takeNightImage
    with picamera.PiCamera() as camera:
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python2.7/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python2.7/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python2.7/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
`

pageauc, 2017-10-27:
You did not say what you did to get things to work?

Here is what I found re first error message using google.
https://forums.pimoroni.com/t/camera-not-working-mmal-mmal-vc-component-enable-failed-to-enable-component-enospc/3789
Indicates a resource error due to possibly firmware issue.  Try updating
RPI firmware manually.  This is covered in the pi-timolo wiki here

https://github.com/pageauc/pi-timolo/wiki/How-to-Install-or-Upgrade#update-firmware-optional

    sudo rpi-update

Make sure to reboot after firmware update and let me know status

Claude ...

DrGlas, 2017-10-30:
>>You did not say what you did to get things to work?
Ah, sorry. I simply reinstalled pi-timolo and rebooted. 

The FW is updated using rpi-update but the error message persists: I tried to google the entire message and get different suggestions, e.g.:

[https://raspberrypi.stackexchange.com/questions/51406/cannot-connect-to-picamera-when-using-it-with-flask](url)

[https://github.com/raspberrypi/firmware/issues/680](url)

Nils

DrGlas, 2017-11-06:
So I solved this by re-writing the Raspbian image to my SD card and re-installed pi-timolo. I suspect the reason for the problems I had, was that I prior to installing pi-timolo had installed and tried a couple of other timelapse solutions, and that the problem was related to a conflict...just a theory. Everything seems to work fine now!

Great piece of sw!
N

closed 2017-11-06 by DrGlas

pageauc, 2017-11-06:
Glad you got things working.  If you post a project online I would love to get a link.
Claude ..