pageauc/pi-timolo #76: Brightness value increasing/decreasing in a 30min interval

marciano1, 2017-11-15:
I am recording 10 secs of video each time sensor detects motion.  At the same time (this is by default) a picture is taken and stored in timelapses directory.
For an unknown reason I found an interval where10 photos were taken increasing exposure and then decreasing it without any reason.  No lights turned on or off, no ambient light variations.
"Brightness Value" metadata goes from 1.2 to 6.75 (very overexposed), Shutter Speed(and Exposure Time) vary from 0.4 to 1.2
[image]
[image]

pageauc, 2017-11-15:
Can you post a copy of your config.py.  Not sure what settings you are using.  
Also
What lighting conditions are being used.  eg room lighting with/without windows?
Problem can be caused by not enough time for camera to warm up.  I have tried to minimize camera warm up and switching from one mode to another but this can delay taking an image.  You can turn off the timelapse images in config.py if you only want video.  Normal timelapse images will use low light transitions.  Most of these use camera automatic features but below a certain light level a manual algorithm has to be used that will vary the exposure time. 

Please note I have just posted an updated pi-timolo.py version 7.93 to detect camera and fix bug with detecting for python3 cv2 library.  You can upgrade using command below in logged in ssh or console session

    wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

You can attach the config.py to this github issue.

Thanks Claude ...

marciano1, 2017-11-15:
[config.py.zip](https://github.com/pageauc/pi-timolo/files/1475955/config.py.zip)
There's a window I understand could impact to exposure leaving shadow areas darker
[image] This is a screenshot of motion dir.
Look at the first and second line where exposure is increased gradually and also see the other thumbnails. 
Is good to have pictures apart of the footages. 
What I have to look at is why every hour after the last motion detection a start video is triggered (during the night)

pageauc, 2017-11-15:
There is a motion force variable in config.py per

motionForce = 3600          # default= 3600 seconds (1 hr) Off=0  Force an image if no Motion Detected in specified seconds.

Set value to 0 per comment to disable this default feature.

pageauc, 2018-02-03:
Please Note that pi-timolo is now at release 10.62.  There are many changes including implementation of rclone for viewing media on a remote storage service, plugins for custom configurations, Using opencv for motion tracking and better low light handling.  I will close this issue.  If you still have a problem reopen this issue and I will be more than happy to help you resolve.  Claude ....

marciano1, 2018-02-03:
Thank you very much Claude!

Daniel Stonek
www.stonek.com
www.UruguayDesdeLoAlto.com
www.bancodefotos.com

cpforbes, 2018-04-12:
Could this be related to: https://github.com/waveform80/picamera/issues/475

pageauc, 2018-04-12:
> Could this be related to: waveform80/picamera#475

Yes I believe this is the issue.  Code is very similar.

mklein9, 2018-04-13:
I'm the OP of the referenced issue [waveform80/picamera#475](https://github.com/waveform80/picamera/issues/475).  I tried running the tiny Python program (in the second comment) for over 6 hours and do not see the exposure oscillating.  It simply drops immediately after starting to the lowest possible exposure speed (9 us), and stays there.

closed 2018-10-18 by pageauc