pageauc/pi-timolo #77: Using sudo in scripts

ragnarjw, 2017-11-19:
Many thanks for a useful application!

I encountered an issue with v 4.10 of makevideo.sh. You use sudo to remove files and directories, but as all files and directories would be owned by the same user, this seems superfluous and causes the script to halt to prompt for a password. Nothing bad seems to happen if those sudos are removed.

pageauc, 2017-11-20:
Thanks.  pi user has wheel rights so should not normally prompt for password.  I wrote this when there was a chance that some files were owned by root.  If users run pi-timolo per is current setup then you are right sudo should not be required.  I have updated makevideo.sh github repo to ver 4.20
Thanks for your issue.

Also FYI
I am currently working on plugins that are python import files that contain config.py variable settings to perform specific tasks.  I currently have about 12 plugins but still working on completing and testing.
Claude

ragnarjw, 2017-11-20:
Thanks Claude. I am not running this as the pi user, so perhaps that is why it showed up. Actually, there is an instance of sudo in the startup/shutdown script as well, you might want to consider changing that as well.

 I have just set up an installation using your application and will report back if I have suggestions. At the moment I am trying to figure out your settings for twilight and how you tweak  the camera settings. Is there some documentation you could suggest to learn more about those?

It will be interesting to see the plugin scheme, will that be e.g. to run motion tracking as a plugin?

Very off topic, but I looked at your other projects, and very happy to see a green NB featuring in the speed camera youtube video :-) I've driven an NC for a few years, brilliant car!

pageauc, 2017-11-20:
plug ins is just a second layer of config.py variable import.  The overrides specific config.py variables but the benefit is you can leave a stock config.py and apply pre done plugin to accomplish a specific task.  For example One user had a multi day/session shop project.  He wanted a project timelapse but only when he was working in the shop.  The variables setup only needs to be done at the start of the project so can be unattended.  Below is the sample plugin to do this.

"""
Senario 

You have a multi day/session work shop project.
You want a timelapse series whenever you work on the project
You do Not want to manage the camera each time you work in the shop

The pi-timolo settings below will monitor area using motion tracking.
When motion tracking is triggered, the camera will take a sequence of
time lapse images for motionQuickTLTimer seconds duration and 
motionQuickTLInterval seconds delay between images.

This means you can setup the camera once take and not have to manage it for a long workshop timelapse 
since it is only activated when you enter the shop.  If you are in the shop longer and the first
timelapse sequence period ends, then the next motion in the shop will trigger another timelapse
sequence. filename will be plugin/timelapse_shop_project.py

"""
# Customize settings below to suit your project needs

motionQuickTLTimer = 3600     # On motion do timelapse sequence for 60 minutes
motionQuickTLInterval =  300  # Take image every 5 minutes until timer runs out

imageNamePrefix = 'TL-'     # default= 'TL-' for all image file names. Eg garage-
motionPrefix = "shop-"         # default= "shop-" Prefix for all Motion Detect images
motionDir = "media/shop_TL"   # default= "media/shop_TL"  Folder Path for Motion Detect Image Storage

imageWidth = 1920             # default= 1920 Full Size Image Width in px
imageHeight = 1080            # default= 1080  Full Size Image Height in px
imageVFlip = True             # default= False True Flips image Vertically
imageHFlip = True             # default= False True Flips image Horizontally
showDateOnImage = False       # default= True False=Do Not display date/time text on images

motionNumOn = True            # default= True  True=filenames by sequenced Number  False=filenames by date/time
motionNumRecycle = False      # default= True when NumMax reached restart at NumStart instead of exiting
motionNumStart = 10000        # default= 10000 Start 0f motion number sequence
motionNumMax  = 0             # default= 0 Max number of motion images desired. 0=Continuous

# Do not change these settings.
motionTrackOn = True          # Turn on Motion Tracking
motionQuickTLOn = True        # Turn on motion timelapse sequence mode
motionVideoOn = False         # Take images not video
motionTrackQuickPic = False   # Turn off quick picture from video stream
timelapseOn = False           # Turn off normal time lapse mode so only motion mode used.
videoRepeatOn = False         # Turn on Video Repeat Mode IMPORTANT Overrides timelapse and motion
motionForce = 0               # Do not force motion image if no motion for a period of time

closed 2017-12-01 by pageauc