pageauc/pi-timolo #82: Timelape: Take the first photo immediately

tibaultbordeaux, 2018-01-09:
Hi there.

Thanks a lot for this great piece of software, genius man!
This a huge time saver, thanks again!

My issue:
pi-timolo takes the first photo _after_ the first loop of the Timer.
I have set the Timer on 86400 seconds since I want to take one photo every 24h.

Therefore I start pi-timolo.sh via cron exactly at 12 o'clock (and check every day at the same time, if it's still running).
But: Now it takes another 24h till the first photo is taken.

How would i (you?) change this?

Thanks in advance

thibault

pageauc, 2018-01-09:
Set up pi-timolo for 24 hrs between images per config.py settings below with whatever image, text values you want. 

    motionTrackOn = True
    timelapseOn = True
    timelapseTimer = 86400

pi-timolo will look after the interval between images eg 24 hours

Set up crontab to start pi-timolo the day before you want the timelapse cycle to begin since pi-timolo will start the timelapse cycle at the end of the timer interval.  Then run pi-timolo.sh start

    00 00 * * * su pi -c "/home/pi/pi-timolo/pi-timolo.sh start"

This will run run every day at midnight but pi-timolo.sh will not restart pi-timolo.py if it is already running so as long as pi-timolo continues to run in the background you will get an image at midnight since pi-timolo.sh will detect that it is already running.  If pi-timolo does stop for some reason then there will be a delay of 24 hours before first image is taken.  I will look at changing the logic to start take an initial timelapse image at the start of the interval.  Please note the camera startup time takes a little while so you will probably have to set cron to start a little before midnight.
I have timelapse cameras that run continuously for many months without problem so this should work.

I will let you know when I change logic to start an initial timelapse image and maybe implement a timelapse initial start time to force start of first image at a particular time as long as the RPI clock is accurate.

Claude ..

tibaultbordeaux, 2018-01-10:
Thanks Claude,

this is exactly what I've done. (Except the motionTrackOn = True, Typo?)

Another approach of mine would be to shut down the Pi after taking a picture at 12 o'clock.
Since all I want to have is one picture a day, the Pi does not have to run 24/7.
I then would boot up at, let's say,  11 o'clock by using a socket timer switch.

In this approach again it would be helpful to have the option to take an initial photo right after starting pi-timolo.sh 

And your idea of implementing a start time for the first image is even more great!

thx!

pageauc, 2018-01-10:
I am roughing out the code for setting a time and or date/time to start timelapse based on variable settings.  If the date is in the past the timelapse would start on the specified hour/minute.  This would allow a restart and still have the same start time.

I was also thinking it would be helpful if timelapse timer could be set using names for values like minutes, hours, days, etc to avoid large seconds values but just a thought.

pageauc, 2018-01-10:
pi-timolo.py release ver 9.2  now takes an image at start of sequence and timelapse will continue per timer settings.  This should solve your original issue.  I am also working on setting a scheduled date/time for start of timelapse sequence and will issue when ready.  

You can do a quick upgrade per

    wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

closed 2018-01-10 by pageauc

pageauc, 2018-01-13:
I have released pi-timolo ver 10.0 that now includes scheduled StartAt variables for motion, timelapse and video repeat.  See Wiki for details [How to Schedule Motion, Timelapse or VideoRepeat](https://github.com/pageauc/pi-timolo/wiki/How-to-Schedule-Motion,-Timelapse-or-VideoRepeat)