pageauc/pi-timolo #84: New Release 10.0 Adds optional Scheduled Start to Motion, Timelapse and Video Repeat

pageauc, 2018-01-13:
I have implemented an optional scheduled start feature.   
or details see [How to Schedule Motion, Timelapse or Video] (https://github.com/pageauc/pi-timolo/wiki/How-to-Schedule-Motion,-Timelapse-or-VideoRepeat)

malialis, 2018-01-13:
Hello.

Thank you for all your amazing work. Since I upgraded i now get the following error when I try run it.

Traceback (most recent call last):
  File "./pi-timolo.py", line 173, in <module>
    from dateutil.parser import parse
ImportError: No module named dateutil.parser

Any ideas what I could of done wrong?

pageauc, 2018-01-13:
Funny I tested this on several machines and it ran fine.  I have added install of dateutil.parser to the pi-timolo-install.sh script.  Rerun the curl upgrade and test again.  Let me know if this fixes the problem
Thanks for your feedback.
Claude

malialis, 2018-01-14:
Hello.

Thank you. That fixed it. It is now working again.

Thank you so much.

pageauc, 2018-01-14:
You are welcome.  I appreciate your feedback.
Claude ...

malialis, 2018-01-14:
Hello.

Sorry, once again. It is working kinda. It takes a photo right away in timelapse. Then nothing. I have even reimaged my rpi. Fresh install. Fully updated. Reinstalled Pi-TiMoLo and it runs no errors. After taking one photo it sits there 

2018-01-14 11:12:07 INFO     postImageProcessing Next Counter=1001 /home/pi/pi-timolo/tl-pi-timolo.dat
2018-01-14 11:12:07 INFO     timolo     Restart PiVideoStream ....

I have tried timelapse at various time lengths. and nothing. Any ideas? 

Sorry for bugging. And thank you so much for all your help.

pageauc, 2018-01-14:
My Stupid Error.  Accidentally turned off timelapse after initial image and
did not notice during testing since I did get initial image and default is
5 minute interval.  Just impatient.
You can. Download pi-timolo.py ver 10.1

wget -O pi-timolo.py
https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

Thanks for your help.  greatly appreciated.  Let me know if this works OK
now.

Claude ...

pageauc, 2018-01-15:
> source/pi-timolo.py

Sorry.  I think I finally have the issue resolved.  Problem was with allow
a first image and not mess up the other restrictions.

Download release 10.2

wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/
source/pi-timolo.py

Had a rush day
Thanks for your patience.

Claude ...

malialis, 2018-01-15:
Hello.

Updated and it is working as intended. Thank you so much.

Most appreciated.

tibaultbordeaux, 2018-01-20:
Just a quick - but HUGE - thank you for this fast implementation of my feature request.
This is amazing!

And as a side note for other users: If I'm using plugins like TLlong.py I have to manually add 

timelapseStartAt = "12:00:00" 

 in the respective file.

pageauc, 2018-01-20:
FYI

I have added StartAt="" to several plugins where I thought it made sense.
Upgrade will not overwrite plugins unless you rename or delete one but new
installs will see the variable.  User would have to then set desired
date/time.

Also I am quite pleased with the new changes to the webserver that uses a <
BACK button and no longer opens selected folder in a separate browser tab.
It took me while to figure out but only took two lines of code.  I also
added a bottom < BACK link on the bottom if there are more than 35 entries
in the right hand listing.  Two more lines of code..

Claude ....

malialis, 2018-01-24:
Hello.

Thank you for this awesome resource it is amazing. I like the StartAt feature.

Was wondering if you could add a EndAt feature to it. If possible. This way I do not have to manually shut the Pi down durring the evenings. I could have it start timelapse at 8am and stop at 7 pm.

If possible.

Thank you again.

pageauc, 2018-01-24:
Can you post both your requests as a separate GitHub issues.  Start At only
kicks in once during initialization and looks for date and/or time past.
If date is past then time is used only if pi-timolo.py is restarted
otherwise task continues.  It might take me a while to figure out how to
implement.  There is already settings for

noNightShots = False # default= False True=No Night Images (Motion or
Timelapse)
noDayShots = False # default= False True=No Day Images (Motion or Timelapse)

These allow turning off day or night (not both or nothing is done and error
message will remind you not to turn on both)

Thanks Claude ...

closed 2018-01-25 by pageauc