pageauc/pi-timolo #86: Unable to complete installation

DavidPower, 2018-01-24:
I'm attempting to install pi-timolo on a Pi Zero W running DietPi (i.e. a minimal Raspian image).

All goes well with the curl install until I'm prompted as follows: 

```
Need to get 70.1 MB of archives.
After this operation, 192 MB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.
```
"Abort." is not something I typed. It shows up automatically and aborts the installation.

Any thoughts on how I might get around this?

pageauc, 2018-01-24:
I am thinking there is maybe a low disk space, library problem or communication issue.  The install sends -y to the apt-get commands so that is why the continue said y automatically.  Then the abort was due to some reason as yet not known.  Try manually installing libraries per below and let me know where it aborts. Also check free disk space.  Let me know results.  Claude ...

    df -h

    sudo apt-get install python-picamera python3-picamera python-imaging
    sudo apt-get install dos2unix python-pyexiv2 libav-tools
    sudo apt-get install  python-scipy 
    sudo apt-get install  gpac   
    sudo apt-get install  fonts-freefont-ttf 
    sudo apt-get install  python-opencv

before doing this do

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get autoremove
    sudo apt-get clean

You should be able to cut and paste apt-get commands individually and see where it is failing otherwise we can try something else.

DavidPower, 2018-01-25:
Claude,

It wasn't a memory issue. I had 28GB available at the time I reported the error. Oddly...

I installed each component manually per your suggestions above... and they all installed perfectly!

However, upon running pi-tomolo.py for the first time, I got the following Python error:

```
pi-timolo.py ver 10.2  written by Claude Pageau
INFO  - Initializing ....
INFO  - Pi Camera Module is Enabled and Connected supported=1 detected=1
INFO  - Import Configuration Variables from File /root/pi-timolo/config.py
INFO  - No Plugins Enabled per pluginEnable=False
INFO  - Logging to Console per Variable verbose=True
INFO  - Loading Python Libraries Please Wait ...
Traceback (most recent call last):
  File "pi-timolo.py", line 173, in <module>
    from dateutil.parser import parse
ImportError: No module named dateutil.parser
```
So I installed the dateutil package and everything's now working as it should.

I'll have to dig into the documentation and figure out the finer points but I've confirmed the app is capturing images and I'm able to see them in the web browser from your server app.

I'm still stumped by the auto-abort. But... I won't lose sleep over it.

Thanks for your help.

pageauc, 2018-01-25:
Seems strange that the manual install worked but the curl bash install
failed?  Do you think it was a communications problem or maybe library
server was off line for a moment..  Did you do the apt-get cleanup and
autoremove.  Can you successfully run the the curl install again or do a
menubox.sh, UPGRADE menu pick.  Would be interested.

Ran the curl install on two systems from scratch without issue then did an
upgrade that worked OK.

Anyway there is good wiki that can walk you through most things.  The
latest version has plugins and I really like them since you and easily
switch configurations.  The plugins are not overwritten during an upgrade.
I also recently changed the webserver so it now has < BACK button and does
not open multiple browser tabs.

Claude ...

closed 2018-01-25 by pageauc

NelsonPJ, 2018-03-14:
Just found the same issue... fresh install of Raspbian Stretch Lite (2018-03-13) on a Pi Zero W with a 32 gig SD card, then used raspi-config to change my localization settings, enabled WiFi, set a password, etc.  Ran sudo apt-get update/upgrade (but no updates found since the Stretch release is so new, I guess)

Then I ran the curl install and it failed right at the end - it didn't give me a chance to type "Y" or "N".  I did the same install on an RPi3 running Stretch a week or two ago and had no issues (although it wasn't a new install)

Here's my log in case it's useful... I'll delete what's been installed so far and do a manual installation.

pi@zerocam:~ $ curl -L https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo-install.sh | bash 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  9681  100  9681    0     0   9482      0  0:00:01  0:00:01 --:--:-- 73900
INFO  : New pi-timolo Install
INFO  : pi-timolo Folder Created

-------------------------------------------------------------
INFO  : bash 10.2  written by Claude Pageau
        New Install from https://github.com/pageauc/pi-timolo
-------------------------------------------------------------

config.py                           100%[==================================================================>]  11.88K  --.-KB/s    in 0.03s   
menubox.sh                          100%[==================================================================>]  21.24K  --.-KB/s    in 0.02s   
pi-timolo.py                        100%[==================================================================>]  80.99K  --.-KB/s    in 0.09s   
pi-timolo.sh                        100%[==================================================================>]   2.03K  --.-KB/s    in 0.001s  
webserver.py                        100%[==================================================================>]  10.39K  --.-KB/s    in 0.004s  
webserver.sh                        100%[==================================================================>]   1.33K  --.-KB/s    in 0s      
watch-app.sh                        100%[==================================================================>]  12.98K  --.-KB/s    in 0.01s   
convid.sh                           100%[==================================================================>]  12.03K  --.-KB/s    in 0.02s   
makevideo.sh                        100%[==================================================================>]  11.61K  --.-KB/s    in 0.004s  
video.conf                          100%[==================================================================>]   2.79K  --.-KB/s    in 0.002s  
mvleavelast.sh                      100%[==================================================================>]   1.90K  --.-KB/s    in 0.001s  
remote-run.sh                       100%[==================================================================>]     638  --.-KB/s    in 0.001s  
config.py.new                       100%[==================================================================>]  11.88K  --.-KB/s    in 0.01s   
watch-app-new.sh                    100%[==================================================================>]  12.98K  --.-KB/s    in 0.01s   
video.conf.new                      100%[==================================================================>]   2.79K  --.-KB/s    in 0.002s  
Readme.md                           100%[==================================================================>]   6.85K  --.-KB/s    in 0.002s  
media/webserver.txt                 100%[==================================================================>]   1.07K  --.-KB/s    in 0s      
rclone-test.sh                      100%[==================================================================>]   4.32K  --.-KB/s    in 0.001s  
INFO  : New Install Check/Install pi-timolo/plugins    Wait ...
__init__.py                             [ <=>                                                               ]       0  --.-KB/s    in 0s      
dashcam.py                          100%[==================================================================>]   2.39K  --.-KB/s    in 0.001s  
secfast.py                          100%[==================================================================>]   3.79K  --.-KB/s    in 0.002s  
secQTL.py                           100%[==================================================================>]   3.17K  --.-KB/s    in 0.002s  
secstill.py                         100%[==================================================================>]   3.38K  --.-KB/s    in 0.001s  
secvid.py                           100%[==================================================================>]   3.47K  --.-KB/s    in 0.006s  
strmvid.py                          100%[==================================================================>]   3.86K  --.-KB/s    in 0.003s  
shopcam.py                          100%[==================================================================>]   2.73K  --.-KB/s    in 0.001s  
slowmo.py                           100%[==================================================================>]   3.51K  --.-KB/s    in 0.001s  
TLlong.py                           100%[==================================================================>]   4.32K  --.-KB/s    in 0.002s  
[... cut ...]
  python-all python-all-dev python-cffi-backend python-crypto python-cryptography python-dbus python-dev python-enum34 python-gi python-idna
  python-ipaddress python-keyring python-keyrings.alt python-pip python-pip-whl python-pkg-resources python-pyasn1 python-secretstorage
  python-setuptools python-six python-wheel python-xdg python2.7-dev
0 upgraded, 31 newly installed, 0 to remove and 0 not upgraded.
Need to get 32.8 MB of archives.
After this operation, 51.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.

NelsonPJ, 2018-03-14:
Update - did these commands as Claude typed above to install dependent packages

```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get autoremove
sudo apt-get clean
df -h
sudo apt-get install python-picamera python3-picamera python-imaging
sudo apt-get install dos2unix python-pyexiv2 libav-tools
sudo apt-get install  python-scipy 
sudo apt-get install  gpac   
sudo apt-get install  fonts-freefont-ttf 
sudo apt-get install  python-opencv
```

Then I ran the curl install command again, which seemed to download all the files from github, but still crashed with an "abort" when trying to install packages, not letting me type a "Y" or "N"  - here's the final part of the output:

```
2018/03/14 17:19:19 NOTICE: Config file "/home/pi/.config/rclone/rclone.conf" not found - using defaults
rclone v1.39
- os/arch: linux/arm
- go version: go1.9.2
2018/03/14 17:19:19 NOTICE: Config file "/home/pi/.config/rclone/rclone.conf" not found - using defaults
INFO  : New Install Install pi-timolo Dependencies Wait ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  gir1.2-glib-2.0 libdbus-glib-1-2 libgirepository-1.0-1 libpython-all-dev python-all python-all-dev python-cffi-backend python-crypto
  python-cryptography python-dbus python-enum34 python-gi python-idna python-ipaddress python-keyring python-keyrings.alt python-pip-whl
  python-pkg-resources python-pyasn1 python-secretstorage python-setuptools python-six python-wheel python-xdg
Suggested packages:
  python-crypto-dbg python-crypto-doc python-cryptography-doc python-cryptography-vectors python-dbus-dbg python-dbus-doc python-enum34-doc
  python-gi-cairo gnome-keyring libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-fs python-gdata python-kde4 python-keyczar doc-base
  python-secretstorage-doc python-setuptools-doc
The following NEW packages will be installed:
  gir1.2-glib-2.0 libdbus-glib-1-2 libgirepository-1.0-1 libpython-all-dev python-all python-all-dev python-cffi-backend python-crypto
  python-cryptography python-dbus python-enum34 python-gi python-idna python-ipaddress python-keyring python-keyrings.alt python-pip
  python-pip-whl python-pkg-resources python-pyasn1 python-secretstorage python-setuptools python-six python-wheel python-xdg
0 upgraded, 25 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,930 kB of archives.
After this operation, 11.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Abort.
```

I'll search through the install script and install any other packages that might be required...

NelsonPJ, 2018-03-14:
Going through the install script, these packages still needed installation:

```
sudo apt-get install python-pip
sudo pip install python-dateutil
sudo apt-get install -yq pandoc
```

Looking back at my logs when the curl script crashed, I think it crashes when trying to install python-pip... is it because there's no "yq" in the statement?

pageauc, 2018-03-14:
I have update the install to add -yp to python-pip install.  Also moved these dependencies to install near the end.  pip is used for dateutil install.  Give the curl command another try since I have updated the GitHub repo   

Let me know if this resolves issue

NelsonPJ, 2018-03-14:
I confirmed that your new script works. I removed the entire pi-timolo directory and re-ran the curl install script.  The script downloaded everything from github, and then went through all the dependencies (and nothing needed to be updated obviously).

Thanks!

pageauc, 2018-03-15:
I am sure the problem was due to the missing -yq with raspbian lite.  I believe lite does not have python-pip installed by default but the full version does.  I have run the install many times but always on the full version of raspbian.  I am thinking that the interaction of the curl script and manually trying to respond to the python-pip install prompt caused the failure.  python-pip is used for the dateutil library install.  This library was added recently for the scheduled start feature and allows reading date/time entries to be read in different date/time formats.  It also allows defaulting to the current day if the time only is input.

Thanks for your feedback and perseverance.  Your hard work is appreciated.  Funny how a little missing parameter can cause grief.   
Please let me know if you have any more problems
Claude ...

pageauc, 2018-03-15:
I have confirmed that pip is Not installed on jessie lite but is installed by default on full version
See pip section on link below
https://www.raspberrypi.org/documentation/linux/software/python.md