pageauc/pi-timolo #87: Exit on timelapseExitSec reached

tibaultbordeaux, 2018-01-26:
Hi Claude!

Here is another question:

When using timelapseExitSec and the given timespan is reached pi-timolo keeps printing an INFO about that - instead of printing the INFO once and then instead exiting at whole.

Is this by desing?

I would prefer pi-timolo to exit, since it does not have to do anything more when timelpaseExitSec is reached.

(And not related: In your comments of config.py your declare the JPEG quality with 1 = highest and 40 = lowest quality. I think it's the other way round)

Thank you so much for your efforts!

Tibault

pageauc, 2018-01-26:
By design I did not want to exit since you may be running motion at the
same time so I just put a reminder in about the camera timelapse being
disabled due to timelapseExitSec.  This does not exit out of pi-timolo.py.
It is possible to put a check in to see if motion is running and exit
pi-timolo but this might be confusing to a user so I just put a reminder in
whenever a timelapse schedule is called.  It might be annoying if you are
running with verbose on but otherwise if running as a background task and
verbose off then it reminder would not show.  I will take a look at message
again and maybe only display occasionally maybe every 10 minutes or per
variable.

Image quality was based on earlier documentation but it appears things may
have changed.  Yes quality is between 0-100 with 0 max compression (lowest
quality) and 100 Min Compression ( highest quality).   I will change the
wording to make this clear.  I set to 10 to get better compression (lower
quality) since I upload images.  Note the scale is not linear or percentage
so values can be misleading.  I believe default picamera quality value is
85.  So it is more of a compression scale with 0 being highest compression
and lowest quality.
I will try to make this clearer.

I will do a bit more testing

Thanks for your feedback.

pageauc, 2018-01-26:
Found this in the docs
http://picamera.readthedocs.io/en/release-1.10/api_camera.html#picamera.camera.PiCamera.start_recording

So 0 = default quality setting

1-100 Settings for quality where higher value is lower compression and
better quality

Note the reference to the video encoding quality between 10 and 40 where
lower value is higher quality (opposite to jpeg settings).

All encoded formats accept the following additional options:

   - *bitrate* - The bitrate at which video will be encoded. Defaults to
   17000000 (17Mbps) if not specified. The maximum value is 25000000 (25Mbps).
   Bitrate 0 indicates the encoder should not use bitrate control (the encoder
   is limited by the quality only).
   - *quality* - Specifies the quality that the encoder should attempt to
   maintain. For the 'h264' format, use values between 10 and 40 where 10
   is extremely high quality, and 40 is extremely low (20-25 is usually a
   reasonable range for H.264 encoding). For the mjpegformat, use JPEG
   quality values between 1 and 100 (where higher values are higher quality).
   Quality 0 is special and seems to be a “reasonable quality” default.
   - *quantization* - Deprecated alias for *quality*.

pageauc, 2018-01-26:
> 10/api_camera.html#picamera.camera.PiCamera.start_recording

Did a minor update to config.py and pi-timolo to clarify settings.  0 value
is not valid in picamera but I have built in range checking and if zero is
used then I set value to 85 and if over 100 I set value to 100.  I have set
the new config.py imageJpegQuality = 95

still looking at timelpaseExitSec.  Normally when verbose is set to True
there is a message for every timelapse image taken even if it is a short
duration eg 10 sec timer value.  All the INFO message is saying is that the
timelapse schedule arrived but timelapseExitSec is suppressing taking the
image.  This is by design.
Anyway still under review on design but thinking of leaving it in unless
you have a better option. Let me know

Your feedback is greatly appreciated.

Claude ...

tibaultbordeaux, 2018-01-27:
Hi Claude,

thanks again for your fast response and clarification.
Great to see the improvement regarding the quality settings.

Only thing I do not get: "...but this might be confusing to a user... "

If timelapseExitSec is reached AND motion == False 
there is nothing to do anymore for pi-timolo.py, is it?

You could send an INFO (even by overriding verbose == false!) like
"Hey guy, your job is done, I'm exiting. Please restart me"
and then exit.

I do not see the confusion - but of course I'm not into the script like you, you probably see more use cases than me.

Thanks for listening!

Tibault

pageauc, 2018-01-27:
timelapseExitSec is not something I use very often.  I will do some testing
again.  The program used to exit at quite a few spots but other users
wanted to avoid exiting program so now there are only a few critical error
points that exit.

1 If camera not connected and enabled
2 Can't find plugin directory
3 Plugin Name not found
4 Could not import cv2
5 Could not import camera module
6 picamera module already in use
7 motion and timelapse both turned off
8 If image counter reached max and recycle is not enabled
9 User pressed ctrl-c

Most of these are checks  before processing loop is started.

pageauc, 2018-01-28:
I have just released pi-timolo.py version 10.4
This fixes the exit condition when timelapse and/or motion NumRecycle=False and NumMax exceeded or timelapseExitSec is exceeded.  Only one log suppression message is displayed for each condition so warning is not repeated. 
In the case when both motion and timelapse are suppressed then a final warning is displayed and the program is exited.  If program is run in the background users should turn on 

    verbose=True

and then run 

    cd~/pi-timolo
    ./pi-timolo.py

The messages will be shown to indicate one or both are suppressed. 
I have tested this and seems to work OK but would appreciate any feedback on release 10.4

You can update pi-timolo.py via curl upgrade or 

    wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

Thanks for your feedback
Claude ...

pageauc, 2018-01-28:
I have just released pi-timolo.py version 10.44
This fixes the exit condition when timelapse and/or motion NumRecycle=False and NumMax exceeded or timelapseExitSec is exceeded.  Only one log suppression message is displayed for each condition so warning is not repeated. 
In the case when both motion and timelapse are suppressed then a final warning is displayed and the program is exited.  If program is run in the background users should turn on 

    verbose=True

and then run 

    cd~/pi-timolo
    ./pi-timolo.py

The messages will be shown to indicate one or both are suppressed. 
I have tested this and seems to work OK but would appreciate any feedback on release 10.4

You can update pi-timolo.py via curl upgrade or

pageauc, 2018-01-28:
I have done some motion tracking fine tuning.  Reduced the timeout to fraction of a second between track movements.  Now times out quicker and seems to work better.  Also improved tracking info if turned on.  You should use the new config.py or update the  motionTrackTimeOut=0.3
I expect this could be reduced more if running on RPI3 but this value should work OK.  If motion stops for more than motionTrackTimeout without reaching the motionTrackTrigLen then the track is reset
Claude ...

tibaultbordeaux, 2018-01-29:
I've upgraded via menubox.sh and did a mv config.py.new config.py.

Unfortunately I got a ValueError after  timelapseExitSec exceeded:
`unsupported format character  'p' (0x70) at index 19`

pageauc, 2018-01-29:
Yep.  Stupid error that crept in.  Put variable name instead of %s.  Must
have crept in somehow since I did do testing.

Release 10.45 should fix this.  Appreciate your feedback.  I think we are
making progress.

You can upgrade using

    wget -O pi-timolo.py
https://raw.github.com/pageauc/pi-timolo/master/source/pi-timolo.py

Thanks
Claude ...

pageauc, 2018-01-29:
> source/pi-timolo.py

FYI

release 10.46 now includes a timelapseExitSec Coundown if >0 to timelapse
logging message.  Should make it easier to track how close value is to
trigger point.

Let me know how testing goes
Thanks Claude ...

tibaultbordeaux, 2018-01-29:
Hi, all is working great now! Countdown is a helpful addition, thank you for for this add on.

BTW, by following the logging messages more closely I was wondering if there is room for time saving in the process of taking a picture. I.E. between taking the picture and printing text to the picture there seems to be one minute of silence where nothing happens. Seems unimportant but my Pi is living on batteries and I try to keep the uptime as short as possible.

But hey, You've coded so hard the last days. Take a rest. :-) This one can wait.

I'm impressed by your engagement!

pageauc, 2018-01-29:
Time delay would be for timelapse timer I believe.  There should be no
delay between taking the image and adding the text.  Not sure If you are
talking about adding some sort of sleep time between timelapses or not.
Please provide details and maybe an example.  Not sure if you are talking
about sleep time that saves power.  The RPI conserves energy quite well.
If you are in timelapse only then I turn off motion tracking.

Try this

Download htop

    sudo apt-get install htop

setup first run with config.py settings

    verbose=False
    motionTrackOn=True
    timelapseOn = False

Measure cpu activity using htop.  You should see quite a bit of cpu
activity due to opencv.
Second run

    verbose=False
    motionTrackOn=False
    timelapseOn = True

You should see a lot less cpu activity.  This represents less energy on
RPI.  You could turn off webserver and retry tests.

Your feedback is appreciated
Thanks Claude ...

tibaultbordeaux, 2018-01-31:
Please look on the timecodes.

From starting pi-timolo.py until taking the first picture it takes 24 seconds.
Then it takes another 61 seconds until the text is written to the image.

`pi-timolo.py ver 10.46  written by Claude Pageau
INFO  - Initializing ....
INFO  - Pi Camera Module is Enabled and Connected supported=1 detected=1
INFO  - Import Configuration Variables from File /home/pi/pi-timolo/config.py
INFO  - No Plugins Enabled per pluginEnable=False
INFO  - Logging to Console per Variable verbose=True
INFO  - Loading Python Libraries Please Wait ...
2018-01-31 18:54:55 INFO     <module>   Testing if Pi Camera is in Use
2018-01-31 18:54:57 INFO     <module>   Pi Camera is Available.
2018-01-31 18:54:57 INFO     <module>   Start pi-timolo per /home/pi/pi-timolo/config.py Settings
2018-01-31 18:55:07 WARNING  timolo     Motion Tracking is Suppressed per motionTrackOn=False
-------------------------------------- Settings ----------------------------------------------
Config File .. configName=config.py  configTitle=pi-timolo ver 10.43 Default Settings
     Plugin .. pluginEnable=False  Disabled (Using Only config.py Settings)

Image Info ... Size=1280x720  Prefix=cam1-  VFlip=False  HFlip=False  Rotation=0  Preview=False
               JpegQuality=95 where 1=Low 100=High(Min Compression) 0=85
   Low Light.. nightTwilightThreshold=90  nightDarkThreshold=50  nightBlackThreshold=4
               nightMaxShutSec=5.90  nightMaxISO=800  nightDarkAdjust=4.70  nightSleepSec=30
   No Shots .. noNightShots=False   noDayShots=False
   Img Text .. On=True  Bottom=True (False=Top)  WhiteText=True (False=Black)  showTextWhiteNight=True
               showTextFontSize=36 px height

Motion ....... motionTrackOn=False  Motion Tracking is Disabled)

Time Lapse ... On=True  Prefix=tl-   Timer=300 sec   timelapseExitSec=3000 (0=Continuous)
               timelapseMaxFiles=0
   Img Path .. timelapsePath=/home/pi/pi-timolo/media/timelapse  timelapseCamSleep=4.00 sec
   Sched ..... timelapseStartAt  blank=Off or Set Valid Date and/or Time to Start Sequence
   Num Seq ... On=True  numRecycle=True  numStart=1000   numMax=9999  current=1046
   Num Path .. numPath=/home/pi/pi-timolo/tl-pi-timolo.dat
   Sub-Dir ... timelapseSubDirMaxHours=0 (0=off)  timelapseSubDirMaxFiles=0 (0=off)
   Recent .... timelapseRecentMax=400 (0=off)  timelapseRecentDir=media/recent/timelapse

Disk Space  .. spaceTimerHrs=0 (Disabled) - Manage Target Free Disk Space. Delete Oldest jpg Files
            .. Check Every spaceTimerHrs=0 (0=Off)  Target spaceFreeMB=500 (min=100 MB)

Logging ...... verbose=True (True=Enabled False=Disabled)
   Log Path .. logDataToFile=False  logFilePath=/home/pi/pi-timolo/pi-timolo.log
------------------------------------ Log Activity --------------------------------------------
2018-01-31 18:55:07 INFO     timolo     Start TimeLapse Loop ... ctrl-c Exits
2018-01-31 18:55:07 INFO     timolo     daymode=False  motionDotsOn=False 

2018-01-31 18:55:14 INFO     timolo     Sched TimeLapse  daymode=False  Timer=300 sec  ExitSec=17/3000 Status
2018-01-31 18:55:19 INFO     takeNightImage BlackThresh=1/4 shutSec=5.9000 camSleepSec=30 MaxISO=800 Size=1280x720
2018-01-31 18:56:20 INFO     writeTextToImage Added White Text [ 1046  20180131_18:56:19 ]
2018-01-31 18:56:20 INFO     writeTextToImage media/timelapse/tl-cam1-1046.jpg
2018-01-31 18:56:20 INFO     postImageProcessing Next Counter=1047 /home/pi/pi-timolo/tl-pi-timolo.dat
`

pageauc, 2018-01-31:
> 2018-01-31 18:54:55 INFO Testing if Pi Camera is in Use
> 2018-01-31 18:54:57 INFO Pi Camera is Available.
> 2018-01-31 18:54:57 INFO Start pi-timolo per /home/pi/pi-timolo/config.py

Yes that is true for night shots in your case 1/4 night means it is almost
black.  From the time the long exposure starts ithere is a few seconds of
camera warm up plus 30 seconds  to check white balance plus 5.9 seconds
exposure time and the extra time to process image.  I tried turning off
text to image and changing to png thinking jpg with high quality would
delay things but the best I can do is approx 60 seconds below pixAve of 4.
I also tried reducing nightSleepSec from 30 to 10 but you get inconsistent
image quality.  30 gives better white balance calculation.
So at very low light it seems like it takes approx 60 seconds.  This can be
reduced by reducing nightSleepSec but this will also affect quality of
night image.  You can give it a try and see.

BTW I reviewed code and decided to simplify day checks and checking pixel
Average by using videostream instead of camera mode (pixAve calculation is
only done during non day lighting conditions).  See release 10.50 that is
on GitHub Repo.

Twilight and nightDark will produce low light images much faster but
nightBlack looks like 60 seconds unless I missed something.

Your Feedback is appreciated.
Let me know if you think there is a way to improve the nightBlack threshold
speed.

Claude ...

pageauc, 2018-02-01:
Did some fixups that crashed single core RPI.  Also fixed nightl images
since there were a few greenish images.  Tested and this
appears to work OK on RPI3 and RPI2 single core.

Still cannot speed night shots

If you can dowlnoad ver pi-timolo.py version 10.60

 wget -O pi-timolo.py https://raw.github.com/pageauc/pi-timolo/master/
source/pi-timolo.py

Thanks Claude ...

closed 2018-02-03 by pageauc

tibaultbordeaux, 2018-02-06:
Hi Claude, sorry for my delay.

Thanks a lot for all your detailed explanations on night shots.
Sounds very reasonable.

Thanks again, I like your work!