pageauc/pi-timolo #89: Take only one single picture per session

tibaultbordeaux, 2018-03-03:
Dear Claude,

I have another feature request. Would love to hear from you what you think about.

I'm using pi-timolo in conjunction with a "witty-Pi 2" which boots up the RPi once a day and shutting it down five minutes or so later. In this time window it just has to take one single picture (of a plant, by the way). 
Therefor I would need a config to tell pi-timolo: Please; take only one picture and than exit. But of course I want to use the image numbering, so that the picture of the next day gets the next number.

What do you think?

Tibault

pageauc, 2018-03-03:
I would suggest you just set timelapseTimer to greater than the time for the RPI to be up.
Timelapse will take one image at the start and wait for timer period but the RPI will shutdown before that happens.  Next day the same thing will happen.  See sample config.py settings.  Note I set timelapseExitSec = 120  actually timelapseTimer = 600 would be redundant but I added both just in case RPI stays up longer than 600 seconds.

    timelapseTimer = 600        
    timelapseNumOn = True
    timelapseNumRecycle = False 
    timelapseNumStart = 1000    
    timelapseNumMax = 0    #  0=Continuous
    timelapseExitSec = 120  

On next startup the numbering will continue where is left off.
I assume you have motion turned off so did not have this setting above.

Give this a try and let me know.

Claude ...

pageauc, 2018-03-05:
I have commented on the github issue and example config.  See
https://github.com/pageauc/pi-timolo/issues/89

Let me know if this meets your needs.
Claude....

tibaultbordeaux, 2018-03-12:
Dear Claude,

I'm so sorry hat I didn't answer to your comments so far.

Thanks a lot. Right now I'm experimenting with your suggestings.
Good to know, that the numbering will continue where it left off - didn't realize that point.

While I would prefer that pi-timolo exits immediately after taking the picture this workaround is acceptable.

Thanks again for your work!

closed 2018-03-20 by pageauc