pageauc/pi-timolo #90: How to use convid.sh to join motion videos?

NelsonPJ, 2018-03-16:
I don't understand how convid.sh is supposed to be used to join motion videos together.

Starting with a default installation of the latest Pi-Timolo, I'm experimenting with taking 15 second long motion videos.  I see that it creates an h264 file in media/motion, and then it's quickly converted to an mp4 file and moved to media/videos (which is done by convid.sh, according to your wiki)

So now I've got a bunch of little mo-#####.mp4 files in media/videos that I'd like to join using "convid.sh join", but when I run it from the command line to test it, I get this error:

```
pi@picam:~/pi-timolo $ ./convid.sh join
================================================
INFO  : convid.sh 5.00  written by Claude Pageau
Batch Join MP4 Videos Using MP4Box
================================================
WARN  : No media/motion Files Found to Process.

```

From looking in the settings in video.conf, when each file is converted after it's taken, convid.sh script puts the converted video file in mp4_dest (media/videos)

But then when you run the "join" command, it's looking in mp4_source (media/motion) - which is now empty?  And it's going to put them in mp4_dest (media/videos) - which is where all the converted videos already are?

So I changed mp4_source to look in media/videos, and then ran "convid.sh join" to make converted videos named "convid-#####.mp4".  So that worked.

But when I ran "convid.sh" a second time, it joined all the "convid-#####" videos together, along with the newer "mo-#####" vidoes, and I ended up with one big convid-#####.mp4 file (!)

Shouldn't there be a different configuration option in video.conf for the "join" destination?  Or a way to keep the initially converted mp4 vidoes in the /media/motion directory?  Or have convid.sh join ignore vidoes prefixed with "convid"?  Or am I missing something?

These are the default settings in video.conf
```
mp4_fps=10                       # Encoding speed of h264 files
mp4_del_h264=true                # delete=true false= rename with .done extension

mp4_source="media/motion"        # Source Folder of mp4's to join
mp4_dest="media/videos"          # Destination Folder for mp4's
mp4_work="media/vid-work"        # Working folder for Encoding

mp4_source_h264=media/motion/*h264 # convertall Path and file type of source files to convert
mp4_video_prefix=convid          # Prefix of output join file
mp4_max_joins=12                 # Number of videos to join before starting New Join

```

pageauc, 2018-03-16:
See the wiki
https://github.com/pageauc/pi-timolo/wiki/How-to-Convert-h264-video-to-MP4#how-to-run-convidsh

You need to specify join parameter

    ./convid.sh join

It will then join files per mp4_max_joins variable.  The output will be a dated filename in per mp4_dest variable.
This feature was mainly written so you could join short motion videos (eg 10 seconds or so) and create longer videos of all the motion events.  It works best if run from a crontab where it will check for mp4 videos generated by pi-timolo and join them but you can run it manually.  You need to specify the join parameter for this to work.  Note the files to join must all have the same file prefix.  default is convid.  

Let me know if you have any questions or problems.
Claude ...

NelsonPJ, 2018-03-17:
I think the issue is that each time a motion capture video is taken, pi-timolo.py immediately calls convid.sh to convert it from *.h264 to *.mp4 (which is good) but when convid.sh is called with just a filename, it moves the converted file into the **mp4_dest** folder (which is set in video.conf to be media/vidoes).  So my media/motion directory is always empty, and media/videos fills up with little mp4 files.

I think convid.sh called with just a filename should put the converted file back in the same directory it came from (in this case, the media/motion directory)  

So then, like you mention, I could set up a crontab job with "./convid.sh join" once or twice a day, and that would combine all the mp4 videos in **mp4_source** (media/motion) in batches of 12, and move them to **mp4_dest** (media/videos).

Otherwise the only way I can join vidoes with "./convid.sh join" is to point it at the media/videos directory - but that will also have timelapse movies - as well as any previously combined videos.

PS: Fantastic job on this software - I experimented with it a few years ago on a Raspberry Pi B and had some fun - but recently came back to using it with and RPi3, which is a much better experience!  Also I'm really liking with your usage of rclone to sync with Google Drive, and then using that for remote management.

NelsonPJ, 2018-03-17:
Also - you wrote this:

> Note the files to join must all have the same file prefix. default is convid. that pi-timolo creates by default. 

But I don't think that's right - isn't that only used for the output filename?  (i.e. all joined files are called convid-#####.mp4).

pageauc, 2018-03-17:
There is a plugin called secvid.py  Default for video saving is media/secvid
You can then configure video.conf to pick up mp4 videos to join from that folder

    mp4_source="media/secvid

and save output videos to media/videos or another folder name.
That way there should be no conflict.  Using plugins makes it easy to switch configurations.

Let me know if this works for you

closed 2018-03-20 by pageauc