pageauc/pi-timolo #94: change rasp-pi camera to USB camera

lidatou1991, 2018-03-22:
Is it possible to change the default raspi-camera to USB cambera since I only have a webcam?

pageauc, 2018-03-22:
Due to the way pi-timolo was written it needs to have a pi-camera for better resolution for timelapse and motion images.  A while back I implemented motion tracking code from my speed camera project into pi-timolo.  This was to reduce motion false positives.  
If you are looking for something that can use either webcam or pi-camera see my speed-camera project.

    https://github.com/pageauc/speed-camera

I have included a plugins feature and there is a security cam plugin if that is what you are interested in.  The resolution for a webcam will not be as good especially if using a frame from a video stream Speed Camera include a lot of features from pi-timolo like rclone, menubox, webserver, plugins, Etc. 

Let me know if this works for you.
Claude ...

lidatou1991, 2018-03-22:
Claude, Thanks for your reply. Regarding the speed-cam project, I have switched from the rasp-pi cam to webcam. It works well with speed.py and webserver.py PID number shown.But it is strange make html page reports error reporting No such file or directory.

For plugins, I will try later

pageauc, 2018-03-22:
I will do some testing with makehtml.
Are you just testing software or do you have something in mind as a project or usage
Claude ...

lidatou1991, 2018-03-23:
Not just to test the software. I am planning to set up a traffic sensing station since I live by a main road in Chengdu China. Others could visit the site to view real-time traffic situation and statistical reporting. But I am a raspi newbie

pageauc, 2018-03-26:
I have done testing with makehtml.py and it works OK for me.  Send me details of your config.py and the details of error message.  Please NOTE makehtml.py has internal variables that may need to be changed if you are not using the default speed camera folder locations.  If you send me details, I will investigate and get back to you. 

> verbose = True
> 
> image_ext = ".jpg"
> source_csv = "speed-cam.csv"
> 
> web_html_dir = "media/html"
> if not os.path.isdir(web_html_dir):
>     logging.info("Creating html Folder %s", web_html_dir)
>     os.makedirs(web_html_dir)
> web_image_dir = "media/images/"   # location of image link off html folder
> 
> # contour width to height ratio
> guess_person = .55
> guess_cart = 1.1

lidatou1991, 2018-03-27:
`# Display and Log settings
# ------------------------
verbose = True              # True= Display basic status information on console False= Off
display_fps = False         # True= Show average frame count every 1000 loops False= Off
log_data_to_CSV = True      # True= Save log data as CSV comma separated values  False= Off
loggingToFile = True       # True= Send logging to file False= No Logging to File
logFilePath = 'speed-cam.log'  # Location of log file when logDataToFile=True

# Motion Event Settings
# ---------------------
SPEED_MPH = False      # Set Speed Units   kph=False  mph=True
MIN_AREA = 200         # Default= 200 Exclude all contours less than or equal to this sq-px Area
track_len_trig = 75    # Default= 75 Length of track to trigger speed photo
show_out_range = True  # Default= True Show Out of Range Events per x_diff settings below False= Off
x_diff_max = 15        # Default= 15 Exclude if max px away >= last motion event x position
x_diff_min = 1         # Default= 1 Exclude if min px away <= last event x position
track_timeout = 0.0    # Default= 0.0 Optional seconds to wait after track End (Avoid dual tracking)
event_timeout = 0.3    # Default= 0.3 seconds to wait for next motion event before starting new track
max_speed_over = 0     # Exclude track if Speed less than or equal to value specified 0=All

# Camera Settings
# ---------------
WEBCAM = True         # Default = False False=PiCamera True=USB WebCamera

# Web Camera Settings
WEBCAM_SRC = 0         # Default= 0   USB opencv connection number
WEBCAM_WIDTH = 640     # Default= 640 USB Webcam Image width
WEBCAM_HEIGHT = 480    # Default= 480 USB Webcam Image height
WEBCAM_HFLIP = True    # Default= False USB Webcam flip image horizontally
WEBCAM_VFLIP = False   # Default= False USB Webcam flip image vertically

# Pi Camera Settings
# ------------------
CAMERA_WIDTH = 640     # Image stream width for opencv motion scanning Default=320
CAMERA_HEIGHT = 480    # Image stream height for opencv motion scanning  Default=240
CAMERA_FRAMERATE = 25  # Default= 25 Frame rate for video stream V2 picam can be higher
CAMERA_ROTATION = 0    # Rotate camera image valid values are 0, 90, 180, 270
CAMERA_VFLIP = True    # Flip the camera image vertically if required
CAMERA_HFLIP = True    # Flip the camera image horizontally if required

# Camera Image Settings
# ---------------------
image_path = "media/images"   # folder name to store images
image_prefix = "speed-"       # image name prefix
image_format = ".jpg"         # Default = ".jpg"  image Formats .jpeg .png .gif .bmp
`

I did not make any changes on the directory path.

pageauc, 2018-03-27:
Can you retry running makehtml.py and send me the error message.  
makehtml.py just reads the csv file and creates individual html files in the media/html folder.  The html files are formatted so you can move up and down the files and view the image and associated data.  It has to be run manually and results viewed in browser.

Claude ...

lidatou1991, 2018-03-28:
Claude speed-cam now is operating on my machine with GUI window prompted up. while there is still some error message as following
`Traceback (most recent call last):
  File "./makehtml.py", line 201, in <module>
    read_from_csv(source_csv)
  File "./makehtml.py", line 158, in read_from_csv
    jpg_exists, next_link = check_row(row)
  File "./makehtml.py", line 130, in check_row
    img_path = row_data[5]
IndexError: list index out of range
`
The setting of config.py is as following
` Display and Log settings
# ------------------------
verbose = True              # True= Display basic status information on console$
display_fps = False         # True= Show average frame count every 1000 loops F$
log_data_to_CSV = True      # True= Save log data as CSV comma separated values$
loggingToFile = True       # True= Send logging to file False= No Logging to Fi$
logFilePath = 'speed-cam.log'  # Location of log file when logDataToFile=True
`

lidatou1991, 2018-03-28:
I checked the media directory, There is nothing in subfolder of html and images

pageauc, 2018-03-28:
Are you running this on Windows?
Make sure you have the latest GitHub files
Delete csv file and any subfolders in media/images folder
Delete files in media/html
Start speed camera and collect speed data and images.  
Once you have accumulated enough data (images in media/images dated subfolder) and speed data in the csv file 
Run makehtml.py   I would recommend you stop speed-cam.py before running makehtml.py just in case.  Once you know everything is running you can leave speed-cam.py running.
makehtml.py is designed to only be run occasionally and not too often especially if there are a lot of entries.
Claude ...

lidatou1991, 2018-03-29:
[image]
[image]
[image]

it is running via ssh from a Linux machine. There is no subfolders under html/images.

pageauc, 2018-03-29:
It looks like you do not have any speed data or images generated.  I will
add code in makehtml.py to check for data and prevent crash if no data
available.
I will also update wiki when I get a chance. Did you follow calibration
steps in wiki
https://github.com/pageauc/speed-camera/wiki/Calibrate-Camera-for-Distance

Try running speed-cam.py from the command line.
https://github.com/pageauc/speed-camera/wiki/How-to-View-Data#view-console-verbose-logging
.
Stop speed-cam.py if it is running in background.  Use menubox.sh to STOP
if it is running.
Then run speed camera from ssh session and view logging information.  Make
sure camera is pointed correctly and there is good lighting.

    ./speed-cam.py

Generate some speed data and images manually by tracking movement in front
of the camera
This can be done by pointing correctly at a roadway per wiki calibration
instructions.
https://github.com/pageauc/speed-camera/wiki/Calibrate-Camera-for-Distance
or manually by generating tracking movement in front of camera.  You should
see tracking data and saving of images logging message with location of
folder where
images are save.  Make sure you have several entries.
ctrl-c to exit speed-cam.py.  Check for a subfolder in media/images  there
should be a subfolder with a dated name.  Check to make sure that subfolder
contains
jpg files
Also ensure there are entries in speed-cam.csv file

    cat speed-cam.csv

I will update the speed-camera wiki to be a bit clearer and add details for
trouble shooting.

Let me know results
Claude ...

pageauc, 2018-03-29:
> Calibrate-Camera-for-Distance
> Data#view-console-verbose-logging

Can you create a new github issue under speed-camera.

I recently added plugins to speed-camera and added capability to use
speed-cam as a security camera.  The documentation and wiki are not clear
on this so I need to do some work to make this easier.   I will do some
code changes to speed-cam.py to add a SECURITY_CAM_MODE setting to
config.py that will put camera in security camera mode without a plugin
needed.  This will use the full image for motion tracking and disable logic
for calculating speed since it would not be needed in security cam mode.  I
will also look at adding IP camera support at the same time.

This will take a while since we are about to migrate north after our winter
here in Texas.

Claude ...

closed 2018-03-29 by pageauc