pageauc/pi-timolo #95: IOError: [Errno 2] No such file or directory: 'media/timelapse/...'

dirkclemens, 2018-07-04:
All,
In a plain new installation I get the following error message. I use the original config.py and I am running Python 2.7.3 on a Raspberry Pi Wheezy with Kernel 4.1.19+

```
Logging ...... verbose=True (True=Enabled False=Disabled)
   Log Path .. logDataToFile=False  logFilePath=/opt/pi-timolo/pi-timolo.log
--------------------------------- Log Activity ---------------------------------
2018-07-04 20:11:33 INFO     timolo     Start Motion Tracking and TimeLapse Loop ... ctrl-c Exits
2018-07-04 20:11:33 INFO     timolo     daymode=True  motionDotsOn=False

2018-07-04 20:11:33 INFO     timolo     Sched TimeLapse  daymode=True  Timer=300 sec  ExitSec=0 0=Continuous
2018-07-04 20:11:33 INFO     timolo     Stop PiVideoStream ...
Traceback (most recent call last):
  File "/opt/pi-timolo/pi-timolo.py", line 1865, in <module>
    timolo()
  File "/opt/pi-timolo/pi-timolo.py", line 1539, in timolo
    takeDayImage(filename, timelapseCamSleep)
  File "/opt/pi-timolo/pi-timolo.py", line 964, in takeDayImage
    camera.capture(filename, quality=imageJpegQuality)
  File "/usr/lib/python2.7/dist-packages/picamera/camera.py", line 1570, in capture
    encoder.start(output)
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 1543, in start
    super(PiCookedOneImageEncoder, self).start(output)
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 819, in start
    self._open_output(output)
  File "/usr/lib/python2.7/dist-packages/picamera/encoders.py", line 774, in _open_output
    output = io.open(output, 'wb', buffering=65536)
IOError: [Errno 2] No such file or directory: 'media/timelapse/tl-cam1-1000.jpg'
```

pageauc, 2018-07-04:
I have pi-timolo running on wheezy.
There are issues with wheezy like no SSL support so pip wont work.  If you take a look at the pi-timolo.install.sh you will see my work around to update pip so SSL support works.  There are also issues with PIL and image for python3.  Some of these issues are due to deprecated packages and wheezy no longer being supported.  My advise is to run python2 only.

Did you install pi-timolo.py from the curl script?  If not then run the UPGRADE menu pick in menubox.sh.  That will run the curl script and update pi-timolo to latest version.
With regards to your error
WRT your error

The media/timelapse folder is automatically created at line 631 of pi-timolo.py and the code did not error out there so you must have a media/timelapse folder.  The problem I see is the error is referring to /opt/pi-timolo/pi-timolo.py.  Not sure why there should be an install at /opt  Install should be at /home/pi/pi-timolo.  There should be no reference to /opt.  Check your /opt folder and see if for some reason there is a pi-timolo folder there. 

Let me know
Claude ...

closed 2018-07-09 by dirkclemens

pageauc, 2018-07-09:
Noticed you closed this item.  Did you resolve the problem and what was the problem
Thanks Claude ...

dirkclemens, 2018-07-09:
I’ve chosen another software, just a PIR sensor and picamera-code, absolutely sufficient for my approach. 
Cheers
Dirk