pageauc/pi-timolo #96: Cannot Ctrl+C in pi-timolo.py

blokdk, 2018-07-17:
When I start ./pi-timolo.py it shows a lot of info and below ------ log Activity ---------
it shows 5 lines of INFO. The last line contains takeNightImage ......
It seems that nothing is happening. So I wave with my hands above the camera and wait a while.
Then I try Ctrl+C to exit. But I am not returned to anything I know. Every command is just sort of echoed.
I see no way to return to the terminal.

Please advise.

pageauc, 2018-07-18:
If camera is in night mode for timelapse or motion, then long exposures will be taken.  Depending on the amount of light this can take a while.  Night motion is very slow and object has to be in camera view for a while to trigger.  I would recommend testing motion in brighter light.  Motion Tracking is used where the object has to move a specified distance in x and/or y dimension in order to trigger an image.  Waving a hand in front of the camera may or may not activate the trigger.  Motion tracking reduces false positives motion like bushes moving back and forth since the would not travel the trigger distance.  A motion tracking video stream image can alternatively be configured to eliminate the delay in taking a still image but the stream image is lower quality.
Timelapse will automatically adapt settings for twilight mode.  

See the wiki for details.  https://github.com/pageauc/pi-timolo/wiki

Let me know your progress
Claude ..

blokdk, 2018-07-18:
Many thanks for your reaction. I think I can deduce that the Ctrl+C is not working because timolo is so busy with taking a 'night' picture?
I made a complete new Raspbian install on my pi3 this morning; installed your software and brought the pi into the brightness. That worked fine. 
No I am going to play with it, feeded with a battery pack and putting it on a location where it will hopefully trap the woodpecker that visited our garden yesterday morning.
Thank you!

closed 2018-07-18 by blokdk

pageauc, 2018-07-18:
You should take a look at the default config.py settings.  By default motion track and timelapse are both turned on.  If you just want motion track then edit the config.py file using nano.  Turn timelapse off per

    timelapseOn = False

Also motion tracking has some settings you may need to change to track small things depending on how close the camera will be to the birds.  If they going to be within 5 or 10 feet then you should be OK but if you are not catching birds the you can adjust

    motionTrackMinArea = 100

Another option is to take a series of pictures when motion is detected per the following settings

    motionTrackQuickPic = False
    motionQuickTLOn = True
    motionQuickTLTimer = 20     
    motionQuickTLInterval = 0

This will take As many stills as fast as possible for 20 seconds or whatever motionQuickTLtimer seconds you specify.

Note it takes time for pi-timolo to switch from streaming mode to camera still mode.  Still mode will take the best image quality.  For very fast action where objects may be out of frame by the time the still image is taken, but at the cost of quality you can have motion track take a stream image per motionTrackQuickPic setting.  I don't think this will be necessary though

    motionTrackQuickPic = True

Video is also an option but if there is a lot of activity then you might trigger a lot of video sequences that take a lot disk storage space.

    motionTrackQuickPic = False
    motionQuickTLOn = False
    motionVideoOn = True      
    motionVideoFPS = 24         
    motionVideoTimer = 20

I have one of my pi's setup with a USB battery pack and a hard drive mounted on a tablet stand I found that allows for adjusting camera angle.  I can use wifi in the back yard and just view media using the web browser.

I use Filezilla with SFTP protocal to access the RPI.  You can download and/or upload using drag and drop from another computer on the network.  Another option is to use a removable memory stick but this takes a bit more knowledge to setup a mount point.

Most information will be in the wiki.   Let me know if you need help.
Claude ...